dsmrreader/dsmr-reader #1102: Hoge CPU load

royheidotting, 2020-09-03:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

Wat gebruik je?
- DSMR-reader versie:   2.15.1
- Type RaspberryPi of server:   Intel NUC DN2820FYKH
- Standaardinstallatie of Docker:  Standaardinstallatie
- DSMR-protocol slimme meter v4/v5:   Heb zo snel geen idee

---

Het postgres process vreet erg veel CPU.
[image]
Hierdoor wordt de gehele setup een stuk langzamer.

Af en toe schiet de website van DSMR-reader in 504 (504 Gateway Time-out)

Ik heb de data retentie terug gezet naar een maand. 
Is er iets wat ik hier aan kan doen?

dennissiemensma, 2020-09-03:
Bedankt voor je melding. Ik denk dat er nog wat extra onderzoek mag komen:

- Welke DSMR-versie heeft je meter?
- Waar staat je datalogger-sleep op?
- Hoe groot is je database?
```
sudo su - postgres
psql dsmrreader
SELECT nspname || '.' || relname AS "relation",
    pg_size_pretty(pg_total_relation_size(C.oid)) AS "total_size"
  FROM pg_class C
  LEFT JOIN pg_namespace N ON (N.oid = C.relnamespace)
  WHERE nspname NOT IN ('pg_catalog', 'information_schema')
    AND C.relkind <> 'i'
    AND nspname !~ '^pg_toast'
  ORDER BY pg_total_relation_size(C.oid) DESC
  LIMIT 20;
```

- Verder kun je kijken of de DB-indexes op de grootste tabellen staan. Zie dit comment voor hoe en wat: https://github.com/dennissiemensma/dsmr-reader/issues/1069#issuecomment-670166046

- Tot slot zie ik dat je nog DSMR-reader v2.x draait. Dat kan uiteraard prima, als het draait dan draait het, maar ik weet niet uit mn hoofd welke performance-verbeteringen er sindsdien zijn gedaan. Mogelijk win je ook nog wat met upgraden, maar dat is geen garantie. Eerst de andere punten maar.

royheidotting, 2020-09-04:
Hallo Dennis, 
Bedankt voor je reactie zo ver.

- De meter heeft DSMR v5
- De datalogger sleep staat op 0.5
- Grootte van de database:
[image]

[image]
Deze komen wel overeen met in het door jou genoemde voorbeeld.

[image]
Deze komen **niet** overeen met in het door jou genoemde voorbeeld, onderstaande 4 rows mis ik.
"dsmr_consumption_electrici_phase_power_current_l1_3d9fd70d" btree (phase_power_current_l1)
"dsmr_consumption_electricityconsumption_2241f8ee" btree (phase_currently_delivered_l2)
"dsmr_consumption_electricityconsumption_5714c500" btree (phase_currently_delivered_l1)
"dsmr_consumption_electricityconsumption_e90842a6" btree (phase_currently_delivered_l3)

[image]
Deze komen wel overeen met in het door jou genoemde voorbeeld.

Wat betreft versie 2.x, ik heb nog niet de tijd kunnen vinden en het risico durven nemen om up te graden.

dennissiemensma, 2020-09-04:
Bedankt voor de info. Ik denk dat ik al zie waardoor het komt. 

Je hebt 5 GB aan metingen opgeslagen en dat is teveel, hoewel je geen Pi draait. 

- Het instellen van de retentie is al 1 ding wat je goed hebt gedaan, maar je zult de datalogger sleep ook echt wat moeten ophogen. Mijn advies is om die datalogger sleep op `10` seconden te zetten. Dan heb je 6 telegrammen per minuut ipv 60 en dat scheelt je dus 10x zo weinig data aan metingen. Je kunt hem ook op 5 seconden ofzo zetten, maar als je niets met de individuele metingen doet, zou ik het zo hoog mogelijk zetten. Voor de uur- en dagtotalen maakt het immers niet uit!

- Als ik even snel terugkijk in de code van v2.15 zie ik dat de retentie doorgaans constant draait op de achtergrond (dat was eerder niet zo), dus na verloop van tijd slinkt je data vanzelf. Dat kan zomaar enkele dagen duren, maar gedurende dat je data steeds kleiner wordt, gaat dat ook sneller. Je kunt de voortgang zien met de queries uit deze comment: https://github.com/dennissiemensma/dsmr-reader/issues/1071#issuecomment-670690844 (aantal records per maand)

- Upgraden zou ik afhouden als het niet nodig is. Kijk even hoe het performt als je dataset eenmaal geslinkt is. Ik denk verder dat die missende indexes in een van de latere releases zijn. Als je vanaf 2.15 omhoog scrollt zie je dat er heel veel wijzigingen zijn geweest en ze zouden zomaar onderdeel daarvan kunnen zijn: https://dsmr-reader.readthedocs.io/en/latest/changelog.html#v2-15-0-2020-01-15. Niet erg dus.

bikeymouse, 2020-09-05:
Ha Dennis,

Je geeft aan:
_Mijn advies is om die datalogger sleep op 10 seconden te zetten. Dan heb je 6 telegrammen per minuut ipv 60 en dat scheelt je dus 10x zo weinig data aan metingen._
Dat is inderdaad ook zoals ik hem gebruik, ik zie echter dat deze optie in V4 (in ieder geval voor de Docker versie) is verdwenen?

royheidotting, 2020-09-07:
De CPU load is al een heel stuk lager, hoewel deze nog redelijk hoog is.

[image]

De Database wordt nog niet direct kleiner;
[image]

Huidige status van het aantal berichten:
[image]

Ik laat het even een weekje rusten en de nieuwe instellingen z'n werk doen.

Voor zover enorm bedankt voor je hulp en ondersteuning.

dennissiemensma, 2020-09-07:
Je kunt kijken of dit wat schijfruimte teruggeeft, gezien de aantallen metingen al flink teruggelopen zijn:

```
sudo su - postgres
vacuumdb -f -v -d dsmrreader
```

royheidotting, 2020-09-09:
Dat heeft behoorlijk geholpen.

[image]

Van 5GB naar ongeveer 450MB.
Het postgres process gebruikt nu nog 10% van de CPU.
[image]

Wat volgens mij wel acceptabel is.

Issue mag wat mij betreft gesloten worden, tenzij je nog een extra suggestie hebt.

dennissiemensma, 2020-09-09:
Bedankt voor je terugkoppeling. In dat geval zou ik het eerst zo aankijken.

Als je nog betere performance wil dan is mijn advies om de datalogger sleep nog hoger te zetten. Over ongeveer een maand (gezien dat je retentie-instelling is) houd je veel minder metingen over omdat dan alles geroteerd is en er niet meer elke seconde een meting bij komt. Ik verwacht dat dat de performance en datagrootte nog meer ten goed komt.

dennissiemensma, 2020-09-09:
En nog even ter vergelijking met mijn situatie, een DSMR-v4 meter die elke 10 seconden een telegram uitspuugt en met retentie van een week:

```
DATA
  Telegram records total                                     141768
                \_ unprocessed                                    5
  Electricity consumption records stored                      91509
  Gas consumption records stored                              40821

POSTGRESQL SIZE OF LARGEST TABLES
  public.dsmr_consumption_electricityconsumption              32 MB
  public.dsmr_datalogger_dsmrreading                          32 MB
  public.dsmr_stats_hourstatistics                          4552 kB
  public.dsmr_consumption_gasconsumption                    4256 kB
  public.dsmr_weather_temperaturereading                    3840 kB
```

closed 2020-09-09 by dennissiemensma