dsmrreader/dsmr-reader #1103: Firefox / Configuration

Edwink18, 2020-09-03:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

Wat gebruik je?
- DSMR-reader versie:   4.3
- Type RaspberryPi of server:  Raspberry in Virtualbox.
- Standaardinstallatie of Docker:   Standard install
- DSMR-protocol slimme meter v4/v5:  v5
---

In Firefox versie 80.0.1 lijkt het configuratie menu niet helemaal goed te werken
in firefox is het niet mogelijk om settings te veranderen.  In Google Chrome werkt het wel.

dennissiemensma, 2020-09-03:
Bedankt voor je melding. Zou je een screenshot willen toevoegen of aangeven welke (relatieve) URL binnen de applicatie het is?

Edwink18, 2020-09-03:
Screenshot van de Firefox email settings in het configuratie menu

[image]

dennissiemensma, 2020-09-03:
Bedankt voor de opheldering. Hiervoor geldt hetzelfde verhaal als in #1100. 

- Zou je eens willen kijken of dit wat oplost: https://github.com/dennissiemensma/dsmr-reader/issues/1100#issuecomment-685072450 ?
- Wellicht zie je nog een foutmelding in de "console" van je browser, als je op F12 drukt en de pagina herlaadt?

Edwink18, 2020-09-03:
Nog een extra ook bij energy supplier prices gaat het niet goed. Zelfde resultaat.  In Google Chrome gaat dit allemaal prima

[image]

Edwink18, 2020-09-03:
De oplossing in #1100 heeft ook dit probleem opgelost. Setting zij weer te wijzigen. Webpage ziet er weer uit zoals je zou verwachten.

dennissiemensma, 2020-09-03:
Bedankt voor je snelle proberen en terugkoppeling! Dan weet ik dat dat het oplost.

closed 2020-09-03 by dennissiemensma