dsmrreader/dsmr-reader #1105: Allow Embedding DSMR reader v4.x

jimz011, 2020-09-05:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

Wat gebruik je?
- DSMR-reader versie:  4.3
- Type RaspberryPi of server:  server (unraid)
- Standaardinstallatie of Docker:  Docker
- DSMR-protocol slimme meter v4/v5:   ESMR 5

---

Ik wil graag dit embedden in Home Assistant, maar ik mag het niet embedden. Voorheen kon ik dit instellen met settings.py om de volgende regel toe te voegen:
```
X_FRAME_OPTIONS = 'ALLOWALL'
```
Er staat nu dat dit in een `.env` file moet. Helaas kan ik die file nergens vinden in de docker install (maar settings.py zelf geeft ook aan dat het niet meer werkt en ik .env moet gebruiken). Kan ik dit anders ook ergens in de settings (UI) aanpassen? Of moet dit echt met een variabele?

Mijn dank voor uw hulp en fantastisch programma overigens!

Groeten jimz011

dennissiemensma, 2020-09-05:
Bedankt voor je vraag. Zie https://dsmr-reader.readthedocs.io/nl/v4/faq/v4_upgrade.html#unsupported-settings en #1085 / #1080. Ik zal de optie later aan .env toevoegen.

jimz011, 2020-09-06:
Dit heeft gewerkt, enorm bedankt!

closed 2020-09-06 by jimz011

dennissiemensma, 2020-10-03:
Vanaf `v4.5` is dit straks in te stellen (overschrijven) via de env als `DJANGO_USE_X_FRAME_OPTIONS`.