dsmrreader/dsmr-reader #1106: fout na upgrade van v3 naar v4

DennisIOT, 2020-09-05:
Dag Dennis,

Na een upgrade van v3 naar 4 kom ik op de web pagina niet verder als het onderstaande en eigenlijk weet ik niet waar ik moet zoeken.
Waar heb ik het gedaan na alle stappen doorlopen te hebben.

Server Error
Sorry, something unexpected happened.
Exception:
ProgrammingError: column dsmr_frontend_frontendsettings.always_require_login does not exist
LINE 1: ...ntend_frontendsettings"."tariff_2_returned_name", "dsmr_fron...
                                                             ^
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 73, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 9, in dispatch
    if not FrontendSettings.get_solo().always_require_login:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1303, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1154, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

dennissiemensma, 2020-09-06:
Bedankt voor je melding. Lost dit het wellicht op? https://github.com/dennissiemensma/dsmr-reader/issues/944#issuecomment-605651767

DennisIOT, 2020-09-06:
Hi Dennis, geen geluk.
Onderstaande fouten komen direct terug

pi@raspberrypi:~ $ sudo su - dsmr
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ ./manage.py dsmr_sqlsequencereset

Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils                             .py", line 82, in _execute
    return self.cursor.execute(sql)
psycopg2.ProgrammingError: relation "dsmr_datalogger_meterstatisticschange" does not exist
LINE 25: ...alesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_data...
                                                              ^

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/_                             _init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/_                             _init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/b                             ase.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/b                             ase.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_sqlsequencereset.py", line                              30, in handle
    cursor.execute(sql)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils                             .py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils                             .py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils                             .py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", lin                             e 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils                             .py", line 82, in _execute
    return self.cursor.execute(sql)
django.db.utils.ProgrammingError: relation "dsmr_datalogger_meterstatisticschange" does not exi                             st
LINE 25: ...alesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_data...
                                                              ^

(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ ./manage.py migrate
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying auth.0012_alter_user_first_name_max_length...Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.IntegrityError: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(155) already exists.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 85, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 245, in handle
[... 16 lines cut ...]
    force_update, using, update_fields,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 892, in _save_table
    results = self._do_insert(cls._base_manager, using, fields, returning_fields, raw)
[... cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(155) already exists.

(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $

dennissiemensma, 2020-09-06:
Je kunt het handmatig proberen:

```
sudo su - postgres
psql dsmrreader
```

Deze queries dan uitvoeren:
```
BEGIN;
SELECT setval(pg_get_serial_sequence('"django_admin_log"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_admin_log";
SELECT setval(pg_get_serial_sequence('"auth_permission"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_permission";
SELECT setval(pg_get_serial_sequence('"auth_group_permissions"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_group_permissions";
SELECT setval(pg_get_serial_sequence('"auth_group"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_group";
SELECT setval(pg_get_serial_sequence('"auth_user_groups"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_user_groups";
SELECT setval(pg_get_serial_sequence('"auth_user_user_permissions"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_user_user_permissions";
SELECT setval(pg_get_serial_sequence('"auth_user"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_user";
SELECT setval(pg_get_serial_sequence('"django_content_type"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_content_type";
SELECT setval(pg_get_serial_sequence('"dsmr_api_apisettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_api_apisettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_backendsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_backendsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_emailsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_emailsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_scheduledprocess"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_scheduledprocess";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_backupsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_backupsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_dropboxsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_dropboxsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_emailbackupsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_emailbackupsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_energysupplierprice"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_energysupplierprice";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_electricityconsumption"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_electricityconsumption";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_gasconsumption"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_gasconsumption";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_consumptionsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_consumptionsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_dataloggersettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_dataloggersettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_retentionsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_retentionsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_dsmrreading"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_dsmrreading";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_meterstatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_meterstatistics";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_meterstatisticschange"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_meterstatisticschange";
SELECT setval(pg_get_serial_sequence('"dsmr_frontend_notification"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_frontend_notification";
SELECT setval(pg_get_serial_sequence('"dsmr_frontend_frontendsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_frontend_frontendsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_frontend_sortedgraph"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_frontend_sortedgraph";
SELECT setval(pg_get_serial_sequence('"dsmr_mindergas_mindergassettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mindergas_mindergassettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_mqttbrokersettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_mqttbrokersettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_jsondaytotalsmqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_jsondaytotalsmqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_splittopicdaytotalsmqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_splittopicdaytotalsmqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_rawtelegrammqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_rawtelegrammqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_jsontelegrammqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_jsontelegrammqttsettings";
[... cut ...]
SELECT setval(pg_get_serial_sequence('"dsmr_influxdb_influxdbintegrationsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_influxdb_influxdbintegrationsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_influxdb_influxdbmeasurement"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_influxdb_influxdbmeasurement";
COMMIT;

        BEGIN;
        SELECT setval(
            pg_get_serial_sequence('"django_migrations"','id'), coalesce(max("id"), 1), max("id") IS NOT null
        ) FROM "django_migrations";
        COMMIT;
```

Daarna nog een keertje:
```
sudo su - dsmr
./manage.py migrate
```

Geeft dat een ander resultaat?

DennisIOT, 2020-09-06:
Hi Dennis,
Dit was hem, tikken en lopen ja waanzinnige applicatie werkt weer. 
Waar het nu fout is gegaan snap ik niet echt, terwijl ik de stappen allemaal doorlopen heb.
Thanks for now again. !!

closed 2020-09-07 by dennissiemensma