dsmrreader/dsmr-reader #1107: Onregelmatig lezen van Telegrams door data logger

bikeymouse, 2020-09-06:
Wat gebruik je?
- DSMR-reader versie:   V4.3
- Type RaspberryPi of server:   Intel NuC / Proxmox
- Standaardinstallatie of Docker:   Docker xirixiz/dsmr-reader-docker:latest
- DSMR-protocol slimme meter v4/v5:   V5

Ha Dennis,
Na upgrade van V3.12 naar V4.3 werkt alles schijnbaar prima, maar ik zag in het status scherm toch regelmatig dat de inkomende Telegrammen niet iedere minuut binnenkomen. 

Als ik de datalogger uitlezingen bekijk, zie ik dat de telegrammen voor de upgrade nog exact iedere minuut binnenkomen, echter dat er na de upgrade soms een paar minuten geen updates zijn en dan weer 3 binnen 1 minuut.

(ik heb "datalogger process sleep" op 60 seconden staan en "backend process sleep" op 30 seconden want geen behoefte aan meerdere updates per minuut).

Dit lijkt heel erg op het issue wat ik eerder had toen de datalogger was aangepast. Dat heb je toen teruggedraaid, [zie  #894](https://github.com/dennissiemensma/dsmr-reader/issues/894)

Wellicht is er weer iets veranderd wat dezelfde issue terugbrengt?
[image]

dennissiemensma, 2020-09-06:
Ik denk dat je het beste kunt beginnen met DEBUG loglevel aan te zetten: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging-levels

En daarna in de log van de datalogger kijken. Die zou heel gedetaileerd moeten aangeven wat die aan het doen is. Voor het gemak zou je de sleep van de dataloggen tijdens het debuggen even op bijvoorbeeld 10 seconden kunnen zetten, en daarna steeds wat ophogen.

bikeymouse, 2020-09-06:
Hi,
Ik heb in de .env file in de DSMR subdirectory deze regel gezet:
DSMRREADER_LOGLEVEL=DEBUG

Vervolgens de Docker-container DSMR opnieuw opgestart.

In /var/log/supervisor/dsmr_backend.log zie ik netjes zoals verwacht iedere 30 seconden:
```
[2020-09-06 20:03:59,472] DEBUG    SP: 1 backend service(s) ready to run
[2020-09-06 20:03:59,472] DEBUG    SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-09-06 20:03:59,493] DEBUG    Compact: Processed reading: 918107 @ 2020-09-06 20:02:33+02:00
[2020-09-06 20:03:59,498] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-09-06 20:04:00.494103+02:00 (ETA 0:00:00.995535)
[2020-09-06 20:03:59,502] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 30.0s
[2020-09-06 20:04:29,523] DEBUG    SP: 1 backend service(s) ready to run
[2020-09-06 20:04:29,524] DEBUG    SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-09-06 20:04:29,535] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-09-06 20:04:30.533643+02:00 (ETA 0:00:00.998420)
etc.
```
Als ik echter in /var/log/supervisor/dsmr_datalogger.log kijk zie ik voor iedere keer dat DSM is opgestart deze regel:

```
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
```
Het lijkt er dus op dat het DSMRREADER_LOGLEVEL=DEBUG statement wel impact heeft op de logging van dsmr_backend.log heeft maar niet op dsmr_datalogger.log? 

Is er een andere env variabele die ik hiervoor moet zetten?

Verder heb ik de datalogger sleep en backend sleep op 6 seconden gezet, wat zou moeten resulteren in 10 metingen per minuut. Ik echter dan ook dat dit meestal maar 9 (en soms 8 of 10) metingen zijn, dus blijkbaar mist hij er dan ook vaak een meting.

Is er wat aan de datalogging veranderd voor de seriële poort (vorige keer resulteerde dat in buffer-overruns)?

dennissiemensma, 2020-09-06:
Staat er ook niets als je tailt op wildcard "dsmr_datalogger.log*"?

dennissiemensma, 2020-09-13:
@bikeymouse ben je hier nog verder mee gekomen?

bikeymouse, 2020-09-13:
Hi Dennis,

Nee eigenlijk niet. 
> Staat er ook niets als je tailt op wildcard "dsmr_datalogger.log*"?

Nee, er komt - behalve na ieder restart - geen extra logging in dsmr_datalogger.log* dus ik kan niet meer verder debuggen. 

In dsmr_backend.log krijg ik wél netjes logregels die beginnen met DEBUG, waaruit ik opmaak dat het commando DSMRREADER_LOGLEVEL=DEBUG in de .env file wel degelijk impact heeft. Maar blijkbaar niet op dsmr_datalogger.log

Zoals verder ook aangegeven, het variëren van de sleep-time bevestigt alleen het beeld: onafhankelijk van de sleep time zie je gewoon dat een bepaalt % van de metingen niet binnenkomt.

Zoals eerder aangegeven: dit is exact hetzelfde gedrag als ik eerder constateerde in #894 weet je zeker dat de oorzaak hetzelfde is?

Ik hoor graag of/wat ik nog kan doen om te achterhalen wat het issue is.

dennissiemensma, 2020-09-13:
De datalogger is op de schop gegaan, inclusief de logging, die nu nog uitgebreider zou moeten zijn.

Ik zie nu dat ik een typo heb gemaakt in de wildcard, dat kan wat verklaren, het moet dit zijn:

```
sudo tail -f /var/log/supervisor/dsmr_datalogger*
```

Want bij mij is de file bijvoorbeeld `/var/log/supervisor/dsmr_datalogger-stderr---supervisor-obQi8C.log` en dan matcht die niet op de wildcard uit eerdere comment.

----

Als voorbeeld van de output:

```
==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-obQi8C.log <==
....
1-0:21.7.0(00.103*kW)
1-0:41.7.0(00.114*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(...)
0-1:24.2.1(200913220000S)(03460.014*m3)
!6992
[2020-09-13 22:40:26,432] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 2.0s
[2020-09-13 22:40:28,686] DEBUG    [2020-09-13 22:40:28.686367] Read 0 Byte(s)
[2020-09-13 22:40:28,938] DEBUG    [2020-09-13 22:40:28.938098] Read 0 Byte(s)
[2020-09-13 22:40:29,189] DEBUG    [2020-09-13 22:40:29.189811] Read 0 Byte(s)
[2020-09-13 22:40:29,441] DEBUG    [2020-09-13 22:40:29.441439] Read 0 Byte(s)
[2020-09-13 22:40:29,693] DEBUG    [2020-09-13 22:40:29.693101] Read 0 Byte(s)
[2020-09-13 22:40:29,944] DEBUG    [2020-09-13 22:40:29.944762] Read 0 Byte(s)
[2020-09-13 22:40:30,196] DEBUG    [2020-09-13 22:40:30.196421] Read 0 Byte(s)
[2020-09-13 22:40:30,447] DEBUG    [2020-09-13 22:40:30.447841] Read 409 Byte(s)
[2020-09-13 22:40:30,699] DEBUG    [2020-09-13 22:40:30.699393] Read 453 Byte(s)
[2020-09-13 22:40:30,950] DEBUG    [2020-09-13 22:40:30.950560] Read 7 Byte(s)
[2020-09-13 22:40:30,957] DEBUG    Received telegram:
...
```

----

Kun je het nogmaals proberen en kijken of je wellicht een patroon of hangende datalogger ziet in debug mode?

bikeymouse, 2020-09-13:
Ha Dennis,

Nee met een andere wildcard geeft geen verschil:

```
bash-5.0# tail -f /var/log/supervisor/dsmr_datalogger*
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging

```
In de log directory heb ik ook maar 3 files:

```
bash-5.0# ls -alh /var/log/supervisor/
total 164K   
drwxr-xr-x    2 root     root        4.0K Sep  5 18:20 .
drwxr-xr-x    1 root     root        4.0K Sep  5 18:19 ..
-rw-r--r--    1 root     root      142.4K Sep 13 22:52 dsmr_backend.log
-rw-r--r--    1 root     root         864 Sep 13 21:42 dsmr_datalogger.log
-rw-r--r--    1 root     root        2.0K Sep 13 22:40 dsmr_webinterface.log
```

bikeymouse, 2020-09-13:
Nog even gezocht of er nog elders logfiles staan, maar dat levert niets extra's:
```
bash-5.0# find -name *.log
./tmp/supervisord.log
./tmp/nginx-stdout---supervisor-m6jqa6od.log
./tmp/nginx-stderr---supervisor-gnubfksk.log
./tmp/dsmr_datalogger-stderr---supervisor-clle91m3.log
find: ./proc/tty/driver: Permission denied
./var/log/supervisor/dsmr_backend.log
./var/log/supervisor/dsmr_webinterface.log
./var/log/supervisor/dsmr_datalogger.log
./var/log/nginx/access.log
./var/log/nginx/error.log
```
Ter info, ik gebruik het Docker image xirixiz/dsmr-reader-docker:latest

dennissiemensma, 2020-09-14:
Bedankt voor de update. Ik zie nu dat ik er initieel overheen gelezen had dat het een Docker-installatie betreft.

Is het dan niet de logfile `/tmp/dsmr_datalogger-stderr---supervisor-clle91m3.log` uit je laatste dump? En ik neem aan dat je telkens de processen herstart nadat de loglevel hebt aangepast he?

En anders kun je overwegen om het handmatig te draaien met debug logging ingeschakeld, dan kun je sowieso niet missen want het zou naar de stdout/stderr moeten gaan:

```
sudo supervisorctl stop dsmr_datalogger

sudo su - dsmr
./manage.py dsmr_datalogger

<afbreken kan met CTRL+C>

logout
sudo supervisorctl start dsmr_datalogger
```

Die laatste werkt ook als ik het zelf even probeer.

bikeymouse, 2020-09-14:
> Is het dan niet de logfile `/tmp/dsmr_datalogger-stderr---supervisor-clle91m3.log` uit je laatste dump?

Inderdaad! Stom data ik dat zelf niet gezien had.. Anyway nu dus wel wat verder met debuggen:

Als ik de log bekijk zie ik de volgende metingen tussen 22:13 en 22:15 uur, waarin opvalt dat er 4 "reads" lijken te zijn om 22:13 waarvan er 2 met 0 bytes. 
Verder ook 1 meting om 22:14 (wat ook raar is, want die staat niet in de database, zie verderop):
```
[2020-09-13 22:12:01,775] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 60.0s
[2020-09-13 22:13:02,054] DEBUG    [2020-09-13 22:13:02.054689] Read 871 Byte(s)
[2020-09-13 22:13:02,305] DEBUG    [2020-09-13 22:13:02.305231] Read 0 Byte(s)
[2020-09-13 22:13:02,555] DEBUG    [2020-09-13 22:13:02.555759] Read 0 Byte(s)
[2020-09-13 22:13:02,806] DEBUG    [2020-09-13 22:13:02.806088] Read 942 Byte(s)
[2020-09-13 22:13:02,807] DEBUG    Received telegram:
/ISK5\2M550T-1012

1-3:0.2.8(50)
0-0:1.0.0(200913221257S)
0-0:96.1.1(4530303434303037323531393537363138)
1-0:1.8.1(007318.932*kWh)
1-0:1.8.2(002599.010*kWh)
1-0:2.8.1(001528.442*kWh)
1-0:2.8.2(003900.023*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.494*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00005)
1-0:99.97.0(3)(0-0:96.7.19)(190614125106S)(0000003280*s)(190909000512S)(0000003281*s)(200228185523W)(0000006231*s)
1-0:32.32.0(00007)
1-0:52.32.0(00019)
1-0:72.32.0(00013)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(226.3*V)
1-0:52.7.0(225.7*V)
1-0:72.7.0(227.2*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.015*kW)
1-0:41.7.0(00.039*kW)
1-0:61.7.0(00.440*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383434393439353138)
0-1:24.2.1(200913221006S)(01904.964*m3)
!9E19
[2020-09-13 22:13:02,823] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 60.0s
[2020-09-13 22:14:02,883] DEBUG    [2020-09-13 22:14:02.883482] Read 2048 Byte(s)
[2020-09-13 22:14:02,884] DEBUG    Received telegram:
/ISK5\2M550T-1012

1-3:0.2.8(50)
0-0:1.0.0(200913221258S)
0-0:96.1.1(4530303434303037323531393537363138)
1-0:1.8.1(007318.932*kWh)
1-0:1.8.2(002599.010*kWh)
1-0:2.8.1(001528.442*kWh)
1-0:2.8.2(003900.023*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.500*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00005)
1-0:99.97.0(3)(0-0:96.7.19)(190614125106S)(0000003280*s)(190909000512S)(0000003281*s)(200228185523W)(0000006231*s)
1-0:32.32.0(00007)
1-0:52.32.0(00019)
1-0:72.32.0(00013)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(226.0*V)
1-0:52.7.0(225.9*V)
1-0:72.7.0(226.3*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.014*kW)
1-0:41.7.0(00.039*kW)
1-0:61.7.0(00.436*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383434393439353138)
0-1:24.2.1(200913221006S)(01904.964*m3)
!25F9
[2020-09-13 22:14:02,906] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 60.0s
[2020-09-13 22:15:02,946] DEBUG    [2020-09-13 22:15:02.946790] Read 2048 Byte(s)
[2020-09-13 22:15:02,948] DEBUG    Received telegram:
/ISK5\2M550T-1012
```
In de database zie ik dus 3 metingen staan om 22:13 uur (terwijl datalogger sleep op 60 seconden staat en backend sleep op 30 seconden) en geen meting om 22:14 uur (en zie ook het gat tussen 22:10 en 22:13 uur):
[image]

Ik zie verder over afgelopen 24 uur ook nog 18 CRC errors (maar dat valt niet samen met de ontbrekende/dubbele metingen). Lijkt me niet dramatisch, maar zou misschien ook niet horen:

```
2020-09-14 21:49:16,796] DEBUG    [2020-09-14 21:49:16.796357] Read 1025 Byte(s)
[2020-09-14 21:49:16,797] DEBUG    Received telegram:
/ISK5\2M550T-1012

1-3:0.2.8(50)
0-0:1.0.0(200914214617S)
0-0:96.1.1(4530303434303037323531393537363138)
1-0:1.8.1(007322.846*kWh)
1-0:1.8.2(002601.339*kWh)
1-0:2.8.1(001528.442*kWh)
1-0:2.8.2(003925.771*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.571*kW42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383434393439353138)
0-1:24.2.1(200914214505S)(01905.676*m3)
!CDAE
[2020-09-14 21:49:16,807] WARNING  Rejected telegram (Invalid telegram CRC. The calculated checksum '14176' (3760) does not match the telegram checksum '52654' (CDAE)) (base64 encoded): b'L0lTSzVcMk01NTBULTEwMTINCg0KMS0zOjAuMi44KDUwKQ0KMC0wOjEuMC4wKDIwMDkxNDIxNDYxN1MpDQowLTA6OTYuMS4xKDQ1MzAzMDM0MzQzMDMwMzczMjM1MzEzOTM1MzczNjMxMzgpDQoxLTA6MS44LjEoMDA3MzIyLjg0NiprV2gpDQoxLTA6MS44LjIoMDAyNjAxLjMzOSprV2gpDQoxLTA6Mi44LjEoMDAxNTI4LjQ0MiprV2gpDQoxLTA6Mi44LjIoMDAzOTI1Ljc3MSprV2gpDQowLTA6OTYuMTQuMCgwMDAyKQ0KMS0wOjEuNy4wKDAwLjU3MSprVzQyLjcuMCgwMC4wMDAqa1cpDQoxLTA6NjIuNy4wKDAwLjAwMCprVykNCjAtMToyNC4xLjAoMDAzKQ0KMC0xOjk2LjEuMCg0NzMwMzAzMzM5MzAzMDMxMzgzNDM0MzkzNDM5MzUzMTM4KQ0KMC0xOjI0LjIuMSgyMDA5MTQyMTQ1MDVTKSgwMTkwNS42NzYqbTMpDQohQ0RBRQ=='
[2020-09-14 21:49:16,807] ERROR    Invalid telegram CRC. The calculated checksum '14176' (3760) does not match the telegram checksum '52654' (CDAE)
Traceback (most recent call last):
  File "/dsmr/dsmr_datalogger/services/datalogger.py", line 86, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/dsmr/dsmr_parser/parsers.py", line 39, in parse
    self.validate_checksum(telegram_data)
  File "/dsmr/dsmr_parser/parsers.py", line 79, in validate_checksum
    raise InvalidChecksumError(
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '14176' (3760) does not match the telegram checksum '52654' (CDAE)
[2020-09-14 21:49:16,809] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 60.0s
```

dennissiemensma, 2020-09-15:
Fijn dat je het gevonden hebt en ik zie daarmee direct de oorzaak.

De details zitten in de tijdstippen van de telegrammen:
```
[2020-09-13 22:13:02,807] DEBUG    Received telegram:
...
0-0:1.0.0(200913221257S)
...

<knip>

...
[2020-09-13 22:14:02,884] DEBUG    Received telegram:
...

0-0:1.0.0(200913221258S)
....
```

Er zit een minuut tussen, maar de telegrammen verschillen slechts 1 seconde:
```
200913221257S
200913221258S
```

Blijkbaar wordt de buffer tussendoor niet geleegd en daardoor is er een willekeur in wanneer berichten ingelezen worden.

Ik zal kijken voor een snelle fix.

bikeymouse, 2020-09-15:
Super dat je het gevonden hebt en hartelijk dank weer voor de goede hulp en snelle fix!

dennissiemensma, 2020-09-15:
Mocht je in de gelegenheid zijn om de fix te proberen:

```
sudo su - dsmr
git fetch
git checkout -b wip/v4.3.1 origin/wip/v4.3.1
./deploy.sh
```

Terugdraaien kan met:
```
sudo su - dsmr
git checkout v4
./deploy.sh
```

Als is het met Docker wellicht wat minder makkelijk.

Als alternatief kun je deze change direct doorvoeren: https://github.com/dennissiemensma/dsmr-reader/commit/239e52e39bdf914611dd72ce3579beec8c2e9599#diff-733a170f04e1a8343f7af62d1953e30e

bikeymouse, 2020-09-15:
Ha Dennis,
Dank voor de tip. Ik draai echter alleen met Docker, handmatige installatie lijkt me wat ingewikkeld. 
Ik wacht wel even tot er een update via Docker beschikbaar is ;-)

dennissiemensma, 2020-09-16:
v4.3.1 is net uit. Mocht dat het niet oplossen, dan hoor ik het graag!

closed 2020-09-16 by dennissiemensma

bikeymouse, 2020-09-17:
Ha Dennis,

Dank! 

Ik heb hem gisteravond geïnstalleerd en zie nu inderdaad weer netjes 1 x per minuut een meting in de datalogger-database verschijnen. 

Toch is het nog niet helemaal goed, want er mist ook nog af en toe wat. Bij nadere inspectie zie ik in afgelopen 14 uur wel 20 CRC-errors voorkomen, dus ruim 1 per uur. 
Heb dat nog niet eerder gezien (maar ook niet echt opgelost), maar dat lijkt me wel veel.
Voorbeeld van 2 opvolgende errors:
```
[2020-09-17 12:02:14,139] ERROR    Invalid telegram CRC. The calculated checksum '53343' (D05F) does not match the telegram checksum '14172' (375C)
Traceback (most recent call last):
  File "/dsmr/dsmr_datalogger/services/datalogger.py", line 86, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/dsmr/dsmr_parser/parsers.py", line 39, in parse
    self.validate_checksum(telegram_data)
  File "/dsmr/dsmr_parser/parsers.py", line 79, in validate_checksum
    raise InvalidChecksumError(
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '53343' (D05F) does not match the telegram checksum '14172' (375C)
[2020-09-17 13:05:19,796] ERROR    Invalid telegram CRC. The calculated checksum '42137' (A499) does not match the telegram checksum '31759' (7C0F)
Traceback (most recent call last):
  File "/dsmr/dsmr_datalogger/services/datalogger.py", line 86, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/dsmr/dsmr_parser/parsers.py", line 39, in parse
    self.validate_checksum(telegram_data)
  File "/dsmr/dsmr_parser/parsers.py", line 79, in validate_checksum
    raise InvalidChecksumError(
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '42137' (A499) does not match the telegram checksum '31759' (7C0F)
```

dennissiemensma, 2020-09-17:
Dank voor de terugkoppeling. Voor de CRC fouten zul je denk ik weer even in de debug-mode moeten gaan en wat info delen over hoe de berichten gelezen worden. Als je ze niet structureel elke minuut krijg, kun je wellicht de logging even twee uur aanlaten oid.

dennissiemensma, 2020-09-18:
Ik denk dat ik in v4.4 een optie maak om de datalogger opnieuw te laten verbinden. 

In jouw geval is het namelijk onnodig om de verbinding in stand te houden, waardoor die buffers ook vollopen en de kans op foute CRC's groter is. De persistente verbindingen zijn meer bedoeld voor uitlezen met lage sleep, waarbij de overhead voor verbinden onnodig is.

bikeymouse, 2020-09-19:
Hi thanks, klinkt logisch, ik hoop dat dat helpt.
Ter info, ik heb hem nog even in debug mode laten draaien, en dit is dan een voorbeeld van een CRC-error:
```
[2020-09-19 21:23:17,618] DEBUG    [2020-09-19 21:23:17.618089] Read 265 Byte(s)
[2020-09-19 21:23:17,619] DEBUG    Received telegram:
/ISK5\2M550T-1012

1-3:0.2.8(50)
0-0:1.0.0(200919212221S)
0-0:96.1.1(4530303434303037323531393537363138)
1-0:1.8.1(007344.941*kWh)
1-0:1.8.2(002614.947*kWh)
1-0:2.8.1(001551.991*kWh)
1-0:2.8.2(004028.994*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00
!335D
[2020-09-19 21:23:17,628] WARNING  Rejected telegram (Invalid telegram CRC. The calculated checksum '60520' (EC68) does not match the telegram checksum '13149' (335D)) (base64 encoded): b'L0lTSzVcMk01NTBULTEwMTINCg0KMS0zOjAuMi44KDUwKQ0KMC0wOjEuMC4wKDIwMDkxOTIxMjIyMVMpDQowLTA6OTYuMS4xKDQ1MzAzMDM0MzQzMDMwMzczMjM1MzEzOTM1MzczNjMxMzgpDQoxLTA6MS44LjEoMDA3MzQ0Ljk0MSprV2gpDQoxLTA6MS44LjIoMDAyNjE0Ljk0NyprV2gpDQoxLTA6Mi44LjEoMDAxNTUxLjk5MSprV2gpDQoxLTA6Mi44LjIoMDA0MDI4Ljk5NCprV2gpDQowLTA6OTYuMTQuMCgwMDAxKQ0KMS0wOjEuNy4wKDAwDQohMzM1RA=='
[2020-09-19 21:23:17,628] ERROR    Invalid telegram CRC. The calculated checksum '60520' (EC68) does not match the telegram checksum '13149' (335D)
Traceback (most recent call last):
  File "/dsmr/dsmr_datalogger/services/datalogger.py", line 86, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/dsmr/dsmr_parser/parsers.py", line 39, in parse
    self.validate_checksum(telegram_data)
  File "/dsmr/dsmr_parser/parsers.py", line 79, in validate_checksum
    raise InvalidChecksumError(
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '60520' (EC68) does not match the telegram checksum '13149' (335D)
[2020-09-19 21:23:17,630] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 60.0s
```

Ik zo afgelopen 12 uur iets van 40 CRC fouten. Daarnaast had ik nog 1 volgende fout:
`[2020-09-19 14:18:55,523] ERROR    Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing.`

dennissiemensma, 2020-09-20:
Bedankt voor de info. Dat telegram is inderdaad afgebroken. Voor een volgende release zal ik het nog wat proberen te verbeteren.

FutureCow, 2020-09-22:
Ik heb sinds versie 4 ook overrun errors
dmesg:

> [502247.338960] ttyUSB ttyUSB0: 5 input overrun(s)
[502258.279377] ttyUSB ttyUSB0: 5 input overrun(s)
[502269.385027] ttyUSB ttyUSB0: 5 input overrun(s)
[502280.228740] ttyUSB ttyUSB0: 5 input overrun(s)

Elke 10 seconden komt er een bij, precies de interval dat ik update. Ook met versie 4.3.1 hetzelfde probleem. CRC fouten zal ik even in de gaten houden, sta nu op 8282 sinds de eerste meeting.

dennissiemensma, 2020-09-22:
Bedankt voor je aanvulling. Ik vermoed dat dit in de volgende release beter zal gaan.