dsmrreader/dsmr-reader #1108: Server Error bij wijzigen energiecontract-prijzen zonder einddatum

bikeymouse, 2020-09-06:
_DSMR-reader versie: V4.3
Type RaspberryPi of server: Intel NuC / Proxmox
Standaardinstallatie of Docker: Docker xirixiz/dsmr-reader-docker:latest
DSMR-protocol slimme meter v4/v5: V5_

Als ik bij de instellingen van de contractprijzen de wijziging wil opslaan krijg ik als ik (zoals tot nu toe wel kon) geen einddatum voor het contract invul de volgende exceptie:

```
Server Error

Sorry, something unexpected happened.
Exception:

ValueError: Cannot use None as a query value
Traceback:

  File "/usr/local/lib/python3.8/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/contrib/admin/options.py", line 614, in wrapper
    return self.admin_site.admin_view(view)(*args, **kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/contrib/admin/sites.py", line 233, in inner
    return view(request, *args, **kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/contrib/admin/options.py", line 1656, in change_view
    return self.changeform_view(request, object_id, form_url, extra_context)

  File "/usr/local/lib/python3.8/site-packages/django/utils/decorators.py", line 43, in _wrapper
    return bound_method(*args, **kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/contrib/admin/options.py", line 1534, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)

  File "/usr/local/lib/python3.8/site-packages/django/contrib/admin/options.py", line 1573, in _changeform_view
    form_validated = form.is_valid()

  File "/usr/local/lib/python3.8/site-packages/django/forms/forms.py", line 177, in is_valid
    return self.is_bound and not self.errors

  File "/usr/local/lib/python3.8/site-packages/django/forms/forms.py", line 172, in errors
    self.full_clean()

  File "/usr/local/lib/python3.8/site-packages/django/forms/forms.py", line 375, in full_clean
    self._clean_form()

  File "/usr/local/lib/python3.8/site-packages/django/forms/forms.py", line 402, in _clean_form
    cleaned_data = self.clean()

  File "/dsmr/dsmr_consumption/forms.py", line 18, in clean
    existing = EnergySupplierPrice.objects.exclude(

  File "/usr/local/lib/python3.8/site-packages/django/db/models/query.py", line 942, in filter
    return self._filter_or_exclude(False, *args, **kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/db/models/query.py", line 962, in _filter_or_exclude
    clone._filter_or_exclude_inplace(negate, *args, **kwargs)

  File "/usr/local/lib/python3.8/site-packages/django/db/models/query.py", line 969, in _filter_or_exclude_inplace
    self._query.add_q(Q(*args, **kwargs))

  File "/usr/local/lib/python3.8/site-packages/django/db/models/sql/query.py", line 1358, in add_q
    clause, _ = self._add_q(q_object, self.used_aliases)

  File "/usr/local/lib/python3.8/site-packages/django/db/models/sql/query.py", line 1377, in _add_q
    child_clause, needed_inner = self.build_filter(

  File "/usr/local/lib/python3.8/site-packages/django/db/models/sql/query.py", line 1237, in build_filter
    return self._add_q(

  File "/usr/local/lib/python3.8/site-packages/django/db/models/sql/query.py", line 1377, in _add_q
    child_clause, needed_inner = self.build_filter(

  File "/usr/local/lib/python3.8/site-packages/django/db/models/sql/query.py", line 1237, in build_filter
    return self._add_q(

  File "/usr/local/lib/python3.8/site-packages/django/db/models/sql/query.py", line 1377, in _add_q
    child_clause, needed_inner = self.build_filter(

  File "/usr/local/lib/python3.8/site-packages/django/db/models/sql/query.py", line 1319, in build_filter
    condition = self.build_lookup(lookups, col, value)

  File "/usr/local/lib/python3.8/site-packages/django/db/models/sql/query.py", line 1170, in build_lookup
    raise ValueError("Cannot use None as a query value")

```

dennissiemensma, 2020-09-06:
Bedankt voor je melding. Dit wordt opgelost in v4.4 via #1094.

closed 2020-09-07 by dennissiemensma

dennissiemensma, 2020-09-25:
Zojuist uitgebracht in [v4.4](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.4)