dsmrreader/dsmr-reader #1110: Remote logger koppelen met DB

Blackpander, 2020-09-11:
Ik vroeg me af of je de remote logger op een pi zero w kan draaien en de data dan direct in een database kan wegschrijven zonder dsmr reader instantie?

Ik bekijk namelijk al mijn data via grafana, dus gebruik de dsmr reader niet.

dennissiemensma, 2020-09-12:
De datalogger is puur een doorgeefluik van telegrammen. Als je een parser zoekt kun je wellicht beter dsmr_parser op GitHub gebruiken en zoeken naar een DB-client.

closed 2020-09-13 by dennissiemensma

ilionxsjambor, 2020-10-28:
Voor andere toepassingen (zoals het stabiel houden van je SD kaart op de Pi) zou het ook wel prettig zijn als een "remote" Postgres db gebruikt kan worden bijv in een clouddienst). Is daar iets voor om dat te kunnen regelen? Ben al een dag alles aan het doorspitten