dsmrreader/dsmr-reader #1111: dsmr_webinterface: ERROR (spawn error)

Milithor, 2020-09-15:
Ik heb helaas gisteren een overheat gehad van mijn server, dit is al wat vaker gebeurd maar nu krijg ik een Bad Gateway 205 Error wanneer ik naar de dsmrreader wil gaan.

Versie: 3.11 -> 3.12

Bij de FAQ gekeken heb ik een nieuwe deploy gedaan zodat de versie zeker up to date is:
```
--- You are currently running version:
3.11

 --- Checking Python version.
Running Python 3.6.9 (default, Nov  7 2019, 10:44:02)
[GCC 8.3.0]

 --- Pulling remote repository for new commits...
remote: Enumerating objects: 1052, done.
remote: Counting objects: 100% (1052/1052), done.
remote: Compressing objects: 100% (110/110), done.
remote: Total 2007 (delta 951), reused 1006 (delta 917), pack-reused 955
Receiving objects: 100% (2007/2007), 941.79 KiB | 2.67 MiB/s, done.
Resolving deltas: 100% (1418/1418), completed with 288 local objects.
From https://github.com/dennissiemensma/dsmr-reader
   a1869c3c..0ae73bf4  v3                                              -> origin/v3
   a1869c3c..f4f605aa  development                                     -> origin/development
 * [new branch]        feature/v4.4/1086/status-monitoring-refactoring -> origin/feature/v4.4/1086/status-monitoring-refactoring
   9731f0a9..f4f605aa  v4                                              -> origin/v4
 * [new branch]        wip/v4.3                                        -> origin/wip/v4.3
 * [new branch]        wip/v4.4                                        -> origin/wip/v4.4
 * [new tag]           v3.12                                           -> v3.12
 * [new tag]           v4.3.0                                          -> v4.3.0
 * [new tag]           v4.0.0                                          -> v4.0.0
 * [new tag]           v4.1.0                                          -> v4.1.0
 * [new tag]           v4.1.1                                          -> v4.1.1
 * [new tag]           v4.2.0                                          -> v4.2.0

 --- Merging/updating checkout.
Updating a1869c3c..0ae73bf4
Fast-forward
 .gitignore                                             |   2 ++
 docs/changelog.rst                                     |  18 ++++++++++++++++++
 dsmr_frontend/migrations/0027_v215_release.py          |   2 +-
 dsmr_frontend/migrations/0037_v3_12_0_release.py       |  37 +++++++++++++++++++++++++++++++++++++
 dsmr_frontend/templates/dsmr_frontend/dashboard.html   |  13 +++++++++++++
 dsmr_frontend/templates/dsmr_frontend/live-graphs.html |  13 +++++++++++++
 dsmr_frontend/urls.py                                  |   4 +++-
 dsmr_frontend/views/generic.py                         |   8 +++++++-
 dsmrreader/__init__.py                                 |   2 +-
 dsmrreader/locales/nl/LC_MESSAGES/django.mo            | Bin 100397 -> 101037 bytes
 dsmrreader/locales/nl/LC_MESSAGES/django.po            |  11 ++++++++++-
 dsmrreader/provisioning/downgrade/v3.12.sh             |  15 +++++++++++++++
 dsmrreader/provisioning/requirements/base.txt          |   8 ++++----
 dsmrreader/provisioning/requirements/dev.txt           |   6 +++---
 14 files changed, 127 insertions(+), 12 deletions(-)
 create mode 100644 dsmr_frontend/migrations/0037_v3_12_0_release.py
 create mode 100644 dsmrreader/provisioning/downgrade/v3.12.sh

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.

 --- Checking Python version.
Running Python 3.6.9 (default, Nov  7 2019, 10:44:02)
[GCC 8.3.0]

 --- Checking & synchronizing base requirements for changes.
Requirement already up-to-date: coreapi==2.3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.3.3)
Requirement already up-to-date: crcmod==1.7 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.7)
Collecting django==3.0.8
  Downloading Django-3.0.8-py3-none-any.whl (7.5 MB)
     |████████████████████████████████| 7.5 MB 1.8 MB/s
Requirement already up-to-date: django-admin-rangefilter==0.6.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.6.1)
[... 71 lines cut ...]
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                      [OK]

 --- Clearing cache...

 --- Deployed version:
3.12

 >>> Deployment complete. <<<
```

Daarna heb ik de supervisor alles laten herstarten en zie ik een **Spawn Error** op de webinterface:
```
:~$ sudo supervisorctl restart all
dsmr_backend: stopped
dsmr_mqtt: stopped
dsmr_webinterface: ERROR (spawn error)
dsmr_mqtt: started
dsmr_backend: started
```

Vroegere reboots en restarts hadden dit probleem niet, dus mogelijks ergens een file corrupted?

Bedankt alvast voor je hulp!

Milithor, 2020-09-15:
Logs tonen het volgende:

```
sudo tail -f /var/log/supervisor/dsmr_webinterface*
Error: Already running on PID 1448 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

[2020-09-15 10:31:15 +0200] [4525] [INFO] Starting gunicorn 20.0.4

Error: Already running on PID 1448 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

[2020-09-15 10:31:18 +0200] [4528] [INFO] Starting gunicorn 20.0.4

Error: Already running on PID 1448 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

```

dennissiemensma, 2020-09-15:
Bedankt voor je melding. Vermoedelijk lost dit het op:

```
sudo rm /var/tmp/gunicorn--dsmr_webinterface.pid
sudo supervisorctl restart dsmr_webinterface
```

Milithor, 2020-09-16:
Hi Dennis

Met die commandos is het inderdaad opgelost!
Dankjewel voor de quick help.

Gr.

Sjors

dennissiemensma, 2020-09-16:
Bedankt voor je terugkoppeling. Ik zal het tevens opnemen in de FAQ bij v4.4

dennissiemensma, 2020-09-22:
Toegevoegd aan docs voor v4.4

closed 2020-09-22 by dennissiemensma