dsmrreader/dsmr-reader #1112: DSMR datalogger network connection fails after x time

onsmam, 2020-09-15:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

Wat gebruik je?
- DSMR-reader versie:   4.3
- Type RaspberryPi of server:   RPI 3B+
- Standaardinstallatie of Docker:   Standaard
- DSMR-protocol slimme meter v4/v5:   V5

---

sinds een tijdje gebruik ik [issue 187](https://github.com/dennissiemensma/dsmr-reader/issues/187) als oplossing voor de netwerk connectie naar een ESP. sinds nieuwe versie 4 heb ik geprobeerd van de interne datalogger te gebruiken als connectie, ook om een time-out in te kunnen stellen, mijn pi loopt erg vol door de V5 en elke seconde data (retentie op een week)

ik zie geen fout meldingen in het log, alleen onderaan de config pagina een rood kruis. het lijkt even te werken maar na een bepaalde tijd stop de logging er mee. ik heb het niet getimed hoe lang het duurt. de tijd op het dashbord dat er al een tijd geen info is ontvangen loopt op.

suggestie van een beginner op programmeer gebied :-)
wellicht een check doen of de data nog binnen komt, zo niet datalogger restart doen? deze check elke 5/10 minuten?

dennissiemensma, 2020-09-15:
Bedankt voor je melding. Heb je al geprobeerd om de datalogger sleep te verhogen? Dan krijg je als het goed is minder telegrammen binnen.

Los daarvan zit er ook nog een bugje in de datalogger, #1107, maar ik weet niet of dat hieraan gerelateerd is.

onsmam, 2020-09-15:
de oplossing van issue 187 gaat toch buiten de datalogger om, rechtstreeks in de api?

omdat ik de sleep wilde gebruiken had ik de datalogger aangezet, en 187 uit, echter na een willekeurige tijd stop de datalogger er mee en komt er niets meer binnen, of wordt niet meer verwerkt. ook wordt de datalogger niet gerestart, ondanks dat er een rood kruis onderaan de config pagina staat...

dennissiemensma, 2020-09-15:
Bedankt voor de aanvulling. Dan denk ik dat dat inderdaad het gevolg van #1107 is.

onsmam, 2020-09-15:
Ah oke, zo begreep ik hem niet, ik wacht het af dan! bedankt!

dennissiemensma, 2020-09-15:
Je kunt overigens nog wel kijken of de fix daarvoor bij jou werkt, samen met een sleep:

```
sudo su - dsmr
git fetch
git checkout -b wip/v4.3.1 origin/wip/v4.3.1
./deploy.sh
```

Terugdraaien kan met:
```
sudo su - dsmr
git checkout v4
./deploy.sh
```

onsmam, 2020-09-16:
ik heb hem ge-deployed, hij lijkt het te doen, ik wacht hem even af bedankt zover!

dennissiemensma, 2020-09-16:
Bedankt voor het proberen! Inmiddels is het ook uit via v4.3.1, dus je kunt weer terug naar de v4 branch met eerdergenoemd command.

closed 2020-09-16 by dennissiemensma

onsmam, 2020-09-16:
Hij doet het inderdaad weer!

onsmam, 2020-09-19:
na 2 dagen is hij er vannacht weer mee gestopt, na het switchen tussen USB en network draait hij weer. (om de data logger te forceren).

het heeft even gewerkt, na 40 minuten is hij er weer mee gestopt. no help dus...

dennissiemensma, 2020-09-20:
Bedankt voor de update. Wat zie je in de datalogger-logs of staat er niets bijzonders in?

onsmam, 2020-09-21:
dsmr_datalogger.log, niets anders dan: `Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging`

dsmr_datalogger-stderr---supervisor-PfcLMH.log, eens in de zoveel tijd: 
```
[2020-09-19 08:42:53,487] ERROR    Invalid telegram CRC. The calculated checksum '21145' (5299) does not match the telegram checksum '3574' (0DF6)
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 86, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 39, in parse
    self.validate_checksum(telegram_data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 85, in validate_checksum
    expected_crc_hex
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '21145' (5299) does not match the telegram checksum '3574' (0DF6)
```

dennissiemensma, 2020-09-21:
Draait het proces zelf nog wel of is die (automatisch) herstart? Omdat je aangaf dat die gestopt was. 

Je kunt eventueel voor een uurtje of wat DEBUG-logging inschakelen (als die na 40 minuten kapt): https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging-levels
Dan kun je meekijken met wat die exact doet. Je hoeft in principe alleen het datalogger proces te herstarten na de config-change. Je zou dan heel uitgebreid in de datalogger-log moeten zien wat die doet. En dan ben ik benieuwd na het gedrag na 40 minuten.

onsmam, 2020-09-22:
ik heb hem op DEBUG gezet, en dan krijg ik in het log allerlei telegrammen voorbijkomen, dat lijkt dus goed te gaan.

[image]

dit zie ik op het dashbord, het verbruiks getal veranderd nu wel elke 10 seconden, echter staat boven in de hoek dus 3 minuten geleden. tevens loopt de grafiek ook niet door

dit zit ik op de status pagina:

[image]

[dsmr_datalogger.log](https://github.com/dennissiemensma/dsmr-reader/files/5262001/dsmr_datalogger.log)

dennissiemensma, 2020-09-22:
Bedankt voor de aanvulling. Je meter lijkt sowieso zo'n 45 seconden achter te lopen (`16:00:47` vs `16:01:26`), dat is niet zo heel erg, maar wel goed om te weten omdat daar de sinds-tijd in DSMR-reader op gebaseerd wordt.

```
...
[2020-09-22 16:01:26,282] DEBUG    Received telegram:
...
0-0:1.0.0(200922160047S)
...
```

Verder zie ik dat het tijdsverschil oploopt (`16:01:48` vs `16:05:00`), ondanks dat de buffer geleegd zou moeten zijn:
```
[2020-09-22 16:05:00,093] DEBUG    Received telegram:

0-0:1.0.0(200922160148S)
```

Dus eigenlijk pakt hij na een paar minuten nog een telegram uit de buffer. Dit zal ik fixen in v4.4 door simpelweg de verbinding te verbreken na het uitlezen, zoals het eerder ook werkte.

onsmam, 2020-09-22:
ik zie niet helemaal waar je de 16:00:47 en 16:01:26 vandaan haalt maar ik geloof je haha.

het viel mij inderdaad ook al op dat die tijd vrij hoog was, ik had het idee dat dit kwam door mijn (trage) pi.. dat deze het niet snel kon verwerken.
aangezien hij via een ESP en wifi aangesloten zit zal dit er ook deels mee te maken hebben.

alvast bedankt voor de fix! ik wacht hem af!

dennissiemensma, 2020-09-22:
Je kunt de datum/tijd uit deze regels halen:
```
0-0:1.0.0(200922160047S)
0-0:1.0.0(20 09 22 16 00 47 S)
0-0:1.0.0(2020-09-22 16:00:47 Summer time)
```