dsmrreader/dsmr-reader #1113: Hoge ratio CRC errors

stappel, 2020-09-18:
Wat gebruik je?
- DSMR-reader versie:   4.3
- Type RaspberryPi of server:   Pi3b
- Standaardinstallatie of Docker:   standaard
- DSMR-protocol slimme meter v4/v5:  v5

---
Ik draai nu DSMR reader slechts sinds 1 maand of zo, maar het viel me op dat ik een goede 10% an CRC errors heb. Ook met een andere kabel geen verschil. Ook in issue #1107 wordt er melding van gemaakt.

ik heb een vermoeden dat het komt door de sleep. Als ik in de log file kijk dan zie ik bijvoorbeeld:

> [2020-09-18 11:04:26,877] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
[2020-09-18 11:04:31,883] DEBUG    [2020-09-18 11:04:31.883542] Read 2048 Byte(s)
[2020-09-18 11:04:31,889] DEBUG    Received telegram:
/XMX5LGBBLA4112019524

>1-3:0.2.8(50)
0-0:1.0.0(2009181
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031373030313137383137)
0-1:24.2.1(200918110006S)(02655.052*m3)
!3BF6
[2020-09-18 11:04:31,902] WARNING  Rejected telegram (Invalid telegram CRC. The calculated checksum '55760' (D9D0) does not match the telegram checksum '15350' (3BF6)) (base64 encoded): b'L1hNWDVMR0JCTEE0MTEyMDE5NTI0DQoNCjEtMzowLjIuOCg1MCkNCjAtMDoxLjAuMCgyMDA5MTgxCjEtMDoyMi43LjAoMDAuMDAwKmtXKQ0KMC0xOjI0LjEuMCgwMDMpDQowLTE6OTYuMS4wKDQ3MzAzMDMzMzkzMDMwMzEzNzMwMzAzMTMxMzczODMxMzcpDQowLTE6MjQuMi4xKDIwMDkxODExMDAwNlMpKDAyNjU1LjA1MiptMykNCiEzQkY2'
[2020-09-18 11:04:31,903] ERROR    Invalid telegram CRC. The calculated checksum '55760' (D9D0) does not match the telegram checksum '15350' (3BF6)
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 86, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 39, in parse
    self.validate_checksum(telegram_data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 85, in validate_checksum
    expected_crc_hex
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '55760' (D9D0) does not match the telegram checksum '15350' (3BF6)
[2020-09-18 11:04:31,913] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s

Hij lijkt dus een half datagram op de vangenomdat hij gaat luisteren terwijl een een datagram pakket al aan het versturen is (vanuit de meter). 
Maar misschien heb ik het totaal mis en zit er gewoon een bug in.

dennissiemensma, 2020-09-18:
Bedankt voor je melding. In principe zijn CRC-fouten niet erg, mits er genoeg goede telegrammen doorkomen. 

Ik ben nog even zoekende wat een goede oplossing is. Aan de ene kant is de datalogger nu het minst zwaar, omdat verbindingen persistent zijn. Aan de andere kant komen hier dus bijeffecten van omdat DSMR-v5 meters de lijn elke seconde (onnodig) volpompen met een telegram, waardoor bij enige sleeps er altijd gedeeltelijke telegrammen gelezen worden.

Ik neig naar een CRC-check in de datalogger zelf zodat die net zo lang blijft lezen totdat er een goed telegram binnenkomt. Dat is waarschijnlijk ook de meest nette oplossing, omdat het de voordelen van de performance en sleep heeft.

stappel, 2020-09-19:
Ik had ongeveer hetzelfde idee als feedback gisteren
Je start luisteren naar een pause, dan komt er een halve telegram voorbij. Die vlag je als een CRC fout. Ok, Maar waarom ga je dan weer in pause voor 5 sec? Waaom blijf je niet luisteren naar de eerst volgende? Je doet toch niks met dat fout pakket.

En misschien kan je de exception afvangen en een warning van maken. Nu lijkt het of het een bug is.

dennissiemensma, 2020-09-20:
De reden dat het op deze manier gebeurt is omdat sleeps de meest efficiente manier zijn om zo min mogelijk load te veroorzaken. Het altijd luisteren is een optie, maar relatief zwaar en tevens nutteloos, helemaal wanneer er hogere sleeps ingesteld worden.

CRC's zijn conditioneel, want soms zijn er ook oprecht veel CRC fouten over de lijn, dus ik kan daar geen onderscheid in maken. Ik zal nog wel eea sleutelen aan de datalogger om het wat beter te maken komende release.

dennissiemensma, 2020-09-23:
Release v4.4 brengt straks weer aardig wat verbeteringen, waaronder #1114. Straks eerst maar even kijken of dat het al wat verbetert voor je.

dennissiemensma, 2020-09-28:
@stappel speelt dit nog steeds in v4.4.x? BIj een hogere datalogger sleep zou die telkens opnieuw moeten verbinden, waardoor ik minder CRC-fouten verwacht.

stappel, 2020-09-29:
Sorry ik had niet door dat 4.4 er al was. Blijkbaar werkt de email functie vanuit DSMR niet meer in mijn setup.

Vanmorgen de upgrade uitgevoerd. Tot nu toe geen meldingen over CRC in het log. Al zie ik de counter voor CRC ook niet meer onder stats pagina.

Wel had ik deze error, mogelijk iets met de uigrade. Ter info zeggen we dan maar.

> [2020-09-29 11:03:40,566] ERROR    dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.ProgrammingError: column dsmr_datalogger_dataloggersettings.log_telegrams does not exist
LINE 1: ...r_datalogger_dataloggersettings"."process_sleep", "dsmr_data...
>                                                             ^
>
>The above exception was the direct cause of the following exception:
>
>Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 30, in run
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 74, in telegram_to_reading
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 31, in get_dsmr_connection_parameters
    is_default_dsmr_protocol = datalogger_settings.dsmr_version != DataloggerSettings.DSMR_VERSION_2_3
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1303, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1156, in execute_sql
    cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.ProgrammingError: column dsmr_datalogger_dataloggersettings.log_telegrams does not exist
LINE 1: ...r_datalogger_dataloggersettings"."process_sleep", "dsmr_data...
                                                             ^

`

dennissiemensma, 2020-09-29:
Bedankt voor de update. Kan het zijn dat er een fout is met de migraties? Je kunt dit even handmatig proberen: https://github.com/dennissiemensma/dsmr-reader/issues/1123#issuecomment-699659888

dennissiemensma, 2020-10-02:
@stappel heeft de migraties uitvoeren het opgelost?

stappel, 2020-10-03:
Ik zie geen meldingen meer in het log. Succes zou ik zeggen

closed 2020-10-03 by stappel