dsmrreader/dsmr-reader #1117: Krijg niet meer iedere seconde data door

bart1604, 2020-09-21:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

Wat gebruik je?
- DSMR-reader versie:   4.3.1
- Type RaspberryPi of server:   Intel NUC
- Standaardinstallatie of Docker:   Docker
- DSMR-protocol slimme meter v4/v5:   V4

---

Voorheen kreeg ik iedere seconde door wat het huidige verbruik is. Momenteel is dit om de paar seconden. Is dit ergens in te stellen? Krijg het zo zelf niet gevonden.

dennissiemensma, 2020-09-21:
Bedankt voor je vraag. Als het goed is heb je ergens tijdens het updaten een melding gekregen met daarin vermeld dat de sleep tijd van de datalogger verhoogd is. Dat is tevens de variabele die je in jouw geval in de admin-interface kan verlagen naar bijvoorbeld `0,5` seconden. Als je in de configuratie-search zoekt naar 'sleep' dan zou je hem moeten vinden.

Let wel dat dit voor extra veel dataopslag en -doorstroom zorgt. Als je echter geen flashgeheugen gebruikt, wat hierdoor sneller slijt, maar iets als een HDD of SSD, dan is dat geen issue.

bart1604, 2020-09-21:
Hoi Dennis,

Bedankt voor je snelle reactie, top.
Als je de "Datalogger proces sleep" bedoelt dan staat deze bij mij al op 0,5 seconden.
Toch duurt het ergens tussen de 6-8 seconden voordat ik nieuwe data te zien krijg op het dashboard.

Opzich allemaal niet zo'n probleem, uiteindelijk gaat het erom dat ik history opbouw en dan zijn dag totalen voldoende.
Maar wellicht heb jij een idee welke kan ik moet zoeken om dit weer terug te krijgen bij het oude.

Mvg,
Bart

dennissiemensma, 2020-09-21:
Weet je zeker dat je een v5 meter hebt? In je initiele bericht heb je neergezet dat je v4 hebt, maar wellicht bedoelde je DSMR-reader v4.

Je kunt nog wat verder graven door DEBUG-logging aan te zetten en de datalogger-log vervolgens te bekijken: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging-levels

Die geeft dan heel gedetaileerd aan wat die aan het doen is.

dennissiemensma, 2020-09-21:
Aangezien je Docker gebruikt kan het trouwens zijn dat de logsfiles ergens anders staan, zoals in /tmp.

bart1604, 2020-09-22:
Ik heb een Kaifa MA304C wat volgens mij hetvolgende is: DSMR 4.2.2
Ik heb DSMR loglevel op DEBUG gezet en zal hieronder een stuk daarvan plakken

`
[2020-09-22 09:18:25,676] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Starting infinite command loop...
[2020-09-22 09:18:25,682] INFO     [2020-09-22 09:18:25.682764] Opening serial connection "/dev/ttyUSB0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
[2020-09-22 09:18:27,651] DEBUG    [2020-09-22 09:18:27.651232] Read 61 Byte(s)
[2020-09-22 09:18:27,902] DEBUG    [2020-09-22 09:18:27.902196] Read 219 Byte(s)
[2020-09-22 09:18:28,153] DEBUG    [2020-09-22 09:18:28.153382] Read 186 Byte(s)
[2020-09-22 09:18:28,404] DEBUG    [2020-09-22 09:18:28.404657] Read 109 Byte(s)
[2020-09-22 09:18:28,655] DEBUG    [2020-09-22 09:18:28.655824] Read 153 Byte(s)
[2020-09-22 09:18:28,907] DEBUG    [2020-09-22 09:18:28.907187] Read 30 Byte(s)
[2020-09-22 09:18:28,911] DEBUG    Received telegram:
/KFM5KAIFA-METER^M
^M
1-3:0.2.8(42)^M
0-0:1.0.0(200922091851S)^M
0-0:96.1.1(4530303236303030303132343637303134)^M
1-0:1.8.1(009676.060*kWh)^M
1-0:1.8.2(007908.632*kWh)^M
1-0:2.8.1(001520.300*kWh)^M
1-0:2.8.2(003409.264*kWh)^M
0-0:96.14.0(0002)^M
1-0:1.7.0(00.512*kW)^M
1-0:2.7.0(00.000*kW)^M
0-0:96.7.21(00010)^M
0-0:96.7.9(00004)^M
1-0:99.97.0(2)(0-0:96.7.19)(190505022558S)(0000000425*s)(000101000013W)(2147483647*s)^M
1-0:32.32.0(00000)^M
1-0:52.32.0(00000)^M
1-0:72.32.0(00000)^M
1-0:32.36.0(00000)^M
1-0:52.36.0(00000)^M
1-0:72.36.0(00000)^M
0-0:96.13.1()^M
0-0:96.13.0()^M
1-0:31.7.0(000*A)^M
1-0:51.7.0(001*A)^M
1-0:71.7.0(001*A)^M
1-0:21.7.0(00.000*kW)^M
1-0:22.7.0(00.037*kW)^M
1-0:41.7.0(00.263*kW)^M
1-0:42.7.0(00.000*kW)^M
1-0:61.7.0(00.286*kW)^M
1-0:62.7.0(00.000*kW)^M
!B6EB
[2020-09-22 09:18:29,156] DEBUG    MQTT: Queued message for dsmr/reading/timestamp
[2020-09-22 09:18:29,174] DEBUG    MQTT: Queued message for dsmr/reading/electricity_delivered_1
[2020-09-22 09:18:29,193] DEBUG    MQTT: Queued message for dsmr/reading/electricity_returned_1
[2020-09-22 09:18:29,205] DEBUG    MQTT: Queued message for dsmr/reading/electricity_delivered_2
[2020-09-22 09:18:29,218] DEBUG    MQTT: Queued message for dsmr/reading/electricity_returned_2
[2020-09-22 09:18:29,228] DEBUG    MQTT: Queued message for dsmr/reading/electricity_currently_delivered
[2020-09-22 09:18:29,241] DEBUG    MQTT: Queued message for dsmr/reading/electricity_currently_returned
[2020-09-22 09:18:29,253] DEBUG    MQTT: Queued message for dsmr/reading/phase_currently_delivered_l1
[2020-09-22 09:18:29,263] DEBUG    MQTT: Queued message for dsmr/reading/phase_currently_delivered_l2
[2020-09-22 09:18:29,274] DEBUG    MQTT: Queued message for dsmr/reading/phase_currently_delivered_l3
[2020-09-22 09:18:29,285] DEBUG    MQTT: Queued message for dsmr/reading/extra_device_timestamp
[2020-09-22 09:18:29,297] DEBUG    MQTT: Queued message for dsmr/reading/extra_device_delivered
[2020-09-22 09:18:29,308] DEBUG    MQTT: Queued message for dsmr/reading/phase_currently_returned_l1
[2020-09-22 09:18:29,320] DEBUG    MQTT: Queued message for dsmr/reading/phase_currently_returned_l2
[2020-09-22 09:18:29,331] DEBUG    MQTT: Queued message for dsmr/reading/phase_currently_returned_l3
[2020-09-22 09:18:29,343] DEBUG    MQTT: Queued message for dsmr/reading/id
[2020-09-22 09:18:29,433] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity1
[2020-09-22 09:18:29,444] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity2
[2020-09-22 09:18:29,454] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity1_returned
[2020-09-22 09:18:29,471] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity2_returned
[2020-09-22 09:18:29,491] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity_merged
[2020-09-22 09:18:29,505] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity_returned_merged
[2020-09-22 09:18:29,529] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity1_cost
[2020-09-22 09:18:29,553] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity2_cost
[... 55 lines cut ...]
1-0:1.7.0(00.521*kW)^M
1-0:2.7.0(00.000*kW)^M
0-0:96.7.21(00010)^M
0-0:96.7.9(00004)^M
1-0:99.97.0(2)(0-0:96.7.19)(190505022558S)(0000000425*s)(000101000013W)(2147483647*s)^M
1-0:32.32.0(00000)^M
1-0:52.32.0(00000)^M
1-0:72.32.0(00000)^M
1-0:32.36.0(00000)^M
1-0:52.36.0(00000)^M
1-0:72.36.0(00000)^M
0-0:96.13.1()^M
0-0:96.13.0()^M
1-0:31.7.0(000*A)^M
1-0:51.7.0(001*A)^M
1-0:71.7.0(001*A)^M
1-0:21.7.0(00.000*kW)^M
1-0:22.7.0(00.036*kW)^M
1-0:41.7.0(00.263*kW)^M
1-0:42.7.0(00.000*kW)^M
1-0:61.7.0(00.294*kW)^M
1-0:62.7.0(00.000*kW)^M
!4959

`

dennissiemensma, 2020-09-22:
Bedankt voor je terugkoppeling. Het lijkt erop dat je geen DSMR-v5-meter hebt, gezien de exact 10 seconden tussenpauze + het versienummer in het telegram (`1-3:0.2.8(42)`).

Dit betekent dat je gebonden bent aan je slimme meter, die elke 10 seconde 1 telegram stuurt. DSMR-reader kan daar niets voor je aan veranderen. Het is mij dan ook een raadsel als je dat eerder wel had (met dezelfde meter).

closed 2020-09-22 by dennissiemensma