dsmrreader/dsmr-reader #1118: dsmr mqtt connectie wordt niet hersteld na herstart mqtt server

mchangsp, 2020-09-21:
Wat gebruik je?
- DSMR-reader versie: v4.3.1
- Type RaspberryPi of server:   RPI 4
- Standaardinstallatie of Docker:   Docker
- DSMR-protocol slimme meter v4/v5:   v4

---
Met v3 wekenlang geen problemen gehad.
Met v4 heb ik nu al een aantal dagen dat de mqtt topics voor electriciteisverbuik plotseling niet meer gepublished worden. De web jnterface werkt normaal. In "Status" werkt alles.
echter in de backend log staat:
ERROR Client no longer connected.

het lijkt erop dat DSMR op een gegeven moment geen contact meer kan maken met de mqtt server.

dennissiemensma, 2020-09-22:
Bedankt voor je melding. Zie #1091 en #1109.

closed 2020-09-22 by dennissiemensma

mchangsp, 2020-09-22:
Dank voor het antwoord. Ik had gezocht op "mqtt" en had #1091 en #1109 niet gezien.