dsmrreader/dsmr-reader #1119: geen waarden in InfluxDB

Phanegem, 2020-09-24:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

Wat gebruik je?
- DSMR-reader versie:   4.3
- Type RaspberryPi of server:   Raspberry Pi 3 Model B+
- Standaardinstallatie of Docker:   Standaard
- DSMR-protocol slimme meter v4/v5:   V5

---
Ik heb op een aparte ubuntu server InfluxDB en grafana draaien.

In grafana zie ik wel:
electricity_live
electricity_positions
[electricity_voltage

Dus deze heeft hij wel doorgestuurd, echter bevatten deze geen waarden. Deze worden of niet doorgestuurd, of, niet opgeslagen. Kun je me op de goede weg zetten om dit op te lossen ?

slashw, 2020-09-24:
Ik heb het zo gedaan als op mn screenshot. De instellingen moet je zelf een beetje uitvogelen. 
Probeer maar eens in het sectie "field" mean naar count te veranderen, krijg je gelijk een heel andere meting.

[image]

dennissiemensma, 2020-09-24:
Bedankt voor je vraag en de aanvulling van slashw. Zelf gebruik ik geen InfluxDB, dus als er nog verbeteringen mogelijk zijn hoor ik het graag! Zie #857 voor achtergrondinformatie. Je kunt verder [debug-logging](https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging-levels) inschakelen om te zien wat er precies gebeurt (in dit geval in `dsmr_datalogger` die de trigger doet).

Wat DSMR-reader betreft is het een vrij eenvoudige constructie die [simpelweg je 'mapping' exporteert naar InfluxDB](https://github.com/dennissiemensma/dsmr-reader/issues/857#issuecomment-658435479) met een simpele ondersteuning voor bepaalde datatypes. Voor het daadwerkelijke gebruik kan ik je helaas niet helpen, hooguit als je iets mist. Dan hoor ik het graag.

dennissiemensma, 2020-09-24:
En anders zou je een mention kunnen doen naar de oorspronkelijke aanvrager van de functionaliteit, @wasperen. Ik heb sindsdien geen aanvullingen meer van diegene ontvangen, dus ik ga er vanuit dat je een heel eind kunt komen met de basis zoals die al beschikbaar is.

Phanegem, 2020-09-25:
> 
> 
> Ik heb het zo gedaan als op mn screenshot. De instellingen moet je zelf een beetje uitvogelen.
> Probeer maar eens in het sectie "field" mean naar count te veranderen, krijg je gelijk een heel andere meting.
> 
> [image]

Ploink, zo simpel en zo lang gezocht. Bedankt voor de tip.

closed 2020-09-25 by dennissiemensma