dsmrreader/dsmr-reader #1120: Dag totaal kosten calculatie klopt niet

JoeyPoortman, 2020-09-24:
Super goede dsmr reader!!! Maar ik loop direct tegen een probleem aan.

Ik zit bij Greenchoice, dus kan alleen over hun praten als klant, probleem:

Bij het opgeven van de kosten en teruglever bedragen is er geen optie om eerst het opgewekte getal van het gebruikte getal af te trekken.
Bij Greenchoice werkt het zo dat als we over 24 uur 5kWh gebruiken hoog tarief, en in diezelfde 20kWh terug hebt geleverd blijft er bij hun alleen -15kWh staan die dag. Is er een optie om dat toe te voegen in de code?

dennissiemensma, 2020-09-24:
Bedankt voor je verzoek! Ik begrijp dat het mooi zou zijn om de (be)rekening van je leverancier na te bootsen, maar DSMR-reader heeft dat niet als doel en het is veel te complex om te maken, laat staan onderhouden voor alle varianten
. 
Je bent overigens niet de eerste met een dergelijk verzoek. Een uitgebreidere redenatie staat o.a. in #1098.

closed 2020-09-24 by dennissiemensma

JoeyPoortman, 2020-09-24:
Bedankt voor de reactie, dan zet ik het weer uit en ga ik proberen om het extern te berekenen! De prijs maakt me geen drol uit aangezien we toch altijd op -xxx uitkomen. Ik heb alleen geen zin om hele grote getallen van elkaar af te trekken voor ik weet wat ik ong teruggeleverd heb haha!