dsmrreader/dsmr-reader #1122: services blijven starten

poudenes, 2020-09-25:
Hoi Allen,

Na een poging tot een upgrade van 3 naar 4 wat niet gelukt is ben ik maar schoon begonnen.
Alles is goed gegaan enige wat niet werkt is MQTT... alle staat goed. Maar weet niet of dit te maken heeft met volgende:

```
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 2393, uptime 0:00:04
dsmr_datalogger                  STARTING
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:45
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 2405, uptime 0:00:02
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:47
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 2415, uptime 0:00:01
dsmr_datalogger                  RUNNING   pid 2405, uptime 0:00:03
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:48
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 2415, uptime 0:00:03
dsmr_datalogger                  STARTING
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:50
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 2438, uptime 0:00:01
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:51
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 2448, uptime 0:00:02
dsmr_datalogger                  RUNNING   pid 2438, uptime 0:00:03
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:53
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 2448, uptime 0:00:03
dsmr_datalogger                  STARTING
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:54
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 2471, uptime 0:00:02
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:56
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 2481, uptime 0:00:01
dsmr_datalogger                  RUNNING   pid 2471, uptime 0:00:03
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:57
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 2481, uptime 0:00:03
dsmr_datalogger                  RUNNING   pid 2471, uptime 0:00:05
dsmr_webinterface                RUNNING   pid 2260, uptime 0:01:59
root@dsmrreader:/home/poudenes# sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 2507, uptime 0:00:02
dsmr_webinterface                RUNNING   pid 2260, uptime 0:02:01
```

Services blijven starten. Kan dit rede zijn dat mijn MQTT info niet doorkomt?

dennissiemensma, 2020-09-25:
Bedankt voor je melding. Dit hoort niet te gebeuren en kan inderdaad een logische oorzaak zijn dat MQTT niets doorkrijgt.

Een goed startpunt is de logs te bekijken en, als daar niets in staat, debug-logging tijdelijk aan te zetten: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging-levels

poudenes, 2020-09-25:
Dit zijn de errors die ik krijg

```
==> /var/log/supervisor/dsmr_datalogger.log <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging

==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-wb7K0h.log <==
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 45, in handle
    self._write_pid_file()
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 135, in _write_pid_file
    with open(self._pid_file, 'w') as file_handle:
PermissionError: [Errno 13] Permission denied: '/var/tmp/dsmrreader--dsmr_datalogger.pid'
```

```
==> /var/log/supervisor/dsmr_backend.log <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 45, in handle
    self._write_pid_file()
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 135, in _write_pid_file
    with open(self._pid_file, 'w') as file_handle:
PermissionError: [Errno 13] Permission denied: '/var/tmp/dsmrreader--dsmr_backend.pid'
```

closed 2020-09-25 by poudenes

dennissiemensma, 2020-09-25:
Bedankt voor de update. Zie de twee permission denied meldingen. Je kunt kijken of /var/tmp/ bestaat, de dsmr-user erbij kan en of er wellicht bestaande bestanden in staan die wegens permissies niet overschreven kunnen worden.

poudenes, 2020-09-25:
> Bedankt voor de update. Zie de twee permission denied meldingen. Je kunt kijken of /var/tmp/ bestaat, de dsmr-user erbij kan en of er wellicht bestaande bestanden in staan die wegens permissies niet overschreven kunnen worden.

Had dit onder de logs nog geplaatst:

/var/tmp/dsmrreader--dsmr_datalogger.pid
/var/tmp/dsmrreader--dsmr_backend.pid

hadden beide 1001:1001 als user en group. Heb ze verwijderd en nu gaat alles goed. MQTT is ook weer terug!

Was door mijzelf wonderlijk al op gelost haha