dsmrreader/dsmr-reader #1123: dsmrreader stopt met publishen op MQTT topic, niets in dsmrreader log te zien (dsmrreader v4.4.1)

mchangsp, 2020-09-26:
DSMR-READER
  Version                                                    v4.4.1
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              5.0 s
  Setting: Datalogger process sleep                           0.5 s
  Setting: Retention cleans up after                         None h

SMART METER
  Telegram version (latest reading)                             v42
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                            1017385
                        \_ unprocessed                            0
  Stored: Electricity consumption records                   1016894
  Stored: Gas consumption records                              2836

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_consumption_electricityconsumption             336 MB
  public.dsmr_datalogger_dsmrreading                         254 MB

---
Ik heb dsmrreader v4.4.1 draaien onder Docker met de instructies van https://github.com/xirixiz/dsmr-reader-docker

Na startup (met DSMRREADER_LOGLEVEL= DEBUG) draait alles goed, maar na een half uurtje of zo stopt het updaten op de mqtt topics. Zie de mqtt explorer screenshot in de zip file. Timestamp = 15:48:51Z = 17:48:51 NL tijd.
In de dsmrreader log file (zie ZIP file) zie aan je de debug statements dat het publishen van de data gewoon doorgaat (regel 6673 en verder in de log file).

Het lijkt erop dat dsmrreader niet door heeft dat er iets mis is?
Ik weet dat de mqtt broker goed werkt omdat andere applicaties (Home Assistant, NodeRed, Domoticz) nog steeds gegevens blijven uitwisselen over de broker heen.

Kan ik nog meer debugging / logging aanzetten om te kijken wat er aan de hand is?
[dsmr v441 200926.zip](https://github.com/dennissiemensma/dsmr-reader/files/5287054/dsmr.v441.200926.zip)

Kan iemand kijken waarom het publishen van de dsmrreader messages in de mqtt topic stopt?

mchangsp, 2020-09-26:
In dsmr_mqtt/services/broker.py in de functie run(mqtt_client) zie ik dat de mqtt messages gepubliceerd worden met mqtt_client.publish.
Wanneer de return value van deze functie in een log gezet kan worden, dan komen we er misschien achter wat er aan de hand is.

dennissiemensma, 2020-09-27:
Bedankt voor je melding. De MQTT client is bijgewerkt, je zou kunnen kijken of je hetzelfde issues hebt met de vorige versie van de client:

```
sudo su - dsmr

pip install paho-mqtt==1.5.0

./reload.sh
```

BobPuts, 2020-09-27:
Ik heb hier ook last van. Echter doet bij mij de configuratie pagina het ook niet krijg de volgende melding. Misschien dat dit verband heeft: 
'ProgrammingError: column dsmr_mqtt_mqttbrokersettings.keep_reconnecting does not exist
LINE 1: ...dsmr_mqtt_mqttbrokersettings"."restart_required", `"dsmr_mqtt...`
een herstart lost het ook niet op.

mchangsp, 2020-09-27:
> 
> 
> Bedankt voor je melding. De MQTT client is bijgewerkt, je zou kunnen kijken of je hetzelfde issues hebt met de vorige versie van de client:
> 
> ```
> sudo su - dsmr
> 
> pip install paho-mqtt==1.5.0
> 
> ./reload.sh
> ```

In de Docker container bestaat het sudo commando niet. Ik krijg "bash: sudo: command not found".

Ik heb QoS op 0 gezet, en nu draait het al ongeveer 20 uur OK. I misschien beter om de return code van de mqtt publish ook te  controleren of alles goed is gegaan.

dennissiemensma, 2020-09-27:
@BobPuts je kunt het beste proberen om opnieuw te deployen en te kijken of er foutmeldingen voorbijkomen. En anders met de hand de DB-migraties uitvoeren:
```
sudo su - dsmr
./manage.py migrate
```

dennissiemensma, 2020-09-27:
@mchangsp gebruikte je eerder een andere QoS?

BobPuts, 2020-09-27:
> @BobPuts je kunt het beste proberen om opnieuw te deployen en te kijken of er foutmeldingen voorbijkomen. En anders met de hand de DB-migraties uitvoeren:
> 
> ```
> sudo su - dsmr
> ./manage.py migrate
> ```

Beide werkt niet krijg beide foutmelding zie foutmelding.txt
[Foutmelding.txt](https://github.com/dennissiemensma/dsmr-reader/files/5288771/Foutmelding.txt)

mchangsp, 2020-09-27:
> @mchangsp gebruikte je eerder een andere QoS?

normaal gesproken gebruik ik QoS 0 (standaard waarde).
Omdat ik connect problemen had, probeerde ik QoS 2.
Na de upgrade stond de QoS nog op 2 toen ik die vreemde problemen kreeg.
Nu weer op 0 gezet en het lijkt nu stabiel te zijn.

dennissiemensma, 2020-09-27:
@BobPuts zie #1126 voor een vervolg

closed 2020-09-27 by dennissiemensma

Mixpower, 2020-09-29:
> Bedankt voor je melding. De MQTT client is bijgewerkt, je zou kunnen kijken of je hetzelfde issues hebt met de vorige versie van de client:
> 
> ```
> sudo su - dsmr
> 
> pip install paho-mqtt==1.5.0
> 
> ./reload.sh
> ```

Heb dit geprobeerd maar blijf dezelfde foutmelding krijgen.