dsmrreader/dsmr-reader #1124: dsmr_webinterface start niet meer

Rawh, 2020-09-26:
Troubleshooting: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

Systeem: Pi 2 Model B v1.1
PostgreSQL: draait op een NAS, pi schrijft daar naartoe

```

DSMR-READER
  Version                                                    v4.4.1
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              1.0 s
  Setting: Datalogger process sleep                           2.0 s
  Setting: Retention cleans up after                         None h

SMART METER
  Telegram version (latest reading)                             v50
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                           24809154
                        \_ unprocessed                            7
  Stored: Electricity consumption records                    625955
  Stored: Gas consumption records                            123995

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                        6472 MB
  public.dsmr_consumption_electricityconsumption             248 MB
  public.dsmr_consumption_gasconsumption                      12 MB
  public.dsmr_stats_hourstatistics                          1224 kB
  public.dsmr_weather_temperaturereading                    1056 kB

```
De dsmr_webinterface wil niet meer starten. Dit heb ik wel eens meer en een reboot van de pi lost dit dan meestal op. Helaas gaat dat niet meer.

De status geeft weer.
```
pi@raspbian:/var/log/supervisor $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 486, uptime 0:06:50
dsmr_datalogger                  RUNNING   pid 485, uptime 0:06:50
dsmr_webinterface                FATAL     Exited too quickly (process log may have details)
```

Een herstart via die methode geeft weer
```
pi@raspbian:/var/log/supervisor $ sudo supervisorctl restart dsmr_webinterface
dsmr_webinterface: ERROR (not running)
dsmr_webinterface: ERROR (spawn error)
```

Logs `dsmr_webinterface.log` van blijven dit herhalen
```[2020-09-26 06:17:36 +0200] [470] [INFO] Starting gunicorn 20.0.4

Error: Already running on PID 464 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

[2020-09-26 06:17:39 +0200] [475] [INFO] Starting gunicorn 20.0.4

Error: Already running on PID 464 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)
```

Rawh, 2020-09-26:
Als ik naar de dsmr user sudo en Gunicorn zelf probeer te draaien, met wat andere variabelen, dan krijg ik dit te zien. Hopelijk dat dit je iets meer zegt :-) 

```
(dsmrreader) dsmr@raspbian:~/dsmr-reader $ cd ../.virtualenvs/dsmrreader/bin/
(dsmrreader) dsmr@raspbian:~/.virtualenvs/dsmrreader/bin $ ls
activate  activate.csh  activate.fish  activate_this.py  autopep8  chardetect  django-admin  django-admin.py  easy_install  easy_install-3.7  gunicorn  miniterm.py  miniterm.pyc  pip  pip3  pip3.7  __pycache__  pycodestyle  python  python3  python3.7  python-config  sqlformat  wheel
(dsmrreader) dsmr@raspbian:~/.virtualenvs/dsmrreader/bin $ ./gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--something.socket dsmrreader.wsgi
[2020-09-27 00:47:38 +0200] [1245] [INFO] Starting gunicorn 20.0.4
[2020-09-27 00:47:38 +0200] [1245] [INFO] Listening at: unix:/var/tmp/gunicorn--something.socket (1245)
[2020-09-27 00:47:38 +0200] [1245] [INFO] Using worker: sync
[2020-09-27 00:47:38 +0200] [1248] [INFO] Booting worker with pid: 1248
[2020-09-27 00:47:38 +0200] [1248] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
    worker.init_process()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
    self.load_wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
    return self.load_wsgiapp()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
    mod = importlib.import_module(module)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'dsmrreader'
[2020-09-27 00:47:38 +0200] [1248] [INFO] Worker exiting (pid: 1248)
[2020-09-27 00:47:38 +0200] [1245] [INFO] Shutting down: Master
[2020-09-27 00:47:38 +0200] [1245] [INFO] Reason: Worker failed to boot.
```

dennissiemensma, 2020-09-27:
Bedankt voor je melding. Je kunt het gevolg oplossen door de 'stale' PID file te verwijderen:
```
sudo rm /var/tmp/gunicorn--dsmr_webinterface.pid
```

Echter zou dit alleen voor moeten komen bij een onverwachte crash, zoals stroomuitval, waardoor Gunicorn z'n PID file niet kan verwijderen.

----

Verder krijg je die andere foutmelding omdat je Gunicorn uit een andere locatie draait dan normaal (de root van het project). Daarom kent die `dsmrreader.wsgi` (`dsmrreader/wsgi.py`) niet.

Rawh, 2020-09-27:
Thanks, dat deed  'm oplossen.

Ik heb idd wat vaker last van stroomuitval. Electricien afspraak staat reeds, hopelijk daar wat uitsluitsel uit wat het kan zijn (aardlek schiet, geen groep)

dennissiemensma, 2020-09-27:
Top, fijn om te horen!

closed 2020-09-27 by dennissiemensma