dsmrreader/dsmr-reader #1128: Server error Energy Contracts

remco-77, 2020-09-28:
Bij opvragen van de Energy Contracts krijg ik nu op v4.4.2 een server error. Wellicht was dit ook al bij 4.4.1 of eerdere versies, dat kan ik helaas niet meer herleiden.

### Foutmelding:
```
Server Error
Sorry, something unexpected happened.

Exception:
TypeError: unsupported operand type(s) for +=: 'decimal.Decimal' and 'NoneType'

Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 11, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 98, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 159, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/energy_contracts.py", line 17, in get_context_data
    context_data['energy_contracts'] = dsmr_consumption.services.summarize_energy_contracts()

  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 511, in summarize_energy_contracts
    summary['total_cost'] += summary[summary_field]
```

### Debug info:

```
DSMR-READER
  Version                                                    v4.4.2
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              1.0 s
  Setting: Datalogger process sleep                           0.5 s
  Setting: Retention cleans up after                         8064 h

SMART METER
  Telegram version (latest reading)                             v42
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                            1531090
                        \_ unprocessed                            8
  Stored: Electricity consumption records                    475515
  Stored: Gas consumption records                             14612

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                         398 MB
  public.dsmr_consumption_electricityconsumption             200 MB
  public.dsmr_stats_hourstatistics                          1672 kB
  public.dsmr_consumption_gasconsumption                    1560 kB
  public.dsmr_weather_temperaturereading                    1328 kB

```

dennissiemensma, 2020-09-28:
Bedankt voor je melding. Heb je toevallig een screenshot of lijst van prijzen die je gebruikt?

remco-77, 2020-09-28:
> Bedankt voor je melding. Heb je toevallig een screenshot of lijst van prijzen die je gebruikt?

Zie hieronder. Heeft eerder wel gewerkt trouwens. Dank alvast voor de fix!

```
dsmrreader=# select * from dsmr_consumption_energysupplierprice;
 id |   start    |    end     | electricity_delivered_1_price | electricity_delivered_2_price | gas_price |       description        | electricity_returned_1_price | electricity_returned_2_price | fixed_daily_cost
----+------------+------------+-------------------------------+-------------------------------+-----------+--------------------------+------------------------------+------------------------------+------------------
  1 | 2016-02-01 | 2019-01-31 |                       0.19360 |                       0.20570 |   0.69030 | Greenchoice              |                      0.00000 |                      0.00000 |          0.00000
  2 | 2019-02-01 | 2020-09-30 |                       0.22360 |                       0.25060 |   0.78610 | Greenchoice Wind & Groen |                      0.00000 |                      0.00000 |          0.00000
  3 | 2020-10-01 | 2023-09-30 |                       0.21562 |                       0.23060 |   0.72739 | Vandebron                |                      0.00000 |                      0.00000 |          0.00000

```

dennissiemensma, 2020-09-28:
Bedankt voor je aanvulling. Dan vermoed ik dat er ergens een dag is waarbij geen gasprijs was, want de contracten hebben geen gekke waardes.

Ben je in de gelegenheid om een fix te proberen of draai je Docker?

```
sudo su - dsmr
git fetch
git checkout -b wip/v4.4.3 origin/wip/v4.4.3
./deploy.sh
```

remco-77, 2020-09-28:
Bedankt voor je snelle reactie. Ik heb de fix geprobeerd en hij doet het weer goed!

dennissiemensma, 2020-09-28:
Dank voor het proberen! Je kunt dan weer terugwisselen met:

```
sudo su - dsmr
git checkout v4
./deploy.sh
```

dennissiemensma, 2020-09-28:
Staat nu ook in v4.4.3

closed 2020-09-28 by dennissiemensma