dsmrreader/dsmr-reader #1129: Server Error

Mixpower, 2020-09-28:
Troubleshooting: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

(Heb je DSMR-reader v4.4 of hoger? Zet dan hieronder ook de output van "dsmr_debuginfo") 

```

DSMR-READER
  Version                                                    v4.4.3
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              1.0 s
  Setting: Datalogger process sleep                           0.5 s
  Setting: Retention cleans up after                         None h

SMART METER
  Telegram version (latest reading)                             v42
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                           10105455
                        \_ unprocessed                            5
  Stored: Electricity consumption records                   1687994
  Stored: Gas consumption records                             28135

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                        2864 MB
  public.dsmr_consumption_electricityconsumption             862 MB
  public.dsmr_stats_hourstatistics                          3376 kB
  public.dsmr_consumption_gasconsumption                    2944 kB

```

---

Zodra ik op Configuratie klik krijg ik het volgende te zien.

`Server Error
Sorry, something unexpected happened.
Exception:
ProgrammingError: column dsmr_mqtt_mqttbrokersettings.keep_reconnecting does not exist
LINE 1: ...dsmr_mqtt_mqttbrokersettings"."restart_required", "dsmr_mqtt...
                                                             ^
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/auth/mixins.py", line 52, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 98, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 159, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/configuration.py", line 39, in get_context_data
    mqtt_broker_settings=MQTTBrokerSettings.get_solo(),

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1303, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1156, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
`

dennissiemensma, 2020-09-29:
Bedankt voor je melding. Zie ook #1123. 

- Wil je eens kijken of de databasemigraties wellicht een foutmelding geven? Dat is namelijk in 99% van de gevallen de reden dat er SQL-fouten komen over missende kolommen.
- Als er een fout uitkomt, kun je dit proberen: https://dsmr-reader.readthedocs.io/en/latest/faq.html#how-do-i-fix-errors-such-as-detail-key-id-123-already-exists

Mixpower, 2020-09-29:
Hoe kan ik dit checken?

Ik heb de "./manage.py dsmr_sqlsequencereset" gebruikt maar dat veranderd niets.

dennissiemensma, 2020-09-29:
Je kunt de DB-migraties zo uitvoeren: https://github.com/dennissiemensma/dsmr-reader/issues/1123#issuecomment-699659888

Normaal is het onderdeel van het deployscript, maar dan vallen fouten wat minder op.

Mixpower, 2020-09-29:
Zojuist gedaan en dit is wat eruit komt.

```
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying dsmr_consumption.0017_energy_supplier_price_refactoring...Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 85, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 245, in handle
    fake_initial=fake_initial,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 227, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/migration.py", line 124, in apply
    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/operations/models.py", line 511, in database_forwards
    getattr(new_model._meta, self.option_name, set()),
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/schema.py", line 380, in alter_unique_together
    self._delete_composed_index(model, fields, {'unique': True}, self.sql_delete_unique)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/schema.py", line 419, in _delete_composed_index
    ", ".join(columns),
ValueError: Found wrong number (2) of constraints for dsmr_consumption_energysupplierprice(start, end)
```

dennissiemensma, 2020-09-29:
Dank voor de aanvulling. Helaas heb ik deze speficieke fout niet eerder gezien. Zou je een dump kunnen maken van de energieprijzen en die willen delen?

Dit kan met:
```
sudo su - postgres
pg_dump --data-only -t dsmr_consumption_energysupplierprice dsmrreader
```

Dan zal ik die lokaal inlezen en kijken of ik het kan reproduceren en oplossen.

Mixpower, 2020-09-29:
Uiteraard geen probleem, voor zover bedankt in ieder geval.
Waar kan ik deze dump terugvinden?

dennissiemensma, 2020-09-29:
Het is de output van het command. Eventueel kun je het doorlussen naar een bestand als je dat makkelijker vindt.

Mixpower, 2020-09-29:
Ah duidelijk, sorry ik dacht dat de dump naar een bestand ging. :)

```
--
-- PostgreSQL database dump
--

-- Dumped from database version 11.5 (Raspbian 11.5-1+deb10u1)
-- Dumped by pg_dump version 11.5 (Raspbian 11.5-1+deb10u1)

SET statement_timeout = 0;
SET lock_timeout = 0;
SET idle_in_transaction_session_timeout = 0;
SET client_encoding = 'UTF8';
SET standard_conforming_strings = on;
SELECT pg_catalog.set_config('search_path', '', false);
SET check_function_bodies = false;
SET xmloption = content;
SET client_min_messages = warning;
SET row_security = off;

--
-- Data for Name: dsmr_consumption_energysupplierprice; Type: TABLE DATA; Schema: public; Owner: dsmrreader
--

COPY public.dsmr_consumption_energysupplierprice (id, start, "end", electricity_delivered_1_price, electricity_delivered_2_price, gas_price, description, electricity_returned_1_price, electricity_returned_2_price, fixed_daily_cost) FROM stdin;
2       2018-03-05      2019-02-28      0.19260 0.20676 0.63429 Nuon VastePrijs Stroom & Gas    0.00000 0.00000 0.00000
1       2017-02-07      2018-03-04      0.17557 0.18912 0.63457 Essent Groene Stroom Zeker 1 jaar (Actietarief) 0.00000 0.00000 0.00000
3       2019-03-01      2020-01-01      0.19190 0.20540 0.67950 Greenchoice     0.00000 0.00000 0.00000
4       2020-01-02      2021-01-02      0.20751 0.20751 0.72971 FENOR Standaard Groen 1 jaar vast       0.20751 0.20751 0.00000
\.

--
-- Name: dsmr_consumption_energysupplierprice_id_seq; Type: SEQUENCE SET; Schema: public; Owner: dsmrreader
--

SELECT pg_catalog.setval('public.dsmr_consumption_energysupplierprice_id_seq', 4, true);

--
-- PostgreSQL database dump complete
--
```

dennissiemensma, 2020-09-29:
De data lijkt goed, zou je eens de output van dit laatste command willen delen?

```
sudo su - postgres
psql dsmrreader

\d public.dsmr_consumption_energysupplierprice
```

Mixpower, 2020-09-29:
```
                                               Table "public.dsmr_consumption_energysupplierprice"
            Column             |          Type          | Collation | Nullable |                             Default                              
-------------------------------+------------------------+-----------+----------+------------------------------------------------------------------
 id                            | integer                |           | not null | nextval('dsmr_consumption_energysupplierprice_id_seq'::regclass)
 start                         | date                   |           | not null | 
 end                           | date                   |           |          | 
 electricity_delivered_1_price | numeric(11,5)          |           | not null | 
 electricity_delivered_2_price | numeric(11,5)          |           | not null | 
 gas_price                     | numeric(11,5)          |           | not null | 
 description                   | character varying(255) |           |          | 
 electricity_returned_1_price  | numeric(11,5)          |           | not null | 
 electricity_returned_2_price  | numeric(11,5)          |           | not null | 
 fixed_daily_cost              | numeric(11,5)          |           | not null | 
Indexes:
    "dsmr_consumption_energysupplierprice_pkey" PRIMARY KEY, btree (id)
    "dsmr_consumption_energysupplierprice_start_07426986_uniq" UNIQUE CONSTRAINT, btree (start, "end")
    "dsmr_consumption_energysupplierprice_start_end_07426986_uniq" UNIQUE CONSTRAINT, btree (start, "end")
```

dennissiemensma, 2020-09-30:
Bedankt voor de info, het lijkt inderdaad in je DB zelf te zitten:
```
   "dsmr_consumption_energysupplierprice_start_07426986_uniq" UNIQUE CONSTRAINT, btree (start, "end")
    "dsmr_consumption_energysupplierprice_start_end_07426986_uniq" UNIQUE CONSTRAINT, btree (start, "end")
```

Hier hoort er maar eentje te zijn, dus nu is het even de vraag hoe we er eentje wegkrijgen en welke dan weg moet.

Als ik bij mezelfde DB-migratie terugdraai maakt die deze aan:
```
dsmr_consumption_energysupplierprice_start_end_07426986_uniq
```

Dus dan denk ik dat deze handmatig bij je weg moet:
```
dsmr_consumption_energysupplierprice_start_07426986_uniq
```

----

Wil je dit eens doen:

```
sudo su - dsmr
psql dsmrreader

drop index dsmr_consumption_energysupplierprice_start_07426986_uniq;
```

En daarna in dezelfde psql sessie weer:
```
\d public.dsmr_consumption_energysupplierprice
```

Om te kijken of die echt weg is. Daarna kun je nogmaals proberen om de migraties uit te voeren om te zien of de fout verholpen is:

```
sudo su - dsmr
./manage.py migrate
```

Mixpower, 2020-09-30:
Top dankjewel, ik  neem aan dat je sudo su - postgres bedoeld? :)

Ik krijg de volgende fout als ik dat probeer.
```
dsmrreader=# drop index dsmr_consumption_energysupplierprice_start_07426986_uniq;
ERROR:  cannot drop index dsmr_consumption_energysupplierprice_start_07426986_uniq because constraint dsmr_consumption_energysupplierprice_start_07426986_uniq on table dsmr_consumption_energysupplierprice requires it
HINT:  You can drop constraint dsmr_consumption_energysupplierprice_start_07426986_uniq on table dsmr_consumption_energysupplierprice instead.
```

dennissiemensma, 2020-09-30:
Ah het is ook een constraint en niet alleen een index, probeer deze eens:

```
sudo su - postgres
psql dsmrreader

ALTER TABLE dsmr_consumption_energysupplierprice DROP CONSTRAINT dsmr_consumption_energysupplierprice_start_07426986_uniq;
```

Mixpower, 2020-09-30:
Ah top dat heeft het opgelost, super bedankt hiervoor! :)

dennissiemensma, 2020-09-30:
Jij bedankt voor het melden! 

Werkt de configuratiepagina nu ook weer helemaal?

Mixpower, 2020-09-30:
Graag gedaan, yes werkt weer top!

closed 2020-09-30 by Mixpower

rodegra-rdg, 2020-10-07:
Ik ervaar na de update v4.3.3 hetzelfde, als ik naar de configuratie wil gaan.

Na de migratie krijg ik alleen deze melding:
ValueError: Found wrong number (0) of constraints for dsmr_consumption_energysupplierprice(start, end)

Zo te zien liggen er bij mij geen indexes? hoe krijg ik die erin?

                                               Table "public.dsmr_consumption_energysupplierprice"
            Column             |          Type          | Collation | Nullable |                             Default
-------------------------------+------------------------+-----------+----------+------------------------------------------------------------------
 id                            | integer                |           | not null | nextval('dsmr_consumption_energysupplierprice_id_seq'::regclass)
 start                         | date                   |           | not null |
 end                           | date                   |           |          |
 electricity_delivered_1_price | numeric(11,5)          |           | not null |
 electricity_delivered_2_price | numeric(11,5)          |           | not null |
 gas_price                     | numeric(11,5)          |           | not null |
 description                   | character varying(255) |           |          |
 electricity_returned_1_price  | numeric(11,5)          |           | not null |
 electricity_returned_2_price  | numeric(11,5)          |           | not null |
 fixed_daily_cost              | numeric(11,5)          |           | not null |

dennissiemensma, 2020-10-07:
@rodegra-rdg je kunt dit proberen:

```
sudo su - postgres
psql dsmrreader

create unique index dsmr_consumption_energysupplierprice_start_end_07426986_uniq
	on dsmr_consumption_energysupplierprice (start, "end");

ALTER TABLE dsmr_consumption_energysupplierprice
    ADD CONSTRAINT dsmr_consumption_energysupplierprice_start_07426986_uniq UNIQUE (start, "end");
```

Daarna nogmaals proberen de migraties/deply uit te voeren.

rodegra-rdg, 2020-10-07:
Krijg nu deze na de migratie
ValueError: Found wrong number (2) of constraints for dsmr_consumption_energysupplierprice(start, end)

resultaat van \d public.dsmr_consumption_energysupplierprice geeft nu dit.
Indexes:
    "dsmr_consumption_energysupplierprice_start_07426986_uniq" UNIQUE CONSTRAINT, btree (start, "end")
    "dsmr_consumption_energysupplierprice_start_end_07426986_uniq" UNIQUE, btree (start, "end")

rodegra-rdg, 2020-10-07:
Nu geeft de migratie dit als resultaat.
django.db.utils.ProgrammingError: constraint "dsmr_consumption_energysupplierprice_start_end_07426986_uniq" of relation "dsmr_consumption_energysupplierprice" does not exist

dennissiemensma, 2020-10-07:
Je kunt eentje verwijderen en kijken of het dan lukt:

```
drop index dsmr_consumption_energysupplierprice_start_end_07426986_uniq;
```

dennissiemensma, 2020-10-07:
Ik heb bij eentje de verkeerde naam gezet zie ik en waarschijnlijk was dat net degene die je moest hebben.

Je kunt de andere ook verwijderen en dan kijken of het helpt om daarna slechts 1 aan te maken.

```
sudo su - postgres
psql dsmrreader

ALTER TABLE dsmr_consumption_energysupplierprice DROP CONSTRAINT dsmr_consumption_energysupplierprice_start_07426986_uniq;

ALTER TABLE dsmr_consumption_energysupplierprice ADD CONSTRAINT dsmr_consumption_energysupplierprice_start_end_07426986_uniq UNIQUE (start, "end");
```

rodegra-rdg, 2020-10-08:
Ja, dat was hem inderdaad.
Alles lijkt weer correct te werken.
Alhoewel ik nu wel het gevoel heb dat ik geen indexen meer heb.
Hoe kan ik die weer aanmaken?
In ieder geval bedankt weer voor de hulp.

rodegra-rdg, 2020-10-08:
Zojuist dit nog even bekeken.
Lijkt mij wat weinig?

The following statement lists all indexes of the schema public in the current database:

SELECT
    tablename,
    indexname,
    indexdef
FROM
    pg_indexes
WHERE
    schemaname = 'public'
ORDER BY
    tablename,
    indexname;

                 tablename                 |                      indexname                      |                                                                indexdef                           
                 dsmr_consumption_energysupplierprice      | dsmr_consumption_energysupplierprice_end_1b6d8f8d   | CREATE INDEX dsmr_consumption_energysupplierprice_end_1b6d8f8d ON public.dsmr_consumption_energysupplierprice USING btree ("end")
                 dsmr_consumption_energysupplierprice      | dsmr_consumption_energysupplierprice_start_3f56c4ff | CREATE INDEX dsmr_consumption_energysupplierprice_start_3f56c4ff ON public.dsmr_consumption_energysupplierprice USING btree (start)
                 dsmr_datalogger_meterstatisticschange     | dsmr_datalogger_meterstatisticschange_pkey          | CREATE UNIQUE INDEX dsmr_datalogger_meterstatisticschange_pkey ON public.dsmr_datalogger_meterstatisticschange USING btree (id)
                 dsmr_frontend_sortedgraph                 | dsmr_frontend_sortedgraph_pkey                      | CREATE UNIQUE INDEX dsmr_frontend_sortedgraph_pkey ON public.dsmr_frontend_sortedgraph USING btree (id)
                 dsmr_frontend_sortedgraph                 | dsmr_frontend_sortedgraph_sorting_order_7a883a44    | CREATE INDEX dsmr_frontend_sortedgraph_sorting_order_7a883a44 ON public.dsmr_frontend_sortedgraph USING btree (sorting_order)
                 dsmr_influxdb_influxdbintegrationsettings | dsmr_influxdb_influxdbintegrationsettings_pkey      | CREATE UNIQUE INDEX dsmr_influxdb_influxdbintegrationsettings_pkey ON public.dsmr_influxdb_influxdbintegrationsettings USING btree (id)
                 dsmr_influxdb_influxdbmeasurement         | dsmr_influxdb_influxdbmeasurement_pkey              | CREATE UNIQUE INDEX dsmr_influxdb_influxdbmeasurement_pkey ON public.dsmr_influxdb_influxdbmeasurement USING btree (id)
                (7 rows)

dennissiemensma, 2020-10-08:
Hmm het is erg bijzonder dat ze bij je missen. Want door de databasemigraties zou iedereen op een zelfde versie uiteindelijk dezelfde structuur en indexes moeten hebben.

Dit is de lijst zoals ik hem nu heb:
```
                    tablename                    |                            indexname                            |                                                                             indexdef                                                                             
-------------------------------------------------+-----------------------------------------------------------------+------------------------------------------------------------------------------------------------------------------------------------------------------------------
 auth_group                                      | auth_group_name_72583608_like                                   | CREATE INDEX auth_group_name_72583608_like ON public.auth_group USING btree (name varchar_pattern_ops)
 auth_group                                      | auth_group_name_key                                             | CREATE UNIQUE INDEX auth_group_name_key ON public.auth_group USING btree (name)
 auth_group                                      | auth_group_pkey                                                 | CREATE UNIQUE INDEX auth_group_pkey ON public.auth_group USING btree (id)
 auth_group_permissions                          | auth_group_permissions_0e939a4f                                 | CREATE INDEX auth_group_permissions_0e939a4f ON public.auth_group_permissions USING btree (group_id)
 auth_group_permissions                          | auth_group_permissions_8373b171                                 | CREATE INDEX auth_group_permissions_8373b171 ON public.auth_group_permissions USING btree (permission_id)
 auth_group_permissions                          | auth_group_permissions_group_id_permission_id_key               | CREATE UNIQUE INDEX auth_group_permissions_group_id_permission_id_key ON public.auth_group_permissions USING btree (group_id, permission_id)
 auth_group_permissions                          | auth_group_permissions_pkey                                     | CREATE UNIQUE INDEX auth_group_permissions_pkey ON public.auth_group_permissions USING btree (id)
 auth_permission                                 | auth_permission_417f1b1c                                        | CREATE INDEX auth_permission_417f1b1c ON public.auth_permission USING btree (content_type_id)
 auth_permission                                 | auth_permission_content_type_id_codename_key                    | CREATE UNIQUE INDEX auth_permission_content_type_id_codename_key ON public.auth_permission USING btree (content_type_id, codename)
 auth_permission                                 | auth_permission_pkey                                            | CREATE UNIQUE INDEX auth_permission_pkey ON public.auth_permission USING btree (id)
 auth_user                                       | auth_user_pkey                                                  | CREATE UNIQUE INDEX auth_user_pkey ON public.auth_user USING btree (id)
 auth_user                                       | auth_user_username_5e5b01d_like                                 | CREATE INDEX auth_user_username_5e5b01d_like ON public.auth_user USING btree (username varchar_pattern_ops)
 auth_user                                       | auth_user_username_key                                          | CREATE UNIQUE INDEX auth_user_username_key ON public.auth_user USING btree (username)
 auth_user_groups                                | auth_user_groups_0e939a4f                                       | CREATE INDEX auth_user_groups_0e939a4f ON public.auth_user_groups USING btree (group_id)
 auth_user_groups                                | auth_user_groups_e8701ad4                                       | CREATE INDEX auth_user_groups_e8701ad4 ON public.auth_user_groups USING btree (user_id)
 auth_user_groups                                | auth_user_groups_pkey                                           | CREATE UNIQUE INDEX auth_user_groups_pkey ON public.auth_user_groups USING btree (id)
 auth_user_groups                                | auth_user_groups_user_id_group_id_key                           | CREATE UNIQUE INDEX auth_user_groups_user_id_group_id_key ON public.auth_user_groups USING btree (user_id, group_id)
 auth_user_user_permissions                      | auth_user_user_permissions_8373b171                             | CREATE INDEX auth_user_user_permissions_8373b171 ON public.auth_user_user_permissions USING btree (permission_id)
 auth_user_user_permissions                      | auth_user_user_permissions_e8701ad4                             | CREATE INDEX auth_user_user_permissions_e8701ad4 ON public.auth_user_user_permissions USING btree (user_id)
 auth_user_user_permissions                      | auth_user_user_permissions_pkey                                 | CREATE UNIQUE INDEX auth_user_user_permissions_pkey ON public.auth_user_user_permissions USING btree (id)
 auth_user_user_permissions                      | auth_user_user_permissions_user_id_permission_id_key            | CREATE UNIQUE INDEX auth_user_user_permissions_user_id_permission_id_key ON public.auth_user_user_permissions USING btree (user_id, permission_id)
[... cut ...]
 dsmr_weather_temperaturereading                 | dsmr_weather_temperaturereading_read_at_key                     | CREATE UNIQUE INDEX dsmr_weather_temperaturereading_read_at_key ON public.dsmr_weather_temperaturereading USING btree (read_at)
 dsmr_weather_weathersettings                    | dsmr_weather_weathersettings_pkey                               | CREATE UNIQUE INDEX dsmr_weather_weathersettings_pkey ON public.dsmr_weather_weathersettings USING btree (id)
(101 rows)
```

Ik weet alleen niet zo goed hoe je dit makkelijk kan oplossen. Ik kan dezelfde dump voor je maken met alleen de queries. En die kan je dan proberen uit te voeren.

dennissiemensma, 2020-10-08:
Houd er wel rekening mee dat dit mogelijk je DB om zeep kan helpen, ik weet niet hoe het uitpakt, dus:

# maak eerst een goede backup van je database!

Queries:
```
psql -d dsmrreader -c "SELECT concat(indexdef, ';') FROM pg_indexes WHERE schemaname = 'public' ORDER BY tablename,indexname;" | head -n 9999
```

```
 CREATE INDEX auth_group_name_72583608_like ON public.auth_group USING btree (name varchar_pattern_ops);
 CREATE UNIQUE INDEX auth_group_name_key ON public.auth_group USING btree (name);
 CREATE UNIQUE INDEX auth_group_pkey ON public.auth_group USING btree (id);
 CREATE INDEX auth_group_permissions_0e939a4f ON public.auth_group_permissions USING btree (group_id);
 CREATE INDEX auth_group_permissions_8373b171 ON public.auth_group_permissions USING btree (permission_id);
 CREATE UNIQUE INDEX auth_group_permissions_group_id_permission_id_key ON public.auth_group_permissions USING btree (group_id, permission_id);
 CREATE UNIQUE INDEX auth_group_permissions_pkey ON public.auth_group_permissions USING btree (id);
 CREATE INDEX auth_permission_417f1b1c ON public.auth_permission USING btree (content_type_id);
 CREATE UNIQUE INDEX auth_permission_content_type_id_codename_key ON public.auth_permission USING btree (content_type_id, codename);
 CREATE UNIQUE INDEX auth_permission_pkey ON public.auth_permission USING btree (id);
 CREATE UNIQUE INDEX auth_user_pkey ON public.auth_user USING btree (id);
 CREATE INDEX auth_user_username_5e5b01d_like ON public.auth_user USING btree (username varchar_pattern_ops);
 CREATE UNIQUE INDEX auth_user_username_key ON public.auth_user USING btree (username);
 CREATE INDEX auth_user_groups_0e939a4f ON public.auth_user_groups USING btree (group_id);
 CREATE INDEX auth_user_groups_e8701ad4 ON public.auth_user_groups USING btree (user_id);
 CREATE UNIQUE INDEX auth_user_groups_pkey ON public.auth_user_groups USING btree (id);
 CREATE UNIQUE INDEX auth_user_groups_user_id_group_id_key ON public.auth_user_groups USING btree (user_id, group_id);
 CREATE INDEX auth_user_user_permissions_8373b171 ON public.auth_user_user_permissions USING btree (permission_id);
 CREATE INDEX auth_user_user_permissions_e8701ad4 ON public.auth_user_user_permissions USING btree (user_id);
 CREATE UNIQUE INDEX auth_user_user_permissions_pkey ON public.auth_user_user_permissions USING btree (id);
 CREATE UNIQUE INDEX auth_user_user_permissions_user_id_permission_id_key ON public.auth_user_user_permissions USING btree (user_id, permission_id);
 CREATE INDEX django_admin_log_417f1b1c ON public.django_admin_log USING btree (content_type_id);
 CREATE INDEX django_admin_log_e8701ad4 ON public.django_admin_log USING btree (user_id);
 CREATE UNIQUE INDEX django_admin_log_pkey ON public.django_admin_log USING btree (id);
 CREATE UNIQUE INDEX django_content_type_app_label_39736169_uniq ON public.django_content_type USING btree (app_label, model);
 CREATE UNIQUE INDEX django_content_type_pkey ON public.django_content_type USING btree (id);
 CREATE UNIQUE INDEX django_migrations_pkey ON public.django_migrations USING btree (id);
 CREATE INDEX django_session_de54fa62 ON public.django_session USING btree (expire_date);
 CREATE UNIQUE INDEX django_session_pkey ON public.django_session USING btree (session_key);
 CREATE INDEX django_session_session_key_9eaaeef_like ON public.django_session USING btree (session_key varchar_pattern_ops);
 CREATE UNIQUE INDEX dsmr_api_apisettings_pkey ON public.dsmr_api_apisettings USING btree (id);
 CREATE UNIQUE INDEX dsmr_backend_backendsettings_pkey ON public.dsmr_backend_backendsettings USING btree (id);
 CREATE UNIQUE INDEX dsmr_backend_emailsettings_pkey ON public.dsmr_backend_emailsettings USING btree (id);
 CREATE INDEX dsmr_backend_scheduledcall_module_path_d8a435fb_like ON public.dsmr_backend_scheduledcall USING btree (module_path varchar_pattern_ops);
 CREATE UNIQUE INDEX dsmr_backend_scheduledcall_module_path_key ON public.dsmr_backend_scheduledcall USING btree (module_path);
 CREATE INDEX dsmr_backend_scheduledcall_next_call_349e681c ON public.dsmr_backend_scheduledcall USING btree (next_call);
 CREATE UNIQUE INDEX dsmr_backend_scheduledcall_pkey ON public.dsmr_backend_scheduledcall USING btree (id);
 CREATE INDEX dsmr_backend_scheduledprocess_active_5ef7244b ON public.dsmr_backend_scheduledprocess USING btree (active);
 CREATE INDEX dsmr_backend_scheduledprocess_module_55cb894b_like ON public.dsmr_backend_scheduledprocess USING btree (module varchar_pattern_ops);
 CREATE UNIQUE INDEX dsmr_backend_scheduledprocess_module_key ON public.dsmr_backend_scheduledprocess USING btree (module);
 CREATE UNIQUE INDEX dsmr_backend_scheduledprocess_pkey ON public.dsmr_backend_scheduledprocess USING btree (id);
 CREATE INDEX dsmr_backend_scheduledprocess_planned_234388f1 ON public.dsmr_backend_scheduledprocess USING btree (planned);
 CREATE UNIQUE INDEX dsmr_backup_backupsettings_pkey ON public.dsmr_backup_backupsettings USING btree (id);
 CREATE UNIQUE INDEX dsmr_backup_dropboxsettings_pkey ON public.dsmr_backup_dropboxsettings USING btree (id);
 CREATE UNIQUE INDEX dsmr_backup_emailbackupsettings_pkey ON public.dsmr_backup_emailbackupsettings USING btree (id);
 CREATE UNIQUE INDEX dsmr_consumption_consumptionsettings_pkey ON public.dsmr_consumption_consumptionsettings USING btree (id);
 CREATE INDEX dsmr_consumption_electrici_phase_power_current_l1_3d9fd70d ON public.dsmr_consumption_electricityconsumption USING btree (phase_power_current_l1);
[... cut ...]
 CREATE INDEX dsmr_stats_note_day_83d05681_uniq ON public.dsmr_stats_note USING btree (day);
 CREATE UNIQUE INDEX dsmr_stats_note_pkey ON public.dsmr_stats_note USING btree (id);
 CREATE UNIQUE INDEX dsmr_weather_temperaturereading_pkey ON public.dsmr_weather_temperaturereading USING btree (id);
 CREATE UNIQUE INDEX dsmr_weather_temperaturereading_read_at_key ON public.dsmr_weather_temperaturereading USING btree (read_at);
 CREATE UNIQUE INDEX dsmr_weather_weathersettings_pkey ON public.dsmr_weather_weathersettings USING btree (id);
```

rodegra-rdg, 2020-10-09:
Ik heb een tijdje geleden alles opnieuw geïnstalleerd en een backup terug geplaatst.
Wel had ik gemerkt dat het sinds dien erg traag was.
Wellicht heb ik hier de indexes gemist.

De meeste creates gingen goed, alleen onderstaande niet.
Ik moet zeggen dat het nu echt als een zonnetje draait :)
Hier ben ik al erg blij nee, bedankt !

> dsmrreader=# CREATE INDEX dsmr_backend_scheduledcall_module_path_d8a435fb_like ON public.dsmr_backend_scheduledcall USING btree (module_path varchar_pattern_ops);

ERROR:  relation "public.dsmr_backend_scheduledcall" does not exist

> dsmrreader=# CREATE UNIQUE INDEX dsmr_backend_scheduledcall_module_path_key ON public.dsmr_backend_scheduledcall USING btree (module_path);

ERROR:  relation "public.dsmr_backend_scheduledcall" does not exist

> dsmrreader=# CREATE INDEX dsmr_backend_scheduledcall_next_call_349e681c ON public.dsmr_backend_scheduledcall USING btree (next_call);

ERROR:  relation "public.dsmr_backend_scheduledcall" does not exist

> dsmrreader=# CREATE UNIQUE INDEX dsmr_backend_scheduledcall_pkey ON public.dsmr_backend_scheduledcall USING btree (id);

ERROR:  relation "public.dsmr_backend_scheduledcall" does not exist

dennissiemensma, 2020-10-09:
Oke top, die `dsmr_backend_scheduledcall` is een tabel die ik waarschijnlijk ooit lokaal had ofzo. Als de rest verder goed werkt is het prima!