dsmrreader/dsmr-reader #1131: DSMR Uitlezing (alleen lezen)

poudenes, 2020-09-29:
Hi All,

When I push on "DSMR Uitlezing (alleen lezen)" the webportal is freezing.
I don't see any errors or other problems. After a while the webportal is back again.

dennissiemensma, 2020-09-29:
Bedankt voor je vraag. Zou je de standaardinformatie willen delen? Als het goed is wordt dat getoond bij het maken van issues:

- Bijvoorbeeld: https://github.com/dennissiemensma/dsmr-reader/issues/new?assignees=&labels=&template=help.md&title=

Het eerste wat namelijk in me op komt is dat je database te groot is geworden.

poudenes, 2020-09-30:
Op gegeven moment krijg ik 504 Gateway Time-out

```
(dsmrreader) dsmr@dsmrreader:~/dsmr-reader$ ./manage.py dsmr_debuginfo

<--- COPY OUTPUT AFTER THIS LINE --->

```

DSMR-READER
  Version                                                    v4.4.1
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              1.0 s
  Setting: Datalogger process sleep                           0.5 s
  Setting: Retention cleans up after                         None h

SMART METER
  Telegram version (latest reading)                             v50
  Setting: Telegram parser                                        4

DATA

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_debuginfo.py", line 29, in handle
    self._dump_data_info()
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_debuginfo.py", line 56, in _dump_data_info
    self._pretty_print('Stored: Telegram records total', DsmrReading.objects.count())
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 411, in count
    return self.query.get_count(using=self.db)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 515, in get_count

    number = obj.get_aggregation(using, ['__count'])['__count']
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 500, in get_aggregation
    result = compiler.execute_sql(SINGLE)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1156, in execute_sql
    cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/encodings/utf_8.py", line 15, in decode
    def decode(input, errors='strict'):
KeyboardInterrupt
```

cat dsmr_webinterface.log
```
[2020-09-30 08:34:48 +0200] [2260] [INFO] Handling signal: term
[2020-09-30 08:34:48 +0200] [25290] [INFO] Worker exiting (pid: 25290)
[2020-09-30 08:34:49 +0200] [2260] [INFO] Shutting down: Master
[2020-09-30 08:35:11 +0200] [625] [INFO] Starting gunicorn 20.0.4
[2020-09-30 08:35:11 +0200] [625] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (625)
[2020-09-30 08:35:11 +0200] [625] [INFO] Using worker: sync
[2020-09-30 08:35:11 +0200] [639] [INFO] Booting worker with pid: 639
[2020-09-30 08:44:07 +0200] [625] [CRITICAL] WORKER TIMEOUT (pid:639)
[2020-09-30 08:44:08 +0200] [837] [INFO] Booting worker with pid: 837
[... 49 lines cut ...]
    num = len(clone)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1303, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1154, in execute_sql
    cursor = self.connection.cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 237, in _cursor
    return self._prepare_cursor(self.create_cursor(name))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
[... cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 237, in _cursor
    return self._prepare_cursor(self.create_cursor(name))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 236, in create_cursor
    cursor = self.connection.cursor()
django.db.utils.InterfaceError: connection already closed
```

dennissiemensma, 2020-09-30:
Bedankt voor de informatie. Alles lijkt erop dat je simpelweg teveel data hebt en dat gebeurt vrij snel met DSMR-v5-meters. Helemaal wanneer je een lage sleep zonder retentie hebt.

Wil je dit eens doen en de output van de onderste query delen:

```
sudo su - postgres
psql dsmrreader

SELECT nspname || '.' || relname AS "relation",
    pg_size_pretty(pg_total_relation_size(C.oid)) AS "total_size"
  FROM pg_class C
  LEFT JOIN pg_namespace N ON (N.oid = C.relnamespace)
  WHERE nspname NOT IN ('pg_catalog', 'information_schema')
    AND C.relkind <> 'i'
    AND nspname !~ '^pg_toast'
  ORDER BY pg_total_relation_size(C.oid) DESC
  LIMIT 10;
```

dennissiemensma, 2020-09-30:
En mocht je inderdaad hele grote tabellen hebben, als in niet tientallen, maar honderden MB's (or meer), dan is dit het advies: 
- https://dsmr-reader.readthedocs.io/nl/v4/faq/data_throughput.html

Daarmee wordt de dataopslag en indirect de performance stukken beter.

poudenes, 2020-09-30:
> Bedankt voor de informatie. Alles lijkt erop dat je simpelweg teveel data hebt en dat gebeurt vrij snel met DSMR-v5-meters. Helemaal wanneer je een lage sleep zonder retentie hebt.
>
> Wil je dit eens [...]

Hier de output:

```
                    relation                    | total_size
------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading             | 3159 MB
 public.dsmr_consumption_electricityconsumption | 86 MB
 public.dsmr_consumption_gasconsumption         | 4552 kB
 public.dsmr_datalogger_meterstatistics         | 576 kB
 public.dsmr_stats_hourstatistics               | 480 kB
 public.dsmr_mqtt_message                       | 168 kB
 public.dsmr_backend_scheduledprocess           | 120 kB
 public.dsmr_stats_daystatistics                | 88 kB
 public.django_admin_log                        | 64 kB
 public.auth_user                               | 64 kB
(10 rows)
```

poudenes, 2020-09-30:
Heb de aanpassingen gedaan in configuratie

poudenes, 2020-09-30:
Nadat ik het heb aangepast, heb ik een reboot gedaan want er kwam geen data meer door via MQTT. Na reboot gebeurt er niks... Wel elke seconde nog in de live viewer van de portal zelf. Lijkt mij dat ik in portal ook om de 5 seconde meting moet zien en niet elke seconde?

poudenes, 2020-09-30:
Heb wederom een reboot gedaan en nu loopt alles weer goed.
Kan ik iets doen aan de grote van de database? Een actie uitvoeren om de total size al naar beneden te krijgen?

dennissiemensma, 2020-09-30:
Als je retentie ingesteld hebt, zal het even duren voordat het effect heeft. De beste indicatie is af en toe kijken naar het aantal records per maand, die zouden op den duur rond een laag aantal moeten zijn. Het opschonen begint bij de oudste metingen.

Voor informatie hoe dit in te zien en wat te verwachten, zie: https://github.com/dennissiemensma/dsmr-reader/issues/1071#issuecomment-670690844

closed 2020-10-01 by poudenes

dennissiemensma, 2020-10-01:
@poudenes betekent dit dat het gelukt is? Zie je de data opgeschoond worden en daarmee ook al een verbetering in de performance?

- Als je denkt dat het opschonen klaar/ingelopen is, kun je eventueel dit uitvoeren: https://github.com/dennissiemensma/dsmr-reader/issues/1102#issuecomment-688444772

Daarmee wordt ook een deel van de database-schijfruimte vrijgegeven. Doorgaans alleen nodig wanneer een zeer groot percentage van de gegevens een Postgresql-database verwijderd wordt.

poudenes, 2020-10-02:
Na opschoon actie is het iets gekrompen de dB:

```
                      relation                      | total_size
----------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading                 | 2428 MB
 public.dsmr_consumption_electricityconsumption     | 75 MB
 public.dsmr_consumption_gasconsumption             | 4552 kB
 public.dsmr_stats_hourstatistics                   | 464 kB
 public.dsmr_datalogger_meterstatistics             | 280 kB
 public.dsmr_backend_scheduledprocess               | 120 kB
 public.auth_user                                   | 64 kB
 public.dsmr_stats_daystatistics                    | 64 kB
 public.django_session                              | 64 kB
 public.dsmr_mqtt_message                           | 64 kB
 public.django_admin_log                            | 64 kB
 public.dsmr_consumption_energysupplierprice        | 56 kB
 public.auth_permission                             | 56 kB
 public.dsmr_notification_statusnotificationsetting | 56 kB
 public.django_content_type                         | 40 kB
 public.dsmr_frontend_sortedgraph                   | 40 kB
 public.django_migrations                           | 40 kB
 public.auth_user_groups                            | 32 kB
 public.dsmr_backup_backupsettings                  | 32 kB
 public.auth_group_permissions                      | 32 kB
```

dennissiemensma, 2020-10-02:
Oke goed om te horen!