dsmrreader/dsmr-reader #1132: Dag statistieken dagelijks uitsturen over MQTT

poudenes, 2020-09-29:
Hoi,

Zou het mogelijk zijn dat de dag statistieken uitgestuurd worden op een opgegeven moment van tijd over MQTT?
Deze MQTT kan ik dan uitlezen in HA en deze dan opslaan in de logger/history. 
Dan hoef ik niet naar de webportal van DSMR en kan de data die in HA zit voor allerlei zaken gebruiken (automations)

dennissiemensma, 2020-09-29:
Bedankt voor je vraag. Kan het zijn dat je er overheen gekeken hebt? Als we hetzelfde bedoelen dan zit die functionaliteit er al in onder `(Data source) Day consumption ....` in zowel JSON als split topic.

poudenes, 2020-09-30:
Die heb ik over het hoofd gezien dank je wel!

closed 2020-09-30 by poudenes

poudenes, 2020-09-30:
Toch een klein vraagje. Het bericht dat dagelijks verzonden wordt (naar mijn pushover) daar staat verbruikers statistieken van de dag ervoor
Electriciteitsverbruik, Gasverbruik en Totale Kosten. Ik bedoel eigenlijk deze informatie. Wordt dat ook via MQTT verzonden al? Want dat kom ik nergens tegen.

dennissiemensma, 2020-09-30:
Ah op die manier. Bovenstaande gegevens zijn incrementeel waardoor een bericht, met een eindtotalen van een dag, nog niet beschikbaar is. Ik zal het opnemen in  #685 maar onbekend wanneer dat doorgevoerd wordt.