dsmrreader/dsmr-reader #1133: Energieontracten geeft foutmelding

poudenes, 2020-09-29:
Hi ALl,

Wanneer ik op Energiecontracten klik krijg ik deze foutmelding

```
Server Error

Sorry, something unexpected happened.
Exception:

TypeError: unsupported operand type(s) for +=: 'decimal.Decimal' and 'NoneType'
Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 11, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 98, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 159, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/energy_contracts.py", line 17, in get_context_data
    context_data['energy_contracts'] = dsmr_consumption.services.summarize_energy_contracts()

  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 511, in summarize_energy_contracts
    summary['total_cost'] += summary[summary_field]
```

dennissiemensma, 2020-09-29:
Bedankt voor je vraag. Zou je de standaardinformatie willen delen? Als het goed is wordt dat getoond bij het maken van issues:

- Bijvoorbeeld: https://github.com/dennissiemensma/dsmr-reader/issues/new?assignees=&labels=&template=help.md&title=

Dit is namelijk waarschijnlijk al opgelost, maar ik weet niet welke versie je draait.

MORD0R, 2020-10-04:
Ík had deze melding ook. Inderdaad waarschijnlijk opgelost. Maar ik gebruikte ook een puntje en na de update een comma ik weet niet of dat uit maakt. Voor 00,00

closed 2020-10-04 by dennissiemensma