dsmrreader/dsmr-reader #1134: telegrams worden niet gelezen sind plaatsen nieuwe meterkast

ghodiamont, 2020-09-30:
Troubleshooting: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

(Heb je DSMR-reader v4.4 of hoger? Zet dan hieronder ook de output van "dsmr_debuginfo") 

---
de status geeft incoming telegrams : most recent reading 1 day and 3 hours

telegrams to consumption: there are too many telegrams unprocessed :  2630

dennissiemensma, 2020-09-30:
Bedankt voor je melding. Ik denk dat je eerst DEBUG-logging mag inschakelen om te kijken wat er aan de hand is: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

Vervolgens kun je mogelijk een foutmelding vinden in de datalogger-log of een hint (waarom er niets gebeurt) in de backend-log.

dennissiemensma, 2020-10-03:
@ghodiamont ben je hier nog verder mee gekomen?

ghodiamont, 2020-10-07:
de index bleek corrupt. na reindex was problem verholpen

closed 2020-10-07 by ghodiamont