dsmrreader/dsmr-reader #1135: Hoog en laagtarief verkeerd uit slimme meter

discofris, 2020-10-01:
Sinds paar weken een slimme meter en ik merk op dat Fluvius een verkeerde interpretatie doet van dag- en nachttarief.
Bvb, dit wordt nu doorgestuurd en aanzien als een laagtarief meting, terwijl het een hoogtarief meting zou moeten zijn:

```
Oct. 1, 2020, 1:36 p.m.
Timestamp indicating when the reading was taken

```
```

/FLU5\253769484_A

0-0:96.1.4(50213)
0-0:96.1.1(3153414733313030313230303831)
0-0:1.0.0(201001133602S)
1-0:1.8.1(000058.290*kWh)
1-0:1.8.2(000084.025*kWh)
1-0:2.8.1(000121.349*kWh)
1-0:2.8.2(000040.531*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(01.972*kW)
1-0:2.7.0(00.000*kW)
1-0:32.7.0(233.8*V)
1-0:52.7.0(235.2*V)
1-0:72.7.0(232.9*V)
1-0:31.7.0(007*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313230303130353231)
0-1:24.4.0(1)
0-1:24.2.3(201001133501S)(00016.279*m3)
!7962
The latest telegram succesfully read. Please note that only the last telegram is saved
```

Ik heb al de DSMR version config aangepast van "Netherlands, v4/5" naar Fluvius. Beiden doen het vreemd...

Ik gebruik DSMR versie 4.4.3 in een Docker container.

dennissiemensma, 2020-10-01:
Bedankt voor je vraag. Dit is inderdaad een bekend probleem in DSMR-reader. Daarom is het mogelijk om [zelf de tarieven te hernoemen](https://github.com/dennissiemensma/dsmr-reader/issues/858). Zie ook #750.

Verder is het wel belangrijk om de juiste DSMR-versie te selecteren, want er zit een subtiel verschil tussen Fluvius en NL telegrammen m.b.t het gas-veld. Deze verschilt namelijk:

```
# NL
0-1:24.2.1

# Fluvius
0-1:24.2.3

# Jouw telegram:
0-1:24.2.3(201001133501S)...
```

dennissiemensma, 2020-10-01:
De namen van de tarieven kun je overigens wijzigen als je zoekt in het configuratiescherm. De achterliggende pagina zit in de admin waar je ook alle grafiekkleuren kan instellen.

discofris, 2020-10-01:
Dank je @dennissiemensma 
Volgende keer zoek ik wat beter, promised!
:-)

closed 2020-10-01 by discofris