dsmrreader/dsmr-reader #1136: MQTT via Node-Red

discofris, 2020-10-01:
Ik vroeg mij af hoe je kan verbinden via MQTT in een Node-RED flow?
Alles in DSMR staat correct, ik heb een Datasource geactiveerd en de Broker aangezet dat ie naar m'n host met Node-RED erop moet sturen.
In Node-RED ben ik gewoon om te subscriben op een MQTT topic, om te luisteren totdat iemand iets stuurt lukt mij niet in Node-RED. Enig advies?
Wanneer ik de Broker in DSMR deactiveer, dan zie ik outgoing messages ... en terug de Broker activeer om te sturen naar ergens, zijn de messages weg. Dus DSMR-gewijs werkt alles ....

dennissiemensma, 2020-10-01:
Bedankt voor je vraag. Ik kan je hier helaas niet mee helpen, ook omdat ik er zelf geen ervaring mee heb.

Ik kan je aanraden je issue te delen op een plek waar velen anderen je kunnen helpen, want hier op Github leest slechts een enkeling mee. Wellicht heb je meer succes op het [Tweakers forum](https://gathering.tweakers.net/forum/list_messages/1876137)?

discofris, 2020-10-03:
T'is me gelukt. Met een [HiveMQ ](https://www.hivemq.com/downloads/docker/)container ertussen werkt het perfect. In het admin panel van HiveMQ zie ik nu dus 2 clients: 1tje is DSMR-reader en de andere is Node-RED.

### MQTT config in DSMR-reader:
[image]

### MQTT broker node in Node-RED:
[image]
[image]

 ### HiveMQ heeft geen instellingen nodig. 
Enkel de container deployen, vanuit DSMR-reader en Node-RED er mee verbinden. Zorg ook dat je 2 **verschillende** client-ids gebruikt:
[image]
[image]

[image]

Hopelijk helpt dit nog iemand anders.

dennissiemensma, 2020-10-03:
Fijn om te horen, dank voor je uitgebreide toelichting!

closed 2020-10-03 by dennissiemensma