dsmrreader/dsmr-reader #1137: "[sudo] password for dsmr:" na "sudo supervisorctl restart all"

Raarrrinator, 2020-10-01:
Hoi Sander & gebruikers!

Ik heb zojuist DSMR geupdate via sudo su - dsmr, ./deploy.sh. Nu blijkt de reader al 9 uur geen gegevens te ontvangen. In Home Assistant is de Datalogger opnieuw ingesteld en deze werkt weer.

Wanneer ik via sudo supervisorctl restart all de applicatie opnieuw wil starten, krijg ik de prompt "[sudo] password for dsmr:"

Hier probeer ik het wachtwoord in te vullen waarmee ik in de configuratie van DSMR Reader kom, maar krijg een foutmelding "sorry, try again". Ik heb echter geen ander wachtwoord dan het wachtwoord behorend bij het admin account.

Hoe kan ik DSMR reader weer aan de praat krijgen?

Bedankt!

```

DSMR-READER
  Version                                                    v4.4.3
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              1.0 s
  Setting: Datalogger process sleep                           5.0 s
  Setting: Retention cleans up after                         8064 h

SMART METER
  Telegram version (latest reading)                             v50
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                              29912
                        \_ unprocessed                            5
  Stored: Electricity consumption records                      1445
  Stored: Gas consumption records                                 0

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                        6576 kB

```

dennissiemensma, 2020-10-01:
Bedankt voor je vraag. Als je sudo gebruikt zul je dat moeten doen vanaf een gebruiker met sudo-rechten, zoals `pi` en niet `dsmr`, want die heeft (bewust) geen sudo-rechten.

dennissiemensma, 2020-10-01:
En voor de rest is dit altijd een goed startpunt, als je die zelf nog niet gevonden had: https://dsmr-reader.readthedocs.io/nl/v4/troubleshooting.html

Raarrrinator, 2020-10-02:
Dankjewel! 

Nadat ik met CTRL + D uit ben gelogd als DSMR user werkte het wel!

closed 2020-10-02 by dennissiemensma