dsmrreader/dsmr-reader #1139: Vraag omtrent CPU load

ph0real, 2020-10-02:
Troubleshooting: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

(Heb je DSMR-reader v4.4 of hoger? Zet dan hieronder ook de output van "dsmr_debuginfo") 

---

Hallo Dennis; een vraag over CPU load...

Ik sta gemiddeld op een load van  2.73, 2.46, 2.34 op een raspberry pi die dedicated voor DSMR bezig is. Geen andere activiteiten hiernaast. Op basis van oudere berichten hier, heb ik al e.e.a. proberen te tweaken. Met name het uitzetten van MQTT (niet nodig) en de sleep process instellingen van backend nog wat aangepast (10sec).

Ik heb al even zitten staren naar Top -c; ik zag voorheen bij het mqtt proces een spike naar 100% cpu, maar nu zijn het enkel de overige processen van Python die wat opvragen echter niet exorbitant veel. Vandaar mijn hulpvraag; ik wil kijken of ik de load times wat verder omlaag kan brengen naar normaal niveau. Weet jij nog zaken die hieraan kunnen bijdragen?

Gr.Dennis

dennissiemensma, 2020-10-02:
Bedankt voor je vraag, ik heb wat meer informatie van je nodig. De nummer 1 oorzaak is namelijk een te grote database of indexes die missen.

Als je DSMR-reader 4.4 of hoger hebt, wil je dan dit uitvoeren en hier plakken:
```
sudo su - dsmr
./manage.py dsmr_debuginfo --with-indices
```

dennissiemensma, 2020-10-02:
En los van bovenstaand issue, wil je wellicht dit ook nog checken, er lijkt een mount of map te missen:
```
[2020-10-02 00:43:55,586] ERROR    (OSError) dsmr_backup.services.backup.run errored: [Errno 19] No such device: '/backup/DSMR'
```

ph0real, 2020-10-02:
> En los van bovenstaand issue, wil je wellicht dit ook nog checken, er lijkt een mount of map te missen:
> 
> ```
> [2020-10-02 00:43:55,586] ERROR    (OSError) dsmr_backup.services.backup.run errored: [Errno 19] No such device: '/backup/DSMR'
> ```
Check, deze moet ik inderdaad nog aanpassen.

> Bedankt voor je vraag, ik heb wat meer informatie van je nodig. De nummer 1 oorzaak is namelijk een te grote database of indexes die missen.
> 
> Als je DSMR-reader 4.4 of hoger hebt, wil je dan dit uitvoeren en hier plakken:
> 
> ```
> sudo su - dsmr
> ./manage.py dsmr_debuginfo --with-indices
> ```

Ik zit nog op 3.x momenteel...kan ik daar een soortgelijke check doen? Of zou t helpen om naar 4 te upgraden wellicht?

dennissiemensma, 2020-10-02:
Als v3.x goed voor je werkt hoef je niet te upgraden, dat kan altijd nog een keer. 

Je kunt de checks gewoon handmatig doen. Ik zoek de output van 5 onderstaande queries:

```
sudo su - postgres
psql dsmrreader 

SELECT nspname || '.' || relname AS "relation",
    pg_size_pretty(pg_total_relation_size(C.oid)) AS "total_size"
  FROM pg_class C
  LEFT JOIN pg_namespace N ON (N.oid = C.relnamespace)
  WHERE nspname NOT IN ('pg_catalog', 'information_schema')
    AND C.relkind <> 'i'
    AND nspname !~ '^pg_toast'
  ORDER BY pg_total_relation_size(C.oid) DESC
  LIMIT 10;

SELECT relname, reltuples as approximate_row_count FROM pg_class WHERE relname IN ('dsmr_datalogger_dsmrreading', 'dsmr_consumption_electricityconsumption', 'dsmr_consumption_gasconsumption')

select process_sleep as be_sleep from dsmr_backend_backendsettings;

select process_sleep as dl_sleep from dsmr_datalogger_dataloggersettings;

select data_retention_in_hours from dsmr_datalogger_retentionsettings;
```

ph0real, 2020-10-03:
**query 1:**
------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading             | 200 MB
 public.dsmr_consumption_electricityconsumption | 36 MB
 public.dsmr_consumption_gasconsumption         | 2968 kB
 public.dsmr_stats_hourstatistics               | 1064 kB
 public.dsmr_weather_temperaturereading         | 888 kB
 public.dsmr_datalogger_meterstatistics         | 328 kB
 public.dsmr_backend_scheduledprocess           | 120 kB
 public.dsmr_stats_daystatistics                | 120 kB
 public.dsmr_frontend_notification              | 104 kB
 public.django_session                          | 96 kB
(10 rows)

**Query 2**
                 relname                 | approximate_row_count 
-----------------------------------------+-----------------------
 dsmr_consumption_electricityconsumption |                 27674
 dsmr_consumption_gasconsumption         |                 19340
 dsmr_datalogger_dsmrreading             |                375985

**Query 3**
 be_sleep 
----------
     10.0

**Query 4:**
 dl_sleep 
----------
     10.0

**Query 5:**
 168

dennissiemensma, 2020-10-03:
Dank voor je aanvulling. Het zal in ieder geval niet liggen aan je datagrootte of instellingen, die lijken me prima.

- Wat voor hardware gebruik je, gewoon een Pi?
----
- Je zou nog kunnen kijken of er veel processen 'wachten' op iets. Bijvoorbeeld op de disk. Zie als voorbeeld: https://github.com/dennissiemensma/dsmr-reader/issues/973#issuecomment-617737324

In `top` zie je dan op de derde regel `0.0 wa` omhoog gaan. Vooral als die 20%, 30% of hoger is dan kan dat een indicatie zijn dat iets anders vertraagt.
```
top - 19:40:05 up 35 days, 23:34,  1 user,  load average: 0.02, 0.03, 0.00
Tasks: 134 total,   1 running, 133 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.1 us,  0.2 sy,  0.0 ni, 99.8 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
```
----
- Verder kun je nog checken of de database de goede indexes heeft. Heel soms ben ik iemand tegengekomen waarbij dat om een of andere reden niet zo was, wat vreemd is, omdat alles via automatische script wordt aangemaakt.

Voor indexes bekijken kun je dit doen:
```
sudo su - postgres
psql dsmrreader

\dS dsmr_datalogger_dsmrreading
\dS dsmr_consumption_gasconsumption
\dS dsmr_consumption_electricityconsumption
```

Ter vergelijking welke er moeten zijn:
```
Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_9d090a40" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1_uniq" btree ("timestamp")
```

```
Indexes:
    "dsmr_consumption_gasconsumption_pkey" PRIMARY KEY, btree (id)
    "dsmr_consumption_gasconsumption_read_at_key" UNIQUE CONSTRAINT, btree (read_at)
```

```
Indexes:
    "dsmr_consumption_electricityconsumption_pkey" PRIMARY KEY, btree (id)
    "dsmr_consumption_electricityconsumption_read_at_key" UNIQUE CONSTRAINT, btree (read_at)
    "dsmr_consumption_electrici_phase_power_current_l1_3d9fd70d" btree (phase_power_current_l1)
    "dsmr_consumption_electrici_phase_voltage_l1_72e3c8f8" btree (phase_voltage_l1)
    "dsmr_consumption_electrici_phase_voltage_l2_917726c7" btree (phase_voltage_l2)
    "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" btree (phase_voltage_l3)
    "dsmr_consumption_electricityc_currently_delivered_5bf9be2d_uniq" btree (currently_delivered)
    "dsmr_consumption_electricityco_currently_returned_5f91fed7_uniq" btree (currently_returned)
    "dsmr_consumption_electricityconsumption_2241f8ee" btree (phase_currently_delivered_l2)
    "dsmr_consumption_electricityconsumption_5714c500" btree (phase_currently_delivered_l1)
    "dsmr_consumption_electricityconsumption_e90842a6" btree (phase_currently_delivered_l3)
```

ph0real, 2020-10-03:
Ja, inderdaad een RPI 3b. De indexen zien er zo uit als je aangeeft.
De .wa waarden schieten wel zeer regelmatig omhoog. 36, 49, 29 % etc..

Als ik je tip uit #973 volg en een Gunicorn proces volg, dan komt onderstaande output steeds terug.
Of dit echt de oorzaak is weet ik niet zeker; ben niet zo bekend met dit tracen :)

sudo strace -v -s 512 -f -p 11111
strace: Process 11111 attached
_newselect(8, [5 7], [], [], {tv_sec=4, tv_usec=124605}) = 0 (Timeout)
fchmod(6, 001)                          = 0
accept4(5, 0x7ed929d8, [110], SOCK_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
getppid()                               = 5863
fchmod(6, 000)                          = 0
_newselect(8, [5 7], [], [], {tv_sec=30, tv_usec=0}) = 0 (Timeout)
fchmod(6, 001)                          = 0
accept4(5, 0x7ed929d8, [110], SOCK_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
getppid()                               = 5863
fchmod(6, 000)                          = 0
_newselect(8, [5 7], [], [], {tv_sec=30, tv_usec=0}) = 0 (Timeout)
fchmod(6, 001)                          = 0
accept4(5, 0x7ed929d8, [110], SOCK_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
getppid()                               = 5863
fchmod(6, 000)                          = 0

dennissiemensma, 2020-10-03:
Oke dat is al een goede aanwijzing. Bij het andere issue kwam het uiteindelijk ook door het grote aantal metingen (10 miljoen), dus wellicht is het een andere oorzaak. Wat nog wel kan is dat je SD-kaartje slecht performt, maar dat is voor mij geen exacte wetenschap. 

- Het doen van `strace` weergeeft alleen de system calls, dus het kan helpen als je **heel gericht** wilt weten waarom een proces ergens lang over die, zoals bij dat andere issue. Vaak zie je dan hoelang het duurt voordat de applicatie een resultaat van een grote query terugkrijgt, of andersom, dat de query snel klaar is maar dat de applicatie lang aan het verwerken is.

- Je zou nog kunnen kijken naar `iotop` of je gerichter kan zoeken, maar ik heb daar inhoudelijk verder geen tips of suggesties voor.

ph0real, 2020-10-04:
Het unmounten van de USB disk heeft de load ongeveer gehalveerd. Ik zit nu rond de 1 gemiddeld. 
iotop loopt redelijk vlot door, daar zie ik eigenlijk weinig vertraging. Mogelijk valt er nog wel wat te winnen op disk i/o, maar ik vraag me of dat niet marginaal is. Ik heb de applicatie en postgresql op de SD draaien. De DB staat op een usb disk. Zijn er best practices hiervoor? Ik zou ook nog kunnen overwegen de applicatieserver te scheiden van de webserver; las laatst zoiets in een van je support documenten. Voor nu is het denk ik voldoende qua support. Heel erg bedankt voor het meedenken!

dennissiemensma, 2020-10-04:
Ik denk dat het scheiden van de webserver niet zoveel zin heeft, want die doet bijna niets. Het meest intensieve is de database en opzich hoef je die ook niet te scheiden, maar als die op betere I/O draait (vooral als het geen SD-kaart is oid) dan win je vermoedelijk het meeste.

Zorg er iig voor dat je af en toe een backup hebt buiten de SD-kaart, want als je SD corrupt raakt ben je alles kwijt. Dat is de voornaamste beste practice denk ik.

closed 2020-10-04 by dennissiemensma