dsmrreader/dsmr-reader #1140: Django settings instellen via envvars

dennissiemensma, 2020-10-03:
N.a.v. #1080 is er regelmatig vraag naar het override van Django settings via envvars. 

Initieel wilde ik slechts een paar mappen, maar ik kan wellicht beter iets maken wat generiek is (met een prefix) om willekeurige settings te overschrijven.

dennissiemensma, 2020-10-03:
Uitdaging is nog wel de type mapping. Dan moet ik even goed testen.

dennissiemensma, 2020-10-03:
Deze voeg ik toe:

```
DJANGO_STATIC_URL=
DJANGO_FORCE_SCRIPT_NAME=
DJANGO_USE_X_FORWARDED_HOST=
DJANGO_USE_X_FORWARDED_PORT=
DJANGO_X_FRAME_OPTIONS=
```

closed 2020-10-03 by dennissiemensma

dennissiemensma, 2020-10-19:
Zojuist uitgebracht in [v4.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.5.0)

discofris, 2020-10-20:
Ik bevestig dat dit netjes werkt :) 🥇

dennissiemensma, 2020-10-20:
Dank voor de terugkoppeling!

Wieter, 2020-11-15:
werkt ook hier prima, thanks!