dsmrreader/dsmr-reader #1145: Samenvoegen data

cmderks93, 2020-10-05:
Is er een mogelijkheid om een oude backup samen te voegen op een reeds draaiende reader?

Heb een backup gemaakt naar dropbox, maar dit zijn meerdere bestanden.

dennissiemensma, 2020-10-05:
Bedankt voor je vraag. Als het goed is zijn alle backups incrementeel, de meeste recente bevat doorgaans de meeste gegevens, dus je zou dan alleen die nodig moeten hebben. Krijg je toevallig een foutmelding bij terugzetten?

cmderks93, 2020-10-05:
Ik heb het terugzetten niet uitgevoerd, omdat er een duidelijke waarschuwing staat:

_To be clear: Do not restore your database if you’ve either started the application and/or ran manage.py migrate in some way.

Doing so WILL cause trouble with duplicate data/ID’s and break your installation at some point._

Ik heb een nieuwe installatie reeds draaien, en zou de data uit de oude installatie (wel zelfde versie) graag samenvoegen.

dennissiemensma, 2020-10-05:
Dat wordt achteraf heel lastig vanwege de reden die er staat. Wat je hooguit kunt doen is zijn je dagtotalen terugzetten. Heb je daar nog een losse backup van?

Die zouden moeten staan in Dropbox in een submap waar je andere backups staan, zijn heel klein en heten bijvoorbeeld `dsmrreader-postgresql-partial-backup-2020-10-05.sql.gz`.

Als je daar de laatste van hebt mag je die wel mailen naar github at dennissiemensma punt nl en dan zal ik de ID eruit halen en terugsturen zodat je die kan importeren.

dennissiemensma, 2020-10-05:
Als je trouwens zelf enige SQL-kennis hebt, kun je het ook zelf proberen. Het is niet heel moeilijk, maar je moet weten wat je doet.

cmderks93, 2020-10-05:
Dat gaat wel lukken. Gewoon het id nummer weghalen, en hierna importeren via het stukje uit de handleiding?

dennissiemensma, 2020-10-05:
Ja in de basis. Je zult meerdere dingen moeten aanpassen:

- `id` weghalen uit velden bij insert
- ID'S weghalen bij elke data regel
- `setval` weghalen

```
# id weghalen
COPY public.dsmr_stats_daystatistics (id, day, total_cost, electricity1, electricity2, electricity1_returned, electricity2_returned, electricity1_cost, electricity2_cost, gas, gas_cost, average_temperature, highest_temperature, lowest_temperature, fixed_cost) FROM stdin;
# Alle eerste velden weghalen, de ID's
415	2016-05-07	0.53	2.344	0.000	0.000	0.000	0.42	0.00	0.183	0.11	18.5	25.7	10.2	0.00
416	2016-05-08	1.31	6.804	0.000	0.000	0.000	1.21	0.00	0.162	0.10	18.1	24.4	9.3	0.00

...

# Deze weghalen
SELECT pg_catalog.setval('public.dsmr_stats_daystatistics_id_seq', 2009, true);

...

cmderks93, 2020-10-05:
Thnx, zal er eens mee stoeien en laten weten of het gelukt is!

dennissiemensma, 2020-10-06:
Mocht je er alsnog niet uitkomen, laat maar weten.

cmderks93, 2020-10-06:
Helaas,  niet gelukt.. Ik krijg: psql:/home/dsmr/backup-dsmr.sql:48: ERROR:  extra data after last expected column
CONTEXT:  COPY dsmr_stats_daystatistics, line 1: "	2020-09-11	-7.80	0.233	0.285	0.000	0.000	0.05	0.07	0.011	0.01	0.0	0.0	0.0	-7.93" deze fout

dennissiemensma, 2020-10-06:
Kloppen het aantal velden wel als je deze telt bij jezelf? En vergelijke met het aantal velden aan data?

```
COPY public.dsmr_stats_daystatistics (day, total_cost, electricity1, electricity2, electricity1_returned, electricity2_returned, electricity1_cost, electricity2_cost, gas, gas_cost, average_temperature, highest_temperature, lowest_temperature, fixed_cost) FROM stdin;
```

cmderks93, 2020-10-06:
tja... als je de id wel weghaalt, maar er een lege waarde laat staan.....
ook de lege waarde dus weggehaald, en nu is het net alsof het er voor gemaakt is. 

Opgelost!
Bedankt!

closed 2020-10-06 by cmderks93