dsmrreader/dsmr-reader #1147: move dsmr docker to other host, influxdb integration not working

dgreebe, 2020-10-08:
I use the latest DSMR in docker and after moving the instance to another OS I cannot enable the influx integration anymore.
I can select the checkbox. save configuration but after 2 seconds the checkbox is unchecked.

Verified the old configuration and settings with the new configuration but its exactly the same.

Does anybody else has this also ? I have no idea how to debug this kind of stuff....

dennissiemensma, 2020-10-08:
Thanks for reporting, please try this first: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#log-files

dgreebe, 2020-10-09:
Its fixed, the problem was somehow authentication eventhough I tested it several times. I saw that that was the issue by changing to DEBUG logleven...

closed 2020-10-09 by dgreebe