dsmrreader/dsmr-reader #1148: Mogelijkheid voor negatieve waarde in Fixed daily costs (€):

jorkzijlstra, 2020-10-08:
Ik zou graag een negatieve waarde in de dagelijkse kosten willen invoeren waardoor ik beter inzicht hebben van de daadwerkelijke kosten per dag. Al moet ik nu alleen maar even wat hoofdrekenen. 

Reden: Vaste leveringskosten (per dag) + netbeheerkosten (per dag) - vermindering energiebelasting (per dag) = een negatieve waarde.

[image]

Een andere optie is misschien ook het veld "vermindering energie belasting" toe te voegen net als vast kosten, maar dat zal waarschijnlijk meer werk zijn.

dennissiemensma, 2020-10-08:
Bedankt voor je verzoek. Dat lijkt me een prima voorstel zodat je daar zelf vrij in bent. 

Ik heb het voor je doorgevoerd in de volgende release.

closed 2020-10-08 by dennissiemensma

dennissiemensma, 2020-10-19:
Zojuist uitgebracht in [v4.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.5.0)