dsmrreader/dsmr-reader #1149: Plotseling geen weginterface 502 Gateway error

poudenes, 2020-10-13:
Ik kan niet achterhalen waar het vandaan komt. Maar ineens is soms de webinterface weg.
Wat ik dan doen is het volgende:

rm /var/tmp/dsmrreader--dsmr_*
sudo supervisorctl restart all

Dan is alles weer in de lucht incl de webiniterface

Gek genoeg doe ik niks met het systeem. Ik update het niet, althans als ik het doe dan reboot ik hele systeem en check ik alles. Dan werkt alles gewoon. Maar na een tijdje valt het soms weg. Dit keer was het de webinterface

willemdej, 2020-10-13:
dit al geprobeerd? sudo supervisorctl tail -f dsmr_webinterface

mijn probleem was namelijk dat de SD kaart vol was. daarom wou de webinterface niet meer starten.

poudenes, 2020-10-13:
Zal ik volgende keer eens proberen. /boot is 22% vol en / is 49% vol (ik gebruik een 32GB Msata SSD)

dennissiemensma, 2020-10-13:
Bedankt voor je melding. Ik denk dat je wat dieper in de log moet zoeken of er een foutmelding gelogd wordt. Bijvoorbeeld met:

```
sudo tail -n 1000 /var/log/supervisor/dsmr_webinterface.log
```

Zie je nog andere foutmeldingen staan? Die `Handling signal: term` komt waarschijnlijk door de `supervisorctl restart all`.

poudenes, 2020-10-14:
@dennissiemensma hier een deel log waar regelmatig `[CRITICAL] WORKER TIMEOUT (pid:604)` te zien is.
Verder de log van 01-10-2020 tot laatste regel

```
[2020-10-01 21:27:25 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:604)
[2020-10-01 21:27:40 +0200] [18867] [INFO] Booting worker with pid: 18867
[2020-10-01 21:28:40 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:18867)
[2020-10-01 21:28:40 +0200] [18867] [INFO] Worker exiting (pid: 18867)
[2020-10-01 21:28:41 +0200] [18879] [INFO] Booting worker with pid: 18879
[2020-10-01 21:29:41 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:18879)
[2020-10-01 21:29:43 +0200] [18882] [INFO] Booting worker with pid: 18882
[2020-10-01 21:31:40 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:18882)
[2020-10-01 21:31:41 +0200] [18927] [INFO] Booting worker with pid: 18927
[2020-10-01 21:32:41 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:18927)
[2020-10-01 21:32:41 +0200] [18927] [INFO] Worker exiting (pid: 18927)
[2020-10-01 21:32:42 +0200] [18931] [INFO] Booting worker with pid: 18931
[2020-10-01 21:35:22 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:18931)
[2020-10-01 21:35:26 +0200] [18986] [INFO] Booting worker with pid: 18986
[2020-10-01 21:36:26 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:18986)
[2020-10-01 21:36:28 +0200] [18998] [INFO] Booting worker with pid: 18998
[2020-10-01 21:38:59 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:18998)
[2020-10-01 21:39:00 +0200] [19016] [INFO] Booting worker with pid: 19016
[2020-10-01 21:40:00 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19016)
[2020-10-01 21:40:01 +0200] [19016] [INFO] Worker exiting (pid: 19016)
[2020-10-01 21:40:01 +0200] [19027] [INFO] Booting worker with pid: 19027
[2020-10-01 21:41:01 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19027)
[2020-10-01 21:41:02 +0200] [19027] [INFO] Worker exiting (pid: 19027)
[2020-10-01 21:41:02 +0200] [19060] [INFO] Booting worker with pid: 19060
[2020-10-01 21:42:02 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19060)
[2020-10-01 21:42:02 +0200] [19060] [INFO] Worker exiting (pid: 19060)
[2020-10-01 21:42:03 +0200] [19072] [INFO] Booting worker with pid: 19072
[2020-10-01 21:43:03 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19072)
[2020-10-01 21:43:03 +0200] [19072] [INFO] Worker exiting (pid: 19072)
[2020-10-01 21:43:05 +0200] [19077] [INFO] Booting worker with pid: 19077
[2020-10-01 21:44:33 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19077)
[2020-10-01 21:44:34 +0200] [19090] [INFO] Booting worker with pid: 19090
[2020-10-01 21:45:34 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19090)
[2020-10-01 21:45:35 +0200] [19090] [INFO] Worker exiting (pid: 19090)
[2020-10-01 21:45:35 +0200] [19136] [INFO] Booting worker with pid: 19136
[2020-10-01 21:46:35 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19136)
[2020-10-01 21:46:35 +0200] [19136] [INFO] Worker exiting (pid: 19136)
[2020-10-01 21:46:35 +0200] [19140] [INFO] Booting worker with pid: 19140
[2020-10-01 21:48:09 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19140)
[2020-10-01 21:48:09 +0200] [19140] [INFO] Worker exiting (pid: 19140)
[2020-10-01 21:48:10 +0200] [19154] [INFO] Booting worker with pid: 19154
[2020-10-01 21:49:10 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19154)
[2020-10-01 21:49:12 +0200] [19157] [INFO] Booting worker with pid: 19157
[2020-10-01 21:51:28 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19157)
[2020-10-01 21:51:29 +0200] [19203] [INFO] Booting worker with pid: 19203
[2020-10-01 21:52:30 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19203)
[2020-10-01 21:52:31 +0200] [19215] [INFO] Booting worker with pid: 19215
[2020-10-01 21:54:38 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19215)
[2020-10-01 21:54:45 +0200] [19223] [INFO] Booting worker with pid: 19223
[2020-10-01 21:55:45 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19223)
[2020-10-01 21:55:45 +0200] [19223] [INFO] Worker exiting (pid: 19223)
[2020-10-01 21:55:46 +0200] [19275] [INFO] Booting worker with pid: 19275
[2020-10-01 21:56:46 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19275)
[2020-10-01 21:56:46 +0200] [19275] [INFO] Worker exiting (pid: 19275)
[2020-10-01 21:56:46 +0200] [19279] [INFO] Booting worker with pid: 19279
[2020-10-01 21:58:44 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19279)
[2020-10-01 21:58:45 +0200] [19293] [INFO] Booting worker with pid: 19293
[2020-10-01 21:59:46 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19293)
[2020-10-01 21:59:46 +0200] [19293] [INFO] Worker exiting (pid: 19293)
[2020-10-01 21:59:46 +0200] [19305] [INFO] Booting worker with pid: 19305
[2020-10-01 22:00:46 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19305)
[2020-10-01 22:00:46 +0200] [19305] [INFO] Worker exiting (pid: 19305)
[2020-10-01 22:00:47 +0200] [19338] [INFO] Booting worker with pid: 19338
[2020-10-01 22:01:47 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19338)
[2020-10-01 22:01:48 +0200] [19341] [INFO] Booting worker with pid: 19341
[2020-10-01 22:02:48 +0200] [590] [CRITICAL] WORKER TIMEOUT (pid:19341)
[... 55 lines cut ...]

[2020-10-13 12:59:19 +0200] [605] [INFO] Starting gunicorn 20.0.4

Error: Already running on PID 592 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

[2020-10-13 12:59:22 +0200] [614] [INFO] Starting gunicorn 20.0.4

Error: Already running on PID 592 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

[2020-10-13 12:59:27 +0200] [617] [INFO] Starting gunicorn 20.0.4

Error: Already running on PID 592 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

[... cut ...]
[2020-10-13 13:03:49 +0200] [791] [INFO] Handling signal: term
[2020-10-13 13:03:49 +0200] [794] [INFO] Worker exiting (pid: 794)
[2020-10-13 13:03:49 +0200] [791] [INFO] Shutting down: Master
[2020-10-13 13:03:55 +0200] [815] [INFO] Starting gunicorn 20.0.4
[2020-10-13 13:03:55 +0200] [815] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (815)
[2020-10-13 13:03:55 +0200] [815] [INFO] Using worker: sync
[2020-10-13 13:03:55 +0200] [820] [INFO] Booting worker with pid: 820
```

dennissiemensma, 2020-10-14:
Er spelen hier twee dingen, dat die PID file stale is komt vermoedelijk doordat de webinterface crasht en die file dus niet zelf kan opruimen.

Als de webinterface timeouts geeft, betekent dat meestal dat die ergens te lang mee bezig is. Je zult denk ik eerst een info-dump moeten maken van wat je draait en hoeveel data je opslaat etc. Dat kan met:

```
sudo su - dsmr
./manage.py dsmr_debuginfo
```

En dan die output hier delen.

dennissiemensma, 2020-10-14:
Die worker timeouts zijn trouwens van met name 1 oktober zie ik nu pas, dus dat is wel een flinke tijd geleden. Of komt dat overeen met wanneer je problemen had?

poudenes, 2020-10-15:
Dit is de uitkomst van de debuginfo
Het probleem was niet bekend op 1 oktober.
```

DSMR-READER
  Version                                                    v4.4.3
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              5.0 s
  Setting: Datalogger process sleep                           5.0 s
  Setting: Retention cleans up after                         8064 h

SMART METER
  Telegram version (latest reading)                             v50
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                           13328581
                        \_ unprocessed                            4
  Stored: Electricity consumption records                    240718
  Stored: Gas consumption records                             47113

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                        2473 MB
  public.dsmr_consumption_electricityconsumption              82 MB
  public.dsmr_consumption_gasconsumption                    4896 kB

```

dennissiemensma, 2020-10-15:
Bedankt voor het delen. Ondanks dat de datalogger een redelijke sleep heeft, en retentie ook aan staat, denk ik dat er nog simpelweg teveel data is. Een metingen-database van 2,5GB is echt te groot, helemaal als dit op hardware zoals een Pi gedraaid wordt.

Ter vergelijking, mijn data (na bijna 5 jaar):

```
DATA
    Telegrams                                                                                 
      - total (est.)                                                                    144135
      - version (latest reading)                                                          "42"
    Consumption records                                                                       
      - electricity (est.)                                                               93186
      - gas (est.)                                                                       41316

POSTGRESQL SIZE OF LARGEST TABLES (> 5 MB)
    public.dsmr_datalogger_dsmrreading                                                   43 MB
    public.dsmr_consumption_electricityconsumption                                       40 MB
```

Ik kan je aanraden om te beginnen met de retentie op een maand of korter te zetten. Dat schoont alleen metingen op. Dagtotalen blijven oneindig bewaard.

poudenes, 2020-10-16:
Kan het zijn dat nadat ik de retentie heb aangepast, dat hij onwijs druk is nu om alles weg te verwijderen dat ouder dan een maand is? Want sinds de aanpassing ben ik gehele systeem kwijt... geeft niks meer door aan MQTT etc... Harde reboot gedaan. Kan wel op de RPi komen, maar de WebUI en alles is gone...

poudenes, 2020-10-16:
Is er een command om handmatig alle oude retentie berichten te verwijderen?
Systeem is weer terug, maar output is alleen maar groter geworden...

DATA
  Stored: Telegram records total                           11365156
                        \_ unprocessed                          267
  Stored: Electricity consumption records                    209754
  Stored: Gas consumption records                             42004

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                        2475 MB
  public.dsmr_consumption_electricityconsumption              83 MB
  public.dsmr_consumption_gasconsumption                    4920 kB

```

dennissiemensma, 2020-10-16:
Dank voor je update. Afhankelijk van je hardware (vooral je opslag) kan dit inderdaad wat performance kosten, welliswaar tijdelijk, maar dat verschilt per installatie.

Je kunt dit even proberen voor nu, om te zien of die wat schijfruimte vrijgeeft:
```
sudo su - postgres
vacuumdb -f -v -d dsmrreader
```

Normaal hoef je dat nooit te doen, maar bij relatief grote database-deletes in korte tijd kan het wat toevoegen. 

Verder kun je even handmatig kijken hoever die is met het opruimen, zie dit issue en specifiek dat comment, maar ook het verhaal er omheen: https://github.com/dennissiemensma/dsmr-reader/issues/1071#issuecomment-670690844

poudenes, 2020-10-17:
Nadat ik retentie naar 1 week heb gedaan toen je commands zoals hierboven en daarna de debug output is dit uiteindelijk nu het resultaat:

```
DATA
  Stored: Telegram records total                           11279412
                        \_ unprocessed                           46
  Stored: Electricity consumption records                    208410
  Stored: Gas consumption records                             41780

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                        2081 MB
  public.dsmr_consumption_electricityconsumption              70 MB
  public.dsmr_consumption_gasconsumption                    4336 kB
```

dennissiemensma, 2020-10-17:
Dat is al een mooi begin, ik zou het nog een dagje of wat geven. Met de link uit mn vorige comment kun je zien hoeveel maanden/weken die nog ongeveer moet.----------Er staan momenteel nog 11 miljoen metingen in je DB dus opschonen duurt even. Het is ook goed voor je om te weten dat opschonen NIET alle metingen weggooit, maar van elk uur de eerste en laatste bewaart. Daarom duurt het zo lang, dat is vrij intensief.

poudenes, 2020-10-17:
Belangrijkste is dat de dagtotalen bewaard blijven. Die zijn handig voor langer termijn.
Ik geef het gewoon even wat rust. Is het aan te raden iets uit te zetten waardoor het makkelijker zal gaan met schonen?

dennissiemensma, 2020-10-17:
Je kunt de backend sleep tijdelijk iets lager zetten (1s) en de datalogger sleep tijdelijk wat hoger (10s).

neverstoppedaskingwhy, 2020-10-18:
Hoi Dennis, 

Sorry voor het hijacken van deze thread, maar ik heb een vraag met betrekking tot 

> sudo su - postgres
> vacuumdb -f -v -d dsmrreader
> ```
> 
> Normaal hoef je dat nooit te doen, maar bij relatief grote database-deletes in korte tijd kan het wat toevoegen.
 [#1071 (comment)](https://github.com/dennissiemensma/dsmr-reader/issues/1071#issuecomment-670690844)

Ik heb ook sporadisch issues met dsmr op een raspberry pi v3 met de performance of bereikbaarheid van de web interface.

Mijn status voor het uitvoeren van vacuumdb:

```

DSMR-READER
  Version                                                    v4.4.3
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              1.0 s
  Setting: Datalogger process sleep                           0.5 s
  Setting: Retention cleans up after                          168 h

SMART METER
  Telegram version (latest reading)                             v50
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                             626490
                        \_ unprocessed                          194
  Stored: Electricity consumption records                     36914
  Stored: Gas consumption records                             28838

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                        5039 MB
  public.dsmr_consumption_electricityconsumption             228 MB
  public.dsmr_consumption_gasconsumption                      11 MB
  public.dsmr_stats_hourstatistics                          1576 kB
  public.dsmr_weather_temperaturereading                    1336 kB

Mijn status na vacuumdb:

```

DSMR-READER
  Version                                                    v4.4.3
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                              1.0 s
  Setting: Datalogger process sleep                           0.5 s
  Setting: Retention cleans up after                          168 h

SMART METER
  Telegram version (latest reading)                             v50
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                             627019
                        \_ unprocessed                           51
  Stored: Electricity consumption records                     36923
  Stored: Gas consumption records                             28840

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)
  public.dsmr_datalogger_dsmrreading                         120 MB
  public.dsmr_consumption_electricityconsumption              13 MB
  public.dsmr_consumption_gasconsumption                    3000 kB
  public.dsmr_stats_hourstatistics                          1552 kB
  public.dsmr_weather_temperaturereading                    1312 kB

```
De retention had ik aan het begin van het jaar gereduceerd naar één week. Kan het zijn, dat op deze hardware met deze instellingen het te lang duurt om "overbodige" database entries te verwijderen, zodat een vacuumdb gewoon noodzakelijk is?

Om problemen bij nieuwe installaties te vermeiden, is het misschien een idee om bij telegram formaat versie v50 de retentie via het backend niet langer als één week instelbaar te maken?

Bedankt,
Mark

poudenes, 2020-10-18:
alles nu gedaan. Ben bang dat ik mijn dB kwijt ga raken. Alles gedaan sleep time aangepast. vacuum actie uitgevoerd. maar systeem blijft nu ook na een hele nacht rust onbereikbaar. Harde reboot werkt even, dan kan ik inloggen, maar als systeem weer de dockers start ben ik het systeem kwijt.

dennissiemensma, 2020-10-18:
@poudenes op wat voor hardware draai je dit en wat heb je er nog meer op draaien?

poudenes, 2020-10-18:
RPi3B+ en alleen dsmr. Draaiende op SD card. Wellicht alles overzetten met rpiclone naar ssd.... Als enige redmiddel

dennissiemensma, 2020-10-18:
Ben je inmiddels in de gelegenheid geweest om deze queries uit te voeren om te zien hoever die is met opschonen? https://github.com/dennissiemensma/dsmr-reader/issues/1071#issuecomment-670690844

En wat je sowieso mag doen is kijken of je een kopie kan maken van de bestanden in `/home/dsmr/dsmr-reader/backups/` (of kijken in Dropbox op daar recente backups in staan). Dan is de kans ook vrij klein dat je iets kwijt gaat raken hoor.

Als alternatief kun je een CSV-export maken van de dagtotalen via `/export`. Dan heb je die in ieder geval.

poudenes, 2020-10-19:
Hier de output. Ik heb gemerkt als ik de backend deel stop dat webinterface en command line stukken sneller is

```
select date_trunc('month', timestamp) as month, count(id) as total from dsmr_datalogger_dsmrreading group by month order by month desc;
```

```
         month          |  total
------------------------+---------
 2020-10-01 00:00:00+01 |  199528
 2020-09-01 00:00:00+01 | 2500309
 2020-08-01 00:00:00+01 | 2420346
 2020-07-01 00:00:00+01 |    1488
 2020-06-01 00:00:00+01 |    1436
 2020-05-01 00:00:00+01 |    1336
 2020-04-01 00:00:00+01 |     318
```

```select date_trunc('month', read_at) as month, count(id) as total from dsmr_consumption_electricityconsumption group by month order by month desc;
```

```
         month          | total
------------------------+-------
 2020-10-01 00:00:00+01 | 20680
 2020-09-01 00:00:00+01 | 42582
 2020-08-01 00:00:00+01 | 40986
 2020-07-01 00:00:00+01 |  1488
 2020-06-01 00:00:00+01 |  1436
 2020-05-01 00:00:00+01 |  1336
 2020-04-01 00:00:00+01 |   318
```

```
select date_trunc('month', read_at) as month, count(id) as total from dsmr_consumption_gasconsumption group by month order by month desc;
```

```
         month          | total
------------------------+-------
 2020-10-01 00:00:00+01 |  4074
 2020-09-01 00:00:00+01 |  8290
 2020-08-01 00:00:00+01 |  8059
 2020-07-01 00:00:00+01 |  1488
 2020-06-01 00:00:00+01 |  1436
 2020-05-01 00:00:00+01 |  1336
 2020-04-01 00:00:00+01 |   318
```

dennissiemensma, 2020-10-19:
Bedankt voor je update. De backend doet het opschonen, dus het is inderdaad vanzelfsprekend dat dat een hoop load trekt. Doe vooral na afloop nog een laatste vacuumdb en daarna een system-herstart om de swap leeg te maken. 

Qua dataaantallen lijkt het allemaal al stukken beter. Zoals je kunt zien heeft die nu nog maar twee maanden aan data te gaan voordat die bij is. Het kan best zijn dat die morgenochtend klaar is.

poudenes, 2020-10-22:
Nog steeds onbereikbaar. Reset gedaan. backend even uit gezet en opnieuw opgevraagd.  Hij gaat wel door zo te zien. Bijna klaar.. :) 
```
dsmrreader=# select date_trunc('month', timestamp) as month, count(id) as total from dsmr_datalogger_dsmrreading group by month order by month desc;
         month          | total
------------------------+--------
 2020-10-01 00:00:00+01 | 199937
 2020-09-01 00:00:00+01 | 165156
 2020-08-01 00:00:00+01 |   1488
 2020-07-01 00:00:00+01 |   1488
 2020-06-01 00:00:00+01 |   1436
 2020-05-01 00:00:00+01 |   1336
 2020-04-01 00:00:00+01 |    318
(7 rows)
```

```
dsmrreader=# select date_trunc('month', read_at) as month, count(id) as total from dsmr_consumption_electricityconsumption group by month order by month desc;
         month          | total
------------------------+-------
 2020-10-01 00:00:00+01 | 20919
 2020-09-01 00:00:00+01 |  4207
 2020-08-01 00:00:00+01 |  1488
 2020-07-01 00:00:00+01 |  1488
 2020-06-01 00:00:00+01 |  1436
 2020-05-01 00:00:00+01 |  1336
 2020-04-01 00:00:00+01 |   318
(7 rows)
```

```
dsmrreader=# select date_trunc('month', read_at) as month, count(id) as total from dsmr_consumption_gasconsumption group by month order by month desc;
         month          | total
------------------------+-------
 2020-10-01 00:00:00+01 |  4123
 2020-09-01 00:00:00+01 |  1897
 2020-08-01 00:00:00+01 |  1488
 2020-07-01 00:00:00+01 |  1488
 2020-06-01 00:00:00+01 |  1436
 2020-05-01 00:00:00+01 |  1336
 2020-04-01 00:00:00+01 |   318
(7 rows)
```

poudenes, 2020-10-22:
De snelheid vliegt er nu in. Denk dat het heel veel heeft geholpen uiteindelijk swap off te doen en dan de backend aanzetten.
Want als je kijkt naar voorgaande post en deze zit daar 35min tussen en nu geheel klaar bijna.

Daarvoor 3 dagen laten lopen en was net 1 maandje klaar.

```
select date_trunc('month', timestamp) as month, count(id) as total from dsmr_datalogger_dsmrreading group by month order by month desc;
         month          | total
------------------------+-------
 2020-10-01 00:00:00+01 | 11132
 2020-09-01 00:00:00+01 |  1424
 2020-08-01 00:00:00+01 |  1488
 2020-07-01 00:00:00+01 |  1488
 2020-06-01 00:00:00+01 |  1436
 2020-05-01 00:00:00+01 |  1336
 2020-04-01 00:00:00+01 |   318

select date_trunc('month', read_at) as month, count(id) as total from dsmr_consumption_electricityconsumption group by month order by month desc;
         month          | total
------------------------+-------
 2020-10-01 00:00:00+01 |  2350
 2020-09-01 00:00:00+01 |  1424
 2020-08-01 00:00:00+01 |  1488
 2020-07-01 00:00:00+01 |  1488
 2020-06-01 00:00:00+01 |  1436
 2020-05-01 00:00:00+01 |  1336
 2020-04-01 00:00:00+01 |   318

select date_trunc('month', read_at) as month, count(id) as total from dsmr_consumption_gasconsumption group by month order by month desc;
         month          | total
------------------------+-------
 2020-10-01 00:00:00+01 |   988
 2020-09-01 00:00:00+01 |  1423
 2020-08-01 00:00:00+01 |  1488
 2020-07-01 00:00:00+01 |  1488
 2020-06-01 00:00:00+01 |  1436
 2020-05-01 00:00:00+01 |  1336
 2020-04-01 00:00:00+01 |   318
```

dennissiemensma, 2020-10-22:
Fijn om te horen. Als het goed is is je systeem nu ook weer snel met alle processen aan. Vanaf nu is het verstandig om ervoor te zorgen dat de data niet te hard groeit. Dus houd de datalogger sleep vooral boven de 5 seconden!

En tot slot denk eraan om af en toe een backup te (laten) maken buiten je SD-kaart om.

closed 2020-10-22 by dennissiemensma