dsmrreader/dsmr-reader #1150: Elektra is geimporteerd, maar gas niet

HellStorm666, 2020-10-14:
Ik heb de data vanuit domoticz geimporteerd volgens de stappen van https://github.com/dennissiemensma/dsmr-reader/issues/908#issuecomment-592456012

Ik heb daar 
`##Filter Specific DeviceRowID
$MULTIMETERCALENDARFILTER = $MULTIMETERCALENDAR | Where {$_.DeviceRowID -eq "29"}`

aangepast naar
`##Filter Specific DeviceRowID
$MULTIMETERCALENDARFILTER = $MULTIMETERCALENDAR | Where {$_.DeviceRowID -eq "1"}`

Aangezien de device id rij in de csv voor 99% vol stond met de waarde 1.

Voor Meter_Calendar.csv is dat trouwens 2.

Nu is alle elektra (alle 4 de meters) netjes geimporteerd.
Maar gas blijft leeg.

Hoe kan ik deze gas data toevoegen?

Ik verwacht dat ik de gas data nog kan toevoegen aan dezelfde datum als de elektra en die datum in de csv opzoek en de waarde van counter toe te voegen... maarja, hoe?

dennissiemensma, 2020-10-14:
Bedankt voor je vraag. Helaas heb ik geen kennis hoe Domoticz werkt en zeggen de meeste waarden of velden mij niets. Voor dat inhoudelijke script kun je het beste bij de maker zijn.

Het hangt er verder denk ik erg vanaf wat je precies wilt importeren. Als je gegevens hebt in de vorm van dagtotalen dan kun je het wellicht omzetten naar een formaat wat je direct kunt importeren in de DSMR-reader-database. 
Maar dat hangt dus echt af van de gegevens die je hebt, wellicht heb je daar wat meer informatie over want daar kan ik je inhoudelijker mee helpen?

HellStorm666, 2020-10-15:
Ik heb uiteindelijk zelf het script wat aangepast.
Daardoor werd hetvolgende verstuurd:
`http://10.8.1.2:7777/api/v2/datalogger/dsmrreading  -Headers $auth -Method Post -Body $Data`
en in $Data zit dan
`-electricity_delivered_1: 10656.456 
-electricity_delivered_2: 9982.104
-electricity_returned_2: 3678.891
-electricity_currently_returned: 0
-electricity_currently_delivered: 0
-timestamp: 2020-10-13T00:00
-extra_device_timestamp: 2020-10-13T00:00
-extra_device_delivered: 6189.285
-electricity_returned_1: 1580.173`

Echter komt alle elektra data er wel in, maar de gas data niet.
Wat gaat hier dan fout?

en als bonus vraag,
Gisteren avond dus alles weg gegooit en dit aangepaste script gedraaid zodat nu ook de gas data mee ging, wat dus niet werkte.
Wat gebeurt er als ik dit script nu nog een keer draai. (desnoods aangepast) wordt de huidige data overschreven, of wordt dit toegevoegd en een rommeltje?

dennissiemensma, 2020-10-15:
- Heb je wel je datalogger en backend proces uit staan? Er kan ruis ontstaan als die processen blijven draaien tijdens de import van oude data, helemaal als de backend nog draait want dan heb je kans dat die vroegtijdig metingen verwerkt terwijl het importeren nog bezig is.

- Ik denk dat je het beste kunt kijken naar de response van het API endpoint die je aanroept. Als het goed is krijg je een object terug met de waardes zoals opgeslagen. Een startpunt zou zijn om daar te kijken (of in de DSMR-reader admin/database).

HellStorm666, 2020-10-15:
> * Heb je wel je datalogger en backend proces uit staan? Er kan ruis ontstaan als die processen blijven draaien tijdens de import van oude data, helemaal als de backend nog draait want dan heb je [...]

Nope, hoe zet ik die datalogger en backend uit?

Hoe kan ik die API response checken?
PS geeft wel reactie terug, daarin is netjes te zien welk ID t heeft gekregen en welke info er geplaatst is. Bedoel je die?
Want daar staat alles gewoon netjes in.

Kan ik het script nog een keer draaien, maar dan met die datalogger en backend uit? of wordt t dan chaos omdat die data er al een keer in staat?

dennissiemensma, 2020-10-15:
Prima, dan heb je in ieder geval de juiste data in de API.

DSMR-reader bestaat uit drie processen: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#supervisor
Je kunt ze per stuk uitzetten met:
```
sudo supervisorctl stop dsmr_backend
sudo supervisorctl stop dsmr_datalogger
```

De webinterface is nodig voor de API.

Voor het beste resultaat importeer je het met een lege database. Daarvoor kun je deze tabellen leeggooien (of een nieuwe database aanmaken en migraties draaien):

```
dsmr_datalogger_dsmrreading

dsmr_consumption_electricityconsumption
dsmr_consumption_gasconsumption

dsmr_stats_hourstatistics
dsmr_stats_daystatistics
```

- Importeren van data via de API zet data in de eerste tabel.  
- Als de backend later weer aangaat, verwerkt die alles om tabellen 2 en 3 te vullen. 
- Voor elke gepaseerde dag vult die, met de gegevens uit tabellen 2 en 3, uiteindelijk de tabellen 4 en 5. Die laatste twee worden gebruikt in het Archief.

HellStorm666, 2020-10-16:
> Prima, dan heb je in ieder geval de juiste data in de API.
>
> DSMR-reader bestaat uit drie processen: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#supervisor
> Je kunt ze per stuk [...]

als ik in de docker console (ingelogd als root) supervisorctl stop dsmr_backend in geef, krijg ik de fout unix:///run/supervisord.sock no such file....

En hoe gooi ik die database tabellen leeg?
Heb postgres 12.3 draaien in docker.

dennissiemensma, 2020-10-16:
Bedankt voor je update. De belangrijkste waardes zijn degene die in de dag- en uurstatistieken komen. De anderen worden alleen (conditioneel) in de live grafieken getoond en zijn vluchtig.

Is het inmiddels gelukt om een nieuwe lege import te doen met de processen uit? Voor hoe de processen in Docker uitgezet kunnen worden weet ik helaas niet, wellicht kun je xirixiz vragen (als je zijn Docker image gebruikt).

HellStorm666, 2020-10-19:
> Bedankt voor je update. De belangrijkste waardes zijn degene die in de dag- en uurstatistieken komen. De anderen worden alleen (conditioneel) in de live grafieken getoond en zijn vluchtig.
>
> Is het [...]

Nope, nog niet gelukt omdat ik het niet voor elkaar krijg om die processen uit te zetten.
Ik blijf de fout unix:///run/supervisord.sock no such file krijgen. Zelfs als ik supervisor installeer op ubuntu en via de mapping die /var/run/supervisor.sock koppel aan de ubuntu versie.

De env. 
SD_AUTOSTART_CLIENT = false
SD_AUTOSTART_BACKEND = false
SD_AUTOSTART_LOGGER = false

werken ook niet.
Doe ik ene kill pid * (*nummer van pid backend) dan start ie daarna direct de backend met een nieuwe pid
ook als ik via de env de restarts disable er van.

HellStorm666, 2020-10-19:
EDIT:
Gefixed, weer alles weggeknikkerd.
Deze keer ook geen bindings gemaakt in docker naar ubuntu (dus geen enkele -v er in).
Geen P1 port aangesloten, direct het script gedraaid, daarna generate day and hour statistics gerund.
En nu komt de data van t gas wel zichtbaar er in.
Zodra dit klaar is de P1 weer mappen in docker en dan hopen dat het goed blijft gaan :)

dennissiemensma, 2020-10-19:
Dank voor je update. Wat was nou de oplossing, inkomende data en bakcend uitschakelen?

HellStorm666, 2020-10-19:
nope, de P1 stekker niet koppelen in docker (dus eigenlijk de P1 stekker er uit trekken) en dan de setup opnieuw beginnen.
Dan de import draaien, dan de generate day and hour statistics draaien, dan de P1 stekker er pas weer in.

Ik kreeg het nl niet voor elkaar om die backend en inkomende data in dsmr uit te schakelen.

dennissiemensma, 2020-10-19:
Ok, helder. Fijn dat het dan met een workaround toch gelukt is!

closed 2020-10-19 by dennissiemensma