dsmrreader/dsmr-reader #1151: Gotify notification support

HellStorm666, 2020-10-15:
Please add Gotify as notification support.
Free, multi-platform and really easy to use notification system.

https://gotify.net/

with a simple HTTP post or curl messages are send.
`$ curl "https://push.example.de/message?token=<apptoken>" -F "title=my title" -F "message=my message" -F "priority=5"
$ http -f POST "https://push.example.de/message?token=<apptoken>" title="my title" message="my message" priority="5"`

For Powershell this method is working fine:
edit the url, api, prio from within the Notification page of DSMR.
fill the title and message with the DSRM message

`$url = "https://your.url"
$api = "your_api"
$prio = "5"
$Data = @{
'title'= "Test";
'message'= "Dit is een test";
'priority'= $prio;
}
Invoke-WebRequest $url/message?token=$api -Method Post -Body $Data`

dennissiemensma, 2020-10-15:
Thanks for your suggestion. I might add other providers when enough users require it, but for now I'd suggest you'll just make it work by using a dedicated plugin for yourself: https://dsmr-reader.readthedocs.io/en/latest/plugins.html

dennissiemensma, 2020-10-15:
In order to easily allow you to do so I'll add **a new hook/event in DSMR-reader 4.5** you can use. The plugin would then be a script such as:

## `dsmr_plugins/modules/relay_notification_to_gotify.py`
```
import requests

from django.dispatch import receiver

from dsmr_notification.signals import notification_sent

@receiver(notification_sent)
def handle_notification_sent(sender, title, message, **kwargs):
    APP_TOKEN = '...'

    try:
        requests.post(
            'https://push.example.de/message?token={}'.format(APP_TOKEN),
            data=dict(
                title=title,
                message=message,
                priority=5,
            ),
            # A low timeout prevents DSMR-reader from hanging, when the remote server is unreachable.
            timeout=5
        )
    except Exception as error:
        print('handle_notification_sent:', error)

```

## `.env`
```
DSMRREADER_PLUGINS=dsmr_plugins.modules.relay_notification_to_gotify
```

dennissiemensma, 2020-10-15:
You will also need to select a (new) Dummy-notification provider in the admin to ensure notifications are triggered without requiring another provider.

dennissiemensma, 2020-10-15:
Added to v4.5 release.

closed 2020-10-15 by dennissiemensma

HellStorm666, 2020-10-15:
great thanks!

tried to make the dsmr_plugins/modules/relay_notification_to_gotify.py with the correct api and webaddress, and then add the DSMRREADER_PLUGINS=dsmr_plugins.modules.relay_notification_to_gotify in docker env.

But, DSMR_Reader doesn't start then, it crashes with an error in web-service.
Removing the env fixes it. But running the .py gives the error that notification service isnt know.
Enabling notification from the admin pannel and selecting just and, click save, doesnt save this.

dennissiemensma, 2020-10-15:
It will not work until the v4.5 release.

dennissiemensma, 2020-10-19:
Zojuist uitgebracht in [v4.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.5.0)

HellStorm666, 2020-10-20:
Tested it, and it works!
Thanks.