dsmrreader/dsmr-reader #1152: Live grafieken laden niet + current energy usage/return in HA onbekend

GerbenKwakkel, 2020-10-15:
Ha!

Al bijna een jaar met volle tevredenheid gebruikmakend van DSMR-reader (laatste versie 4.4.3), loop ik toch enige tijd tegen het probleem aan dat de live grafieken niet meer laden.
De pagina volledig laden doet het niet en op een gegeven moment klapt hij eruit. Na een 20-30 seconden is alles dan ook weer up.

Daarnaast push ik de metingen via MQTT naar HA en daarbij zijn alleen de waarden 'Current energy usage' of 'Current energy return' niet bekend. Is dit gelinkt aan bovenstaand probleem? Ditgene vind ik misschien nog wel meer prio hebben dan de live grafieken, maar iets zegt mij dat het gelinkt is aan elkaar.

Gisteravond heb ik op de database al een vacuumjob gedraaid, de datalogger sleep proces op 10 seconden gezet en de retentie op een week gezet.
Nu laad de grafiekenpagina wel, heeeel langzaam, maar klapt er na een 30-seconden alsnog uit.

Zijn er nog ideeën wat ik kan doen of uitvoeren? Wellicht is mijn informatie wat beknopt.

wh33ly, 2020-10-15:
Waarschijnlijk zit je match ergens fout tussen HA/MQTT

- Check HA en controleer de  'Current energy usage' en 'Current energy return' sensors en kijk op welk topic deze luisteren
- Check DSMR configuratie -> MQTT -> en open het de configuratie van het Topic waar je HA naar verwijst.
- Controleer de velden ik vermoed dat je het  hier al gevonden hebt, 

Je zou ook gebruik kunnen maken van MQTT explorer en naar MQTT connecten. Zo kun je handig browsen :)

Als het goed is zoek je: 
- electricity_currently_delivered
- electricity_currently_returned

GerbenKwakkel, 2020-10-15:
Bedankt voor je snelle comment @wh33ly .

Ik heb het even gecontroleerd met MQTT explorer maar daar zie ik die twee waarden niet:

[image]

wh33ly, 2020-10-15:
Ik denk dat je “(Databron) Telegram” niet hebt ingeschakeld. Daar zitten de waardes in die je zoekt. Zie ze iig niet in je screenshot

GerbenKwakkel, 2020-10-15:
@wh33ly enorm bedankt! Dat was de fout (stom van mijzelf)... Zag het woord Telegram en dacht aan de app...

Wat zou het nog kunnen zijn met de live grafieken? Dat deze enorm traag laden.

wh33ly, 2020-10-15:
Graag gedaan, ik had precies dezelfde gedachte afgelopen week :-)

Grafieken kan ik je zo snel ook niet mee helpen.

dennissiemensma, 2020-10-15:
@GerbenKwakkel dank voor je vraag en @wh33ly voor de voorlopige hulp.

Ik denk dat je het beste kunt beginnen met het delen van de output van dit:
```
sudo su - dsmr
./manage.py dsmr_debuginfo
```

Ik vermoed namelijk dat er (nog) te veel data opgeslagen is, ondanks dat je aangeeft daar zelf ook al mee bezig geweest te zijn.

dennissiemensma, 2020-11-09:
@GerbenKwakkel ben je hier nog verder mee gekomen?

GerbenKwakkel, 2020-11-10:
Ha @dennissiemensma sinds ik de vacuum-cmd heb gedraaid, de retentie op een week heb gezet, en de interval naar 10-seconden heb gezet i.p.v. 1 seconde draait het nu weer als een zonnetje!

Bedankt voor het meedenken.

dennissiemensma, 2020-11-10:
Fijn om te horen, dank voor je terugkoppeling!

closed 2020-11-10 by dennissiemensma