dsmrreader/dsmr-reader #1158: InfluxDB Measurements Gas Consumption

Djowie, 2020-10-19:
Alle data wordt vanuit DSMR doorgezet naar de InfluxDB instance, dit werkt helemaal perfect. Zou alleen graag wat meer measurements willen meegeven zoals Gas Consumption. Krijg dit alleen niet werkend, is dit uberhaupt mogelijk, las wel iets over het definiëren van je 'eigen' measurements. Kan ik ergens vinden hoe dit werkt? Of zou iemand mij hierbij kunnen helpen?

Op dit moment gebruikt is de 'default' measurements voor de InfluxDB integratie.

slashw, 2020-10-20:
[image]

Dit zijn mijn instellingen,
Onder field selecteer je "deliverd", hierna zou je je verbruik moeten zien.

dennissiemensma, 2020-10-20:
@Djowie bedankt voor je vraag. Ik ken InfluxDB zelf niet zo goed, maar wellicht is het mogelijk om die zelf de aggregatie te laten maken tussen de meterstand-verschillen. 

Ik probeer te voorkomen dat ik voor elk stukje data in DSMR-reader + elke soort externe database een aparte integratie moet maken en vervolgens ook onderhouden.
In principe is e.e.a. namelijk ook uit te lezen via de API van DSMR-reader, gecombineerd met een script dat het doorstuurt naar een externe opslag, zoals InfluxDB. De ingebouwde integraties blijven dan beperkt tot de initiele brongegevens: de telegrammen uit de slimme meter.

Djowie, 2020-10-20:
@dennissiemensma Dat is een heel slim idee :), dat zou me moeten lukken. Bedankt voor je snelle antwoord en klinkt inderdaad logisch zodat je niet alle integraties met externe systemen hoeft te onerhouden.

closed 2020-10-20 by Djowie