dsmrreader/dsmr-reader #1159: Invalid telegram CRC

gerjaneikens, 2020-10-20:
(Heb je DSMR-reader v4.4 of hoger? Zet dan hieronder ook de output van "dsmr_debuginfo") 
DSMR-READER
  Version                                                    v4.4.3
  Database engine/vendor                                 postgresql
  Setting: Backend process sleep                             60.0 s
  Setting: Datalogger process sleep                          60.0 s
  Setting: Retention cleans up after                         8064 h

SMART METER
  Telegram version (latest reading)                             v42
  Setting: Telegram parser                                        4

DATA
  Stored: Telegram records total                                  1
                        \_ unprocessed                            1
  Stored: Electricity consumption records                         0
  Stored: Gas consumption records                                 0

POSTGRESQL SIZE OF LARGEST TABLES (> 1MB)

```

---
Gister installatie gedaan van dsmr-reader. Geeft vanaf het begin al 
"Invalid telegram CRC. The calculated checksum does not match the telegram checksum" aan.

Hele nacht laten lopen en heeft blijkbaar 1 keer goed uitgelezen.

datalogger & backend process sleep op 60 seconden gezet en een nieuwe kabel geprobeerd.
Meter is een Landis Gyr e350 met DSMR 4.0 
DSMR Reader loopt op een raspberry pi 4

dennissiemensma, 2020-10-20:
Bedankt voor je melding. Ik denk dat je allereerst mag kijken of je wel constant telegrammen krijgt via de tool `cu`: https://dsmr-reader.readthedocs.io/nl/v4/installation/explained.html#your-first-reading-optional

Als dat wel goed lijkt te werken (elke 10 seconden een telegram), dan kun je het beste DEBUG-logging aanzetten in DSMR-reader en dan de logfile van de dsmr_datalogger checken: https://dsmr-reader.readthedocs.io/nl/v4/troubleshooting.html#log-files
Voor dat laatste is het wel aan te raden om je datalogger sleep juist zo laag mogelijk te zetten, zodat je niet lang hoeft te wachten totdat er wat gebeurt.

gerjaneikens, 2020-10-20:
cu output:
`pi@raspberrypi:~ $ cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q
Connected.
/XMX5LGBBFFB231028291

1-3:0.2.8(42)
0-0:1.0.0(201020183426S)
0-0:96.1.1(4530303035303031353430363839343134)
0-0:1.8.1(020756.680*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:1.8.2(025323.323*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(01.096*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00001)
1-0:99.97.0(1)(0-0:96.7.19)(180713122834S)(0000007353*s)
1-0:32.32.0(00000)
1-0:52.32.0(00001)
1-0:72.32.0(00000)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(004*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.831*kW)
1)0:41.7.0(00.245*kW)
1-0:61.7.0(00.019*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0 003)
0-1:96.1.0(4730303136353631323130333038393134)
0-1:24.2,1(201020180000S)(12046.433*m3)
!2FEB
`

debug log:
`[2020-10-20 18:33:15,540] INFO     [2020-10-20 18:33:15.540557] Opening connection "/dev/ttyUSB0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
[2020-10-20 18:33:15,881] DEBUG    [2020-10-20 18:33:15.881459] Read 0 Byte(s)
[2020-10-20 18:33:16,215] DEBUG    [2020-10-20 18:33:16.215778] Read 0 Byte(s)
[2020-10-20 18:33:16,550] DEBUG    [2020-10-20 18:33:16.550171] Read 0 Byte(s)
[2020-10-20 18:33:16,884] DEBUG    [2020-10-20 18:33:16.884541] Read 0 Byte(s)
[2020-10-20 18:33:17,218] DEBUG    [2020-10-20 18:33:17.218814] Read 0 Byte(s)
[2020-10-20 18:33:17,553] DEBUG    [2020-10-20 18:33:17.553268] Read 0 Byte(s)
[2020-10-20 18:33:17,887] DEBUG    [2020-10-20 18:33:17.887676] Read 0 Byte(s)
[2020-10-20 18:33:18,222] DEBUG    [2020-10-20 18:33:18.222060] Read 0 Byte(s)
[2020-10-20 18:33:18,556] DEBUG    [2020-10-20 18:33:18.556477] Read 0 Byte(s)
[2020-10-20 18:33:18,890] DEBUG    [2020-10-20 18:33:18.890831] Read 0 Byte(s)
[2020-10-20 18:33:19,225] DEBUG    [2020-10-20 18:33:19.225154] Read 0 Byte(s)
[2020-10-20 18:33:19,559] DEBUG    [2020-10-20 18:33:19.559483] Read 0 Byte(s)
[2020-10-20 18:33:19,893] DEBUG    [2020-10-20 18:33:19.893218] Read 285 Byte(s)
[2020-10-20 18:33:20,227] DEBUG    [2020-10-20 18:33:20.227027] Read 548 Byte(s)
[2020-10-20 18:33:20,562] DEBUG    [2020-10-20 18:33:20.562174] Read 7 Byte(s)
[2020-10-20 18:33:20,572] DEBUG    Received telegram:
/XMX5LGBBFFB231028291

1-3:0.2.8(42)
0-0:1.0.0(201020183226S)
0-0:96.1.1(4530303035303031353430363839343134)
1-0:1.8.1(020756.680*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:1.8.2(025323.286*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(01.103*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00001)
1-0:99.97.0(1)(0-0:96.7.19)(180713122834Q)(0000007353*s)
1-0:32.32.0(00000)
1-0:52.32.0(00001)
1-0:72.32.0(00000)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:7".3.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(004*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.838*kW)
1-0:41.7.0(00.246*kW)
1-0:61.7.0(00.019*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303136353631323130333038393134)
0-1:24.2.1(201020180000S)(12046.433*m3)
!A8ED
[2020-10-20 18:33:20,590] WARNING  Rejected telegram: Invalid telegram CRC. The calculated checksum '52236' (CC0C) does not match the telegram checksum '43245' (A8ED)
`

dennissiemensma, 2020-10-20:
Bedankt voor de snelle update. Ik kan het niet met zekerheid zeggen, maar het kan heel goed te maken hebben met je kabel.

Ik zie namelijk in de dump van `cu` al met het blote oog twee afwijkingen:

```
# Het tweede teken hoort een streepje (-) te zijn.
1)0:41.7.0(00.245*kW)

# De komma hier hoort een punt te zijn.
0-1:24.2,1(201020180000S)(12046.433*m3)
```

In de dump van de datalogger zelf zie ik er ook al eentje:
```
# Het dubbele aanhalingsteken hoort een "2" te zijn (zie je cu dump)
1-0:7".3.0(00000)
```

Al met al lijkt de communicatie dus niet goed te zijn. Meestal is de kabel de oorzaak, maar soms is het ook storing rond de kabel (die kan gevoelig zijn) zoals een stroomsnoer dat tegen de kabel hangt, etc. 

Die CRC-check beschermt dus tegen deze afwijkingen en verwerkt "corrupte" telegrammen niet tenzij de checksum (laatste 4 tekens in de laatste regel) overeenkomt met het telegram zelf.

dennissiemensma, 2020-10-20:
Verder kun je ook nog even zoeken (en evt daar vragen) op het Tweakers-forum in [bijvoorbeeld dit topic](https://gathering.tweakers.net/forum/list_messages/1578510) of er soortgelijke situaties zijn bij anderen. Ik kan namelijk niet uit eigen ervaring spreken, zelf heb ik zelden foutieve CRC's gehad.

gerjaneikens, 2020-10-20:
Ik heb het probleem gevonden.  Mijn pi werd gevoed met een POE splitter vanuit mijn ubiquiti switch.  Rechtstreeks via de voeding doet hij het wel. Bedankt voor de tijd en moeite!

closed 2020-10-20 by gerjaneikens