dsmrreader/dsmr-reader #1160: Table header latest_timestamp optimaliseren voor mobiele weergave

knirefel, 2020-10-21:
Regelmatig bekijk ik het DSMR Reader Dashboard op een mobiel apparaat. Ik stoor me enigszins aan het feit dat de tabel kolommen continu verspringen omdat de tekst in 'latest_timestamp' wisselt van lengte. Zie hier de verschillen:

[image]
[image]

Het zou prettig zijn als de kolombreedte niet automatisch verspringt op basis van de tekstgrootte.

dennissiemensma, 2020-10-21:
Bedankt voor je melding. Ik zal er in een toekomstige release naar kijken.

dennissiemensma, 2020-11-08:
Ik heb hier laatst een tijdje naar gekeken, naar ik kreeg het niet helemaal lekker. Wellicht dat ik hiervoor die header met de 'huidige waardes' hiervoor weer los moet trekken, zoals het in vroegere versies van DSMR-reader was. Maar dat is wel wat werk, dus ik laat het voor nu even voor wat het is.

dennissiemensma, 2021-03-07:
Ik heb hier een mogelijk fix voor. Het variabele stukje tekst staat straks op een losse regel en daarnaast is de header ook losgetrokken, zodat die niet meer de hele tabel beinvloed, mocht het toch weer gebeuren. 

[image]

Tot slot zijn er nu geen twee tellers meer naast elkaar voor huidige verbruik/teruglevering, want dat scheelt breedte. Plus dat er gegarandeerd altijd 1 op `0` staat en de andere niet. Dus was het wat logischer om of de een, of de andere te tonen. Teruglevering wordt nog wel groen getoond, met een `-` er voor.

closed 2021-03-07 by dennissiemensma

dennissiemensma, 2021-03-09:
Zojuist uitgebracht in [v4.13](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.13.0).