dsmrreader/dsmr-reader #1161: Postgres update 12.4 naar 13.

michelotten, 2020-10-22:
Gisteren heb ik mijn postgres bijgewerkt van versie 12.4 naar versie 13.
Via [deze](https://peter.grman.at/upgrade-postgres-9-container-to-10/) stappen heb ik de database overgezet van 12.4 naar versie 13. Mijn postgres container kan deze database weer bereiken.
Echter, vanmorgen gaf DSMR aan dat de backup mislukt was. dsmr_backend.log zit vol met de volgende melding:
`2020-10-22 08:03:01,207] CRITICAL  - Unexpected exit code (1) for backup: b'could not find a "pg_dump" to execute\npg_dump: error: server version: 13.0 (Debian 13.0-1.pgdg100+1); pg_dump version: 12.4\npg_dump: error: aborting because of server version mismatch\n'
[2020-10-22 08:03:01,211] ERROR    (OSError) dsmr_backup.services.backup.run errored: b'could not find a "pg_dump" to execute\npg_dump: error: server version: 13.0 (Debian 13.0-1.pgdg100+1); pg_dump version: 12.4\npg_dump: error: aborting because of server version mismatch\n'`

Lijkt er sterk op dat ik aan de DSMR kant nog iets moet doen om deze ook te 'matchen' met de nieuwe database, echter welke stappen dien ik hiervoor te ondernemen?

Hierbij de debug info dump:
```

OS
    Python version                                                                                                    v3.9.0
    Platform                                                                                                  Linux (x86_64)
    System                                                                                         Linux-4.4.59+-x86_64-with

DSMR-READER
    Version                                                                                                             v4.5
    Backend sleep                                                                                                      1.0 s
    Datalogger sleep                                                                                                   0.5 s
    Retention cleans up after                                                                                          672 h
    Telegram parser                                                                                                      "4"
    Database engine/vendor                                                                                        postgresql

DATA
    Telegrams                                                                                                               
      - total (est.)                                                                                                 2446736
      - version (latest reading)                                                                                        "50"
    Consumption records                                                                                                     
      - electricity (est.)                                                                                             51068
      - gas (est.)                                                                                                     18566

POSTGRESQL SIZE OF LARGEST TABLES (> 5 MB)
    public.dsmr_datalogger_dsmrreading                                                                                428 MB
    public.dsmr_consumption_electricityconsumption                                                                     17 MB

```

Alvast bedankt.

closed 2020-10-22 by michelotten

Sjeff, 2020-11-02:
@michelotten Did you find a fix for this mismatch?

michelotten, 2020-11-02:
I rolled back to version 12.4
In this <https://github.com/xirixiz/dsmr-reader-docker/issues/132> similar
issue I saw Xirixiz linking to this
<https://pkgs.alpinelinux.org/packages?name=postgresql-client*&branch=v3.12>
webpage, showing that the latest stable branch of the Alpine image is still
using Postgres 12.4, meaning that the version 13 is not yet supported.