dsmrreader/dsmr-reader #1162: Fout tijdens upgrade 4.4.3 -> 4.5

Varsame, 2020-10-22:
Dag Dennis,

Een upgrade van 4.4.3 naar 4.5 via deploy.sh loopt vast met onderstaande traceback.. Enig idee? :)

---

Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying dsmr_consumption.0017_energy_supplier_price_refactoring...Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 85, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 245, in handle
    fake_initial=fake_initial,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 227, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/migration.py", line 124, in apply
    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/operations/models.py", line 511, in database_forwards
    getattr(new_model._meta, self.option_name, set()),
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/schema.py", line 380, in alter_unique_together    self._delete_composed_index(model, fields, {'unique': True}, self.sql_delete_unique)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/schema.py", line 419, in _delete_composed_index
    ", ".join(columns),
ValueError: Found wrong number (2) of constraints for dsmr_consumption_energysupplierprice(start, end)
>>>>> [!] Executing database migrations failed again! <<<<<
        - Running dsmr_sqlsequencereset did not resolve the problem.
        - Create an issue on Github and attach the exeception trace listed above.
[!] Halted. Post-deployment script exited with non-zero code

dennissiemensma, 2020-10-22:
Bedankt voor je melding, zie issue #1129.

Je kunt dit even checken (https://github.com/dennissiemensma/dsmr-reader/issues/1129#issuecomment-701588246) en daarna kijken of het overeen komt met dit (https://github.com/dennissiemensma/dsmr-reader/issues/1129#issuecomment-701601153) en dan evt uitvoeren.

## Zorg dat je een recente DB-backup hebt voor de zekerheid!

Varsame, 2020-10-23:
Dank voor de verwijzing, dat heeft het inderdaad opgelost! Super!

closed 2020-10-23 by Varsame