dsmrreader/dsmr-reader #1163: Vaste kosten

cmderks93, 2020-10-24:
Het zou misschien een idee zijn om voor stroom en gas apart de vaste kosten te kunnen vermelden. 
Als ik bijvoorbeeld twee verschillende leveranciers heb voor beide producten, kan ik niet voor elk product apart de vaste kosten invullen.

dennissiemensma, 2020-10-25:
Bedankt voor je verzoek. Ik probeer de tarifering wat recht-toe-recht-aan te houden om het niet complexer te maken dan nodig. Het is sowieso puur ter indicatie voor jezelf, niet om een exacte herberekening van de leverancier te maken.

Ik denk dat je het beste de vaste kosten bij elkaar kunt optellen. Indien je verschillende contractdatums hebt, zul je dan wel op twee momenten per contractduur de wisseling moeten maken en de duur aanpassen, zodat de overgang blijft kloppen.

closed 2020-11-01 by dennissiemensma

SaturnusDJ, 2021-07-05:
Na de vaste kosten optie gevonden te hebben in v4.3, kwam ik dit item tegen. Ik zou het erg handig vinden als deze optie er een keer komt. Hier één leverancier maar verschillende vaste kosten voor elektra en gas.

De enige workaround voor splitten die ik op dit moment kan bedenken is om aan het eind van een contract de vaste kosten te verdelen over het verbruik en een 'recalculate' (https://dsmr-reader.readthedocs.io/en/v4/how-to/troubleshooting/recalculate-prices.html) te doen. Maar dan kijk je dus een jaar lang tegen foute kosten aan en gaat vergelijken met het voorgaande jaar ook niet op. Een workaround daarvoor is dan weer om een inschatting van het verwachte verbruik te gaan doen.

Zou het splitsen van vaste kosten juist niet helpen om het minder complex te maken aan de voorkant doordat je dan kosten hebt voor elektra en gas (vaste kosten geïntegreerd), zodat vaste kosten niet per se los weergegeven hoeven te worden?

Een ander voordeel is dat bijv. Grafana-kostgrafieken volledig correct zijn en er niet gepruts hoeft te worden met het toevoegen van een getal, of een variabel (totale vaste kosten) dat vervolgens weer naar ratio gedeeld moet worden, waarbij beide elk jaar anders zijn.

dennissiemensma, 2021-07-05:
@SaturnusDJ bedankt voor je vraag. Ik begrijp de wens, echter is het goed om te weten dat DSMR-reader (bewust) alleen een indicatie geeft van de kosten. 

Het is namelijk nooit de bedoeling geweest om het exact te laten matchen met de werkelijke kosten, omdat dat al een heel project op zichzelf zou zijn. Elke variabele of wijziging die ik voor de ene situatie toevoeg, is weer niet afdoende voor een andere situatie. 
En ook niet voor het vergelijken van complete kosten van voorgaande jaren. Daar zijn nog veel meer factoren bij van toepassing, zoals contractduur, kortingen, saldering, etc etc. 
De vaste kosten in DSMR-reader zijn meer een compensatie voor alle overige kosten die enigzins bij een dag kunnen tellen. De ene laat de leeg, de andere voert er kosten van de leverancier in, weer een ander de kosten van de netbeheerder, iemand anders weer beiden en nog iemand anders past de welkomstkorting toe. Oftewel, er zijn zoveel factoren.

Een doel van de slimme meters (en daarmee o.a. voor DSMR-reader) is vooral om inzicht te geven in het verbruik van een huishouden. 
Daarbij zijn er tarieven (in uitgeklede vorm) in te voeren in DSMR-reader, om gebruikers een gevoel te geven wat een dag kost. Met name in de winterdagen of bij hoog elektriciteitsverbruik zijn de verschillen dan interessant voor de gebruikers. Vandaar dat het op deze manier ingericht is.