dsmrreader/dsmr-reader #1164: Dropbox API

PascalBu, 2020-10-25:
Trying to setup a dropbox backup on a new install, but doesn't seem to work. API of Dropbox seems to have changed? Also when setting all Dropbox app permissions to allow no backup...

OS
    Python version                                                                                                    v3.7.3
    Platform                                                                                                  Linux (armv7l)
    System                                                                          Linux-5.4.65-v7+-armv7l-with-debian-10.6

DSMR-READER
    Version                                                                                                             v4.5
    Backend sleep                                                                                                      1.0 s
    Datalogger sleep                                                                                                   5.0 s
    Retention cleans up after                                                                                          672 h
    Telegram parser                                                                                                      "3"
    Database engine/vendor                                                                                        postgresql

DATA
    Telegrams                                                                                                               
      - total (est.)                                                                                                  224192
      - version (latest reading)                                                                                      "None"
    Consumption records                                                                                                     
      - electricity (est.)                                                                                             36349
      - gas (est.)                                                                                                         -

ISSUES
    Wachten op de volgende Dropbox-sync die uitgevoerd moet worden                    Since 2020-10-25 08:08:03.198167+01:00

POSTGRESQL SIZE OF LARGEST TABLES (> 5 MB)
    public.dsmr_datalogger_dsmrreading                                                                                 50 MB
    public.dsmr_consumption_electricityconsumption                                                                     15 MB

```

dennissiemensma, 2020-10-25:
Bedankt voor je meldingen. Ik denk dat je het beste kunt beginnen bij de logfiles van de `dsmr_backend` om te zien of/welke fouten er naar voren komen. https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging-levels

Verder heb ik net een Dropbox app geprobeerd te maken en ik zag dat er tegenwoordig wat nieuwe opties zijn, zoals de lifetime van het API token. Zorg ervoor dat je iig kiest voor de oneindige duur.

PascalBu, 2020-10-26:
Ja, dat had ik aangezet. Dit zijn de foutmeldingen in het log:

`[2020-10-26 12:25:56,707] ERROR    (AuthError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: AuthError('28991d8e90f54134a00cb7ca6b11e142', AuthError('missing_scope', TokenScopeError(required_scope='files.metadata.read')))
[2020-10-26 12:25:56,712] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-10-26 12:25:57,722] DEBUG    SP: 0 backend service(s) ready to run
[2020-10-26 12:25:58,905] ERROR    (AuthError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: AuthError('1377e1eae96a48c4b71083f7b216e732', AuthError('missing_scope', TokenScopeError(required_scope='files.metadata.read')))
[2020-10-26 12:25:58,911] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-10-26 12:25:59,920] DEBUG    SP: 0 backend service(s) ready to run
[2020-10-26 12:26:00,768] ERROR    (AuthError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: AuthError('71048c99cc284150a96b05b5385e7652', AuthError('missing_scope', TokenScopeError(required_scope='files.metadata.read')))
[2020-10-26 12:26:00,777] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-10-26 12:26:01,789] DEBUG    SP: 0 backend service(s) ready to run
[2020-10-26 12:26:02,713] ERROR    (AuthError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: AuthError('1b03a93a7fca435abf7ff231c04379e3', AuthError('missing_scope', TokenScopeError(required_scope='files.metadata.read')))
[2020-10-26 12:26:02,720] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s`

dennissiemensma, 2020-10-26:
Bedankt voor je update. Het lijkt een rechten issue:

```
 AuthError('missing_scope', TokenScopeError(required_scope='files.metadata.read'))
```

Begrijp ik het goed dat je de 'App", voor zover mogelijk, wel alle rechten hebt gegeven binnen de App folder?

PascalBu, 2020-10-27:
Ja, dat dacht ik wel. Ik heb voor de zekerheid nogmaals geprobeerd, maar nu met mijn eigen Dropbox account. En nu werkt het wel. Misschien dat ik de volgorde nu heb veranderd? Dus eerst rechten geven en longlived token. En dan pas token genereren. Anders snap ik niet waarom het nu wel werkt.

Misschien wel goed om de wiki te updaten...

Bedankt voor de hulp iig!

closed 2020-10-27 by PascalBu

dennissiemensma, 2020-10-27:
Bedankt voor je update. Mochten meer gebruikers er tegenaan lopen, dan zal ik het bijwerken!