dsmrreader/dsmr-reader #1168: Remote Postgres db

ilionxsjambor, 2020-10-28:
Voor het stabiel houden van je SD kaart op de Pi zou het (denk ik) wel prettig zijn als ook een andere dan de lokaal geïnstalleerde Postgres db gebruikt kan worden (bijv in een clouddienst). Even ter verduidelijking: Ik bedoel niet een datalogger-pi (oid) die het doorstuurt naar een "echte" installtie waar dan backend, db en webinterface draaien, maar dat de postgres db ergens anders draait. Is daar iets voor om dat in te kunnen stellen? Ben al een dag alles aan het doorspitten maar vind eigenlijk nergens de instellingen mbt op welke host etc postgres staat.

jeroenpeters1986, 2020-10-28:
Hallo @ilionxsjambor, dat kan hoor.

Als je de Toegelichte installatie van Dennis hebt gevolgd heb je wellicht deze stap uitgevoerd:
https://dsmr-reader.readthedocs.io/nl/v4/installation/explained.html#application-configuration-setup

Daar kopieer je een instellingen bestand met standaardwaardes. In dat .env-bestand kun je dit aanpassen.

Je ziet standaard dit staan: 
```
DJANGO_DATABASE_HOST=localhost
DJANGO_DATABASE_PORT=5432
```

Ook staan daar nog andere database instellingen, die kun je allemaal aanpassen. Zo kun je ook een database in de cloud of op een andere machine gebruiken.

dennissiemensma, 2020-10-28:
Aanvullend staat hier de hele lijst met instellingen: https://dsmr-reader.readthedocs.io/nl/v4/env_settings.html

Degene die Jeroen noemt zie je daar ook terug. Mocht je er desondanks niet uitkomen, laat maar weten!

closed 2020-10-28 by dennissiemensma

svenjambor, 2020-10-29:
@dennissiemensma & @jeroenpeters1986 Geen idee hoe ik dit heb kunnen missen, maar ik heb het .env bestand gevonden & aangepast & getest. Works like a charm.  Ik zal er een kleine blogpost van maken voor als anderen hier ook tegen aan lopen. Thanks!!!!

iepmaatje, 2021-01-15:
@svenjambor Hallo Sven, heb jij een blogpost geschreven die ik zou kunnen lezen over wat je hebt gedaan ?

svenjambor, 2021-01-15:
> @svenjambor Hallo Sven, heb jij een blogpost geschreven die ik zou kunnen lezen over wat je hebt gedaan ?

Ja, maar vergeten online te zetten. Doe ik morgen voor je. 

Het komt er op neer dat ik 1) een postgresdb in Azure heb opgesponnen (ik heb het geluk dat dat gratis kan) , 2) daar een backup van de bestaande db heb teruggezet, en 3) de variabelen die Jeroen aan heeft gegeven in het .env bestand heb aangepast. Daarna werkte het prima.