dsmrreader/dsmr-reader #1169: backup naar nas werkt echter zijn de backupfiles niet compressed

karia001, 2020-10-28:
Troubleshooting: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

(Heb je DSMR-reader v4.4 of hoger? Zet dan hieronder ook de output van "dsmr_debuginfo") 

```

OS
    Python version                                                                                                    v3.7.3
    Platform                                                                                                  Linux (armv7l)
    System                                                                          Linux-5.4.72-v7+-armv7l-with-debian-10.6

DSMR-READER
    Version                                                                                                             v4.5
    Backend sleep                                                                                                      1.0 s
    Datalogger sleep                                                                                                   5.0 s
    Retention cleans up after                                                                                          672 h
    Telegram parser                                                                                                      "4"
    Database engine/vendor                                                                                        postgresql

DATA
    Telegrams                                                                                                               
      - total (est.)                                                                                                  720261
      - version (latest reading)                                                                                        "50"
    Consumption records                                                                                                     
      - electricity (est.)                                                                                             33576
      - gas (est.)                                                                                                      6681

POSTGRESQL SIZE OF LARGEST TABLES (> 5 MB)
    public.dsmr_datalogger_dsmrreading                                                                                179 MB
    public.dsmr_consumption_electricityconsumption                                                                     14 MB

```

---

ik heb de backup lokatie aangepast naar mijn QNAP NAS door deze te mounten, nu worden de backup's netjes weggeschreven in de opgegeven mount lokatie, echter zie ik dat de backupfiles nu niet worden compressed.
zip level in de web interface heb ik op 9 staan.
wat doe ik fout!!!
ps dmsr draaiend krijgen op de NAS zelf is mij helaas niet gelukt dus toch maar op de PI laten draaien

dennissiemensma, 2020-10-28:
Bedankt voor je melding. Ik begrijp niet helemaal wat je bedoelt, hoe zien de file die je ziet er uit qua naam en extensie?

Verder kun je het beste je eigen logfiles doorspitten: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#log-files
Ik verwacht dat je in de log van `dsmr_backend` eventuele fouten zult zien. 

Zo niet, dan kun je debug logging tijdelijk aanzetten en kijken wat die precies doet op het moment van het maken van de backup. In dat geval zoek je deze output:

```
[2020-10-28 23:45:40,429] DEBUG    SP: Running "Create daily backup" (dsmr_backup.services.backup.run)
[2020-10-28 23:45:40,440] INFO      - Creating new partial backup: /home/dennis/PycharmProjects/dsmr-reader/backups/archive/2020/10/dsmrreader-postgresql-partial-backup-2020-10-28.sql
[2020-10-28 23:45:40,548] DEBUG     - Backup exit code: 0
[2020-10-28 23:45:40,591] INFO      - Created and compressed statistics backup: /home/dennis/PycharmProjects/dsmr-reader/backups/archive/2020/10/dsmrreader-postgresql-partial-backup-2020-10-28.sql.gz
[2020-10-28 23:45:40,592] INFO      - Creating new full backup: /home/dennis/PycharmProjects/dsmr-reader/backups/dsmrreader-postgresql-backup-Wednesday.sql
[2020-10-28 23:45:41,167] DEBUG     - Backup exit code: 0
[2020-10-28 23:45:45,162] INFO      - Created and compressed full backup: /home/dennis/PycharmProjects/dsmr-reader/backups/dsmrreader-postgresql-backup-Wednesday.sql.gz
[2020-10-28 23:45:45,183] DEBUG    SP: Rescheduled "Create daily backup" to 2020-10-29 07:00:00+01:00 (ETA 7:14:14.816650)
```

karia001, 2020-10-29:
hoi Dennis

hier een voorbeeld filename: dsmrreader-postgresql-backup-Thursday.sql ipv de file die in de backup can de pi nog zelf staat dsmrreader-postgresql-backup-Friday.sql.gz.
ik zal zoeken in mijn dsmr_backend log file of i eventuele meldingen zoals je omschrijft kan zien.

dennissiemensma, 2020-10-30:
De melding is dan duidelijk, de permissies staan niet goed voor die mount. 

Wellicht dat je die wat losser in kan stellen voor de bovenliggende folder. En anders een diepere folder maken waar DSMR-reader alleen daar bij kan.

karia001, 2020-10-30:
Hoi Dennis,

een diepere folder aangemaakt en nu werkt het goed, dank je wel voor je hulp

closed 2020-10-30 by karia001