dsmrreader/dsmr-reader #1171: [Fout] Changelog niet geupdate. Kan geen info over 4.5 vinden op readthedocs.io

Rawh, 2020-10-30:
Zoals in de topic;

Ik zie 4.4.3 als laatste entry terwijl ik net de upgrade naar 4.5 heb gedaan.

https://dsmr-reader.readthedocs.io/en/v4/changelog.html

Overigens vind ik het nieuwe support kopje in het menu links een goede uitbreiding. Veel informatie samengevat in 1 lokatie = fijn.

dennissiemensma, 2020-10-30:
Bedankt voor je melding. Er zijn wat issues met ReadTheDocs,  in de [Nederlandse variant](https://dsmr-reader.readthedocs.io/nl/v4/changelog.html) staan ze wel. 

En anders kun je altijd nog kijken bij [de reelases in Github zelf](https://github.com/dennissiemensma/dsmr-reader/releases), daar dupliceer ik ze sowieso eveneens.

Mogelijk dat ik dit project dit weekend verplaats naar een ander account op Github. Ik zal kijken of dat dan ook het issues met de docs oplost.

dennissiemensma, 2020-10-30:
Ik heb ReadTheDocs opnieuw handmatig laten bouwen, nu lijkt het wel goed en staat 4.5 erbij (na een harde browser refresh): https://dsmr-reader.readthedocs.io/en/v4/changelog.html

closed 2020-10-30 by dennissiemensma