dsmrreader/dsmr-reader #1172: DSMR laten weten dat MQTT Broker weer live is

poudenes, 2020-10-30:
Hoi Hoi,

Ik heb gisteren en vandaag mijn MQTT Broker even moeten resetten. Het is mij opgevallen dat DSMR daarna geen nieuwe payloads meer stuurt. Ik heb al geprobeerd:

```
topic: homeassistant/status
payload: online
```

```
topic: hass/status
payload: online
```

maar dit werkte niet. Enige wat werkt is dsmr een restart geven:

`sudo supervisorctl restart all`

Is er iets wellicht dat ik niet goed heb staan waardoor DSMR niet de online payload ziet?

Greets,

dennissiemensma, 2020-10-30:
Bedankt voor je melding. Welke versie draai je momenteel? Dit zou opgelost moeten zijn in DSMR-reader v4.5.

poudenes, 2020-10-31:
> Bedankt voor je melding. Welke versie draai je momenteel? Dit zou opgelost moeten zijn in DSMR-reader v4.5.

Ik zit nog op 4.4.3. Ik ga hem updaten. Zou trouwens mooi zijn als een nieuwe update versie via MQTT ook wordt aangekondigd en/off via notify (mijn geval pushover)

closed 2020-10-31 by poudenes