dsmrreader/dsmr-reader #1173: DSMR v4.5 2 waarschuwingen over telegram aantal en database size. Zijn deze uit te zetten?

Rawh, 2020-10-30:
Troubleshooting: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

```

OS
    Python version                                                                                                    v3.7.3
    Platform                                                                                                  Linux (armv7l)
    System                                                                         Linux-5.4.51-v7l+-armv7l-with-debian-10.6

DSMR-READER
    Version                                                                                                             v4.5
    Backend sleep                                                                                                      1.0 s
    Datalogger sleep                                                                                                   2.0 s
    Retention cleans up after                                                                                            - h
    Telegram parser                                                                                                      "4"
    Database engine/vendor                                                                                        postgresql

DATA
    Telegrams                                                                                                               
      - total (est.)                                                                                                24150600
      - version (latest reading)                                                                                        "50"
    Consumption records                                                                                                     
      - electricity (est.)                                                                                            656193
      - gas (est.)                                                                                                    126987

ISSUES
    Approximately 24150600 readings stored, consider data cleanup (if not already enabled)  Since 2020-10-30 09:32:57.564612+01:00
    Database growing large: 6971 MB, consider data cleanup (if not already enabled)   Since 2020-10-30 09:32:57.559633+01:00

POSTGRESQL SIZE OF LARGEST TABLES (> 5 MB)
    public.dsmr_datalogger_dsmrreading                                                                               6676 MB
    public.dsmr_consumption_electricityconsumption                                                                    268 MB
    public.dsmr_consumption_gasconsumption                                                                             13 MB

```

Ik heb sinds vorige week mijn pi 3 vervangen door de pi 4 (4gb) versie. Ik laat daarbij de database niet schrijven naar mn SD kaartje maar naar een postgresql server die op mijn NAS staat. 

Sinds de upgrade naar 4.5 zie ik dat er een rood rondje met een 2 daarin staat en geeft hij twee waarschuwingen aan. 
[image]

Ik kan niet goed beoordelen of de eerste waarschuwingen "erg" is of niet.
De tweede waarschuwing over de database is mij minder relevant omdat ik die ergens anders host.

Ik heb opzettelijk geen retentie van de database aangezet om ik die graag een jaartje (of misschien meer) wil opslaan, ter vergelijking van de stats e.d. (of in geval van crash, dat ie die opnieuw aan kan maken voor op de site). Ruimte van de NAS speelt daar geen rol.

Is er een mogelijkheid om per waarschuwing deze te onderdrukken of af te vinken als "negeer maar" ?

dennissiemensma, 2020-10-30:
Bedankt voor je melding. Je bent uiteraard vrij om alle data te bewaren. Weet wel dat Postgresql niet oneindig schaalt met grote datasets, zelfs met goede DB-indexes en een SSD storage. Waarschijnlijk kom je een heel eind, voor lange tijd, maar weet dat het ergens ophoudt.

Er is momenteel nog geen officiele optie om de melding uit te zetten. Ik zal dat in de volgende versie opnemen.

Voor nu kun je, mits je geen Docker draait, het overschrijven in `dsmrreader/settings.py`. Standaard zou de inhoud zoiets moeten zijn:
```
"""
    This used to be the local settings file until DSMR-reader 4.x.
    Please use .env instead. See: https://dsmr-reader.readthedocs.io/en/latest/faq/v4_upgrade.html
"""
from dsmrreader.config.production import *
```

Je kunt dat aanpassen naar:
```
"""
    This used to be the local settings file until DSMR-reader 4.x.
    Please use .env instead. See: https://dsmr-reader.readthedocs.io/en/latest/faq/v4_upgrade.html
"""
from dsmrreader.config.production import *

DSMRREADER_STATUS_WARN_OVER_EXCESSIVE_READING_COUNT = 20 * 1000 * 1000  # In millions
DSMRREADER_STATUS_WARN_OVER_EXCESSIVE_DATABASE_SIZE = 1 * 1000 * 1000 * 1000  # In GB
```

En dan zelf de waardes pakken die je wilt. Bijvoorbeeld 100 miljoen metingen en 25 GB grootte.

dennissiemensma, 2020-10-30:
Bijna vergeten: Herstart het webinterface-proces (of alle processen) om de instellingen door te voeren.

dennissiemensma, 2020-11-01:
In `v4.6` wordt het mogelijk om een nieuwe instelling `DSMRREADER_IGNORE_STORAGE_SIZE_WARNINGS=true` te gebruiken.

closed 2020-11-01 by dennissiemensma

dennissiemensma, 2020-11-01:
Uitgebracht in [v4.6](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.6.0)

RichieB2B, 2020-11-02:
De nieuwe `DSMRREADER_SUPPRESS_STORAGE_SIZE_WARNINGS` setting werkt prima, maar waarom `DSMRREADER_STATUS_WARN_OVER_EXCESSIVE_DATABASE_SIZE` en `DSMRREADER_STATUS_WARN_OVER_EXCESSIVE_READING_COUNT` niet zelf instelbaar maken? Ik vind de warnings an sich een prima idee, alleen de defaults een beetje laag.

dennissiemensma, 2020-11-02:
Bedankt voor je input. Ik denk dat je deze waardes het beste kunt zien als een waarschuwing vanuit de ontwikkelaar, om aan te geven dat dit **mogelijk** invloed kan hebben op opslag, performance etc. 
Daarbij is het belangrijk om te weten dat de gemiddelde installatie op een Pi met SD-kaart draait. Want waar jij de waardes bijvoorbeeld wat laag vindt, vind ik ze namelijk nu nog vrij hoog voor een doorsnee-installatie. 

Sommige gebruikers, zoals de maker van dit issue en jijzelf, hebben waarschijnlijk andere hardware of wensen. Jullie krijgen de melding, weten er dus van en dat is prima. Je bepaalt namelijk zelf of je de melding wilt uitschakelen (en niets doen) of actie ondernemen (bijvoorbeeld retentie op een jaar + melding uitschakelen).

Het issue is namelijk dat er regelmatig issues zijn van gebruikers waarbij alles traag is en opschonen duurde lang. Eerder ingrijpen is beter dan achteraf.