dsmrreader/dsmr-reader #1175: FreeBSD compatibility

Wieter, 2020-11-01:
First of all, thanks for all your work on this neat piece of software!

To increase interoperability over various systems, may I suggest to implement a different shebang for all .sh scripts?

For example, while installing on FreeBSD I had to edit the file:
https://github.com/dsmrreader/dsmr-reader/blob/0ee58370f43c6ebe32564660b47f212b518c227f/tools/generate-secret-key.sh#L1
and replace` #!/bin/bash` with `#!/usr/bin/env bash`

Note: at FreeBSD, bash is not installed by default! use `pkg install bash`

Additionally, in https://github.com/dsmrreader/dsmr-reader/blob/0ee58370f43c6ebe32564660b47f212b518c227f/dsmrreader/config/production.py#L6 the `STATIC_ROOT` is statically defined as '/var/www/dsmrreader/static'. As different systems (and users) have different default and preferred locations for www data, it might be a good idea to be able to set this as parameter in the .env(.local) file. 

Now, whenever there is an update, I first have to `git reset --hard`, then `pull`, and then re-incorporate above changes. 

For the setup configuration in FreeBSD, I think people can manage. It deviates a bit from the documentation and example configs (mostly due to different paths), and I myself prefer to use uwsgi in emperor mode.

Wieter, 2020-11-01:
Coming back to the `STATIC_ROOT`, it might be better to set in https://github.com/dsmrreader/dsmr-reader/blob/0ee58370f43c6ebe32564660b47f212b518c227f/dsmrreader/config/production.py#L6 to something universal like `os.path.join(BASE_DIR, 'static')`

dennissiemensma, 2020-11-01:
Thanks for reporting! I've updated the shebangs and added an env var (`DJANGO_STATIC_ROOT`) for overriding `STATIC_ROOT`. These will be available in `v4.6`.

closed 2020-11-01 by dennissiemensma

dennissiemensma, 2020-11-01:
Uitgebracht in [v4.6](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.6.0)