dsmrreader/dsmr-reader #1180: [Question] How is usage calculated?

zuidwijk, 2020-11-02:
Just a question... How is the usage calculated? 

For example: today my container stopped logging between 3am and 10:30am. Now those telegrams are gone, can't get them back ;) yet we do have totals ... the moment the day started, and the last one at the end of the day.

So will this loss only effect on the graph? Or also on the day total?

dennissiemensma, 2020-11-02:
Day consumption is calculated using the first and last reading of the day: https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_consumption/services.py#L250-L268

It _should_ be fine. I'm not sure which data will be missing in the Archive, but you can check it out tomorrow morning. Probably just some missing hour totals, which is okay.

zuidwijk, 2020-11-02:
For hour totals and current graph it's too bad... most important (to me) is daily, monthly and yearly totals... So I guess it's fine indeed 👍

closed 2020-11-02 by zuidwijk