dsmrreader/dsmr-reader #1181: Initial zoom for RECENT GAS CONSUMPTION

RichieB2B, 2020-11-02:
Ik heb in de Frontend instellingen de defaults staan voor **Live graphs initial zoom** (10) en **Live graphs hours range** (24), maar de RECENT GAS CONSUMPTION en RECENT TEMPERATURES grafieken trekken zich niet aan van de initial zoom instelling en laten 24 uur zien.

[image]

```

DSMR-READER
    App / Python / Database                                                        v4.6 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                  1.0s / 0.5s / 8064h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                           14498900
    Consumption records electricity / gas (est.)                                              502143 / 106681

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                2907 MB

```

dennissiemensma, 2020-11-02:
Bedankt voor je melding. Dit is 'bewust' geweest omdat vooral de elektriciteitsmetingen, in sommige extreme situaties, een factor 300 tot 3600 meer datapunten hebben t.o.v. gas en temperatuur. Om ervoor te zorgen dat die elektra-grafieken in die situaties nog enigzins schalen, is deze instelling gemaakt.

Ik zou nog wel de tekst wat kunnen verduidelijken, of de grafieken er ook op kunnen aansluiten, maar ik weet niet of het echt een issue is in de praktijk.

RichieB2B, 2020-11-02:
Ik heb net geüpgraded vanaf v2.9 en daar was de gas grafiek wel initieel ingezoomd. Dat lijkt me ook logisch: ik ben meestal geïnteresseerd in de recente data (laatste half uur). Een overzicht van 24 uur is dan vaak niet duidelijk, zeker als daar pieken inzitten waardoor de recente datapunten nauwelijks zichtbaar zijn. Voor de temperatuur is het wat minder kritisch denk ik.

dennissiemensma, 2020-11-02:
Pfoe, v2.9 is voor mij een eeuwigheid geleden en sindsdien zijn heel veel wijzigingen geweest. Ik zal er in een toekomstige release wel een keer naar kijken.

dennissiemensma, 2020-11-08:
Ik zal kijken of het mogelijk is om de grafiek van gasmetingen ook op een week te zetten en dan de zoom-instelling daarvoor te gebruiken. Dit i.vm. DSMR-v4-meters, die gelimiteerd zijn op uurlijkse gasmetingen.

dennissiemensma, 2020-11-08:
@RichieB2B gezien je een v5 meter hebt: Hoeveel gasmetingen worden er ongeveer per uur bij jou gelogd in DSMR-reader? Zijn dat er altijd ~12 per uur? (elke 5 minuten eentje)

RichieB2B, 2020-11-08:
Ja, elke 5 minuten dus 12 per uur.

dennissiemensma, 2020-11-08:
Dank voor de input. In dat geval zal ik voor een volgende release scheiding maken in de meterversie. In mijn geval (v4) verandert er niets, in jouw geval (v5) zal die de door jou voorgestelde fix gebruiken. 

Dat laatste zou dan goed moeten werken voor v5 gebruikers, zonder dat de grafiek 'te weinig' meetpunten heeft (toevallig ook 24 meetpunten, bij de standaardinstelling van 10% in de laatste 24 uur, uit 240 meetpunten).

dennissiemensma, 2020-11-08:
Doorgevoerd voor de volgende release.

closed 2020-11-08 by dennissiemensma

dennissiemensma, 2020-11-09:
Uitgebracht in [v4.7](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.7.0)

dennissiemensma, 2020-11-09:
De tekst onder de grafieken is hierbij overigens eveneens wat uitgebreid om meer toelichting te geven.

RichieB2B, 2020-11-09:
Top, bedankt!